OctoPrint/OctoPrint #3068: UnicodeDecodeError: 'utf8' codec can't decode byte 0xec in position 545: invalid continuation byte

michaelkuty, 2019-03-03:
#### What were you doing?

4. See error

```
(oprint) root@octopi:/home/pi/OctoPrint# octoprint 
Starting the server via "octoprint" is deprecated, please use "octoprint serve" from now on.
2019-03-03 00:22:12,104 - octoprint.startup - INFO - ******************************************************************************
2019-03-03 00:22:12,105 - octoprint.startup - INFO - Starting OctoPrint 1.3.10
2019-03-03 00:22:12,106 - octoprint.startup - INFO - ******************************************************************************
2019-03-03 00:22:13,114 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins, /root/.octoprint/plugins and installed plugin packages...
2019-03-03 00:22:13,496 - octoprint.plugin.core - ERROR - Error loading plugin softwareupdate
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 847, in _import_plugin
    instance = imp.load_module(key, f, filename, description)
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/softwareupdate/__init__.py", line 27, in <module>
    from octoprint.server.util.flask import restricted_access, with_revalidation_checking, check_etag
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/__init__.py", line 9, in <module>
    from octoprint.vendor.sockjs.tornado import SockJSRouter
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/vendor/sockjs/tornado/__init__.py", line 3, in <module>
    from .router import SockJSRouter
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/vendor/sockjs/tornado/router.py", line 11, in <module>
    from octoprint.vendor.sockjs.tornado import transports, session, sessioncontainer, static, stats, proto
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/vendor/sockjs/tornado/transports/__init__.py", line 3, in <module>
    import octoprint.vendor.sockjs.tornado.transports.pollingbase
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/vendor/sockjs/tornado/transports/pollingbase.py", line 9, in <module>
    from octoprint.vendor.sockjs.tornado import basehandler
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/vendor/sockjs/tornado/basehandler.py", line 13, in <module>
    from tornado.web import asynchronous, RequestHandler
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/web.py", line 87, in <module>
    from tornado import gen
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/gen.py", line 113, in <module>
    from singledispatch import singledispatch  # backport
UnicodeDecodeError: 'utf8' codec can't decode byte 0xec in position 545: invalid continuation byte

```

#### What did you expect to happen?

#### What happened instead?

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPrint

1.3.10 (stable)

#### Operating System running OctoPrint

OctoPi

#### Printer model & used firmware incl. version

#### Browser and version of browser, operating system running browser

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

foosel, 2019-03-04:
Thank you for reporting this!

Sadly your report lacks some of the information that is needed in order to look into this issue:

  * whether you have been able to reproduce this issue in safe mode too
  * the operating system running OctoPrint (name *and* version please)
  * unmodified full length [`octoprint.log`](https://discourse.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299/1)

(If you need help with locating this stuff, please take a look [into the Contribution Guidelines](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#what-should-i-include-in-a-bug-report) which explains everything - and if anything is still unclear then just ask)

Could you please provide that so this issue can be analyzed and ideally reproduced and fixed? Thank you!

michaelkuty, 2019-03-12:
Add full log and os version.

foosel, 2019-04-08:
You have still not provided the information whether this also happened in safe mode as well, but regardless of that, it looks like some parts of the OctoPrint installation got corrupted somehow. Did you ever just pull the power on the Pi without shutting it down properly first by any chance?

Try logging into your OctoPi via SSH. Then delete the corrupt installation and reinstall:

  * `rm -r ~/oprint` 
  * `virtualenv ~/oprint`
  * `~/oprint/bin/pip install https://get.octoprint.org/latest`

If that doesn't fix it I'm out of ideas.

In any case this is not a bug in OctoPrint but rather something caused by your install being broken in some way. Closing, comments are still possible.

closed 2019-04-08 by foosel