OctoPrint/OctoPrint #3070: [Request] Serial over Ethernet

Danny89530, 2019-03-04:
Hello everyone,
I'm here to propose a very small change at __openSerial_ and _serialList_ functions.
I've 2 printers directly connected to a Pi ZeroW and ESP8266 respectively:
These two printers are connected via WiFi to a remote server with two LXC containers where OctoPrint was installed from source.
These two little boards was configured for create a remote socket on port 4000 (in my case) using the _rfc2217 standard_ at startup.

Here there are two example on how to do this on the printer connected boards:
[Python](https://github.com/pyserial/pyserial/tree/master/examples) (for Pi Zero W)
[Arduino](https://github.com/jeelabs/esp-link) (For ESP8266)

I've used this configuration in the past six month without any kind of problem or dataloss but to be able to use this configuration, I've modified the functions in **comm.py** adding two lines of code:

### _openSerial FROM:
```
			[...]
			# connect to regular serial port
			self._log("Connecting to: %s" % port)
			if baudrate == 0:
				baudrates = baudrateList()
				serial_obj = serial.Serial(str(port),
				                           baudrates[0],
				                           timeout=read_timeout,
				                           write_timeout=10000,
				                           parity=serial.PARITY_ODD)
			else:
				serial_obj = serial.Serial(str(port),
				                           baudrate,
				                           timeout=read_timeout,
				                           write_timeout=10000,
				                           parity=serial.PARITY_ODD)
			[...]
```
TO:
```
			[...]
			# connect to regular serial port
			self._log("Connecting to: %s" % port)
			if '://' in str(port):
				serial_obj = serial.serial_for_url(str(port))
			else:
				if baudrate == 0:
					baudrates = baudrateList()
					serial_obj = serial.Serial(str(port),
					                           baudrates[0],
					                           timeout=read_timeout,
					                           write_timeout=10000,
					                           parity=serial.PARITY_ODD)
				else:
					serial_obj = serial.Serial(str(port),
					                           baudrate,
					                           timeout=read_timeout,
					                           write_timeout=10000,
					                           parity=serial.PARITY_ODD)
			[...]
```

### serialList FROM:
```
	[...]
	additionalPorts = settings().get(["serial", "additionalPorts"])
	for additional in additionalPorts:
		baselist += glob.glob(additional)
	[...]
```
TO:
```
	[...]
	additionalPorts = settings().get(["serial", "additionalPorts"])
	for additional in additionalPorts:
		if '://' in additional:
			baselist.append(additional)
		else:
			baselist += glob.glob(additional)
	[...]
```

I'm sure there is a more elegant way to do this but, in any case, do you wanna consider add my changes in the official release?

Regards,
Danilo

OutsourcedGuru, 2019-03-10:
I have run through the scenario and have tested it on a Raspberry Pi 3B (OctoPi 0.16.0) rig -> Raspberry Pi Zero W (Raspbian Stretch Lite 11/13/2018) with a MEGA2560 controller board and it appears to work as suggested.

I have created a [step-by-step tutorial](https://github.com/OutsourcedGuru/Octo-Redirect) for this approach if it will help. I have bumped the GPU memory to the minimum since the Zero only has 512MB of memory.

It may be worth upgrading the remote computer to a Raspberry Pi 3A+ since it would have the expected Type A style of USB connector.

redge76, 2020-11-11:
Any reason why this is not merged ? It seems like a very simple and clever solution to access a remote printer over IP network.
Is there any other simple solution ?
My octoprint is currently running on a small orangePi and for performance reasons, I would like to move it in a docker container running on a much more powerful and easy to maintain system.
I don't want to connect my printer directly to my docker server. 
So what are my options ?
This RFC2217 seems like a perfect solution. What are you using to connect your printer to you octoprint docker container.

frenck, 2020-11-11:
> Any reason why this is not merged ?

This is not a PR...

redge76, 2020-11-11:
:-P hooo yes.. That may explain this situation :-D

foosel, 2020-11-11:
> Is there any other simple solution ?

http://www.dest-unreach.org/socat/

hellerbarde, 2021-01-15:
I wrote a quick proof of concept plugin to do this. No need to complicate the core source.  I will publish it over the weekend. And leave a comment here.
Out of curiosity, which esp firmware is able to properly talk to a rfc2217 client?

hellerbarde, 2021-01-16:
Here we go :)
https://github.com/hellerbarde/OctoPrint-Network-Printing
I have not officially entered it into the plugin-manager repository, but I will do so soon. Also, it is Python 3 only, because I have no interest in supporting Python 2 past it's EOL in my spare time.

LazeMSS, 2021-03-31:
Possible by plugin:
https://github.com/hellerbarde/OctoPrint-Network-Printing

- @hellerbarde you should really submit it to plugin repro :)

hellerbarde, 2021-04-02:
> Possible by plugin:
> https://github.com/hellerbarde/OctoPrint-Network-Printing
> 
>     * @hellerbarde you should really submit it to plugin repro :)

Umm... It's very untested, so far. I don't have the capacity right now to deal with incoming bug reports and support requests. And it can be installed through the URL directly, for motivated people.

redge76, 2021-04-15:
I was able to use the hellerbarde's plugin to connect my Octoprint running in a docker container on a x86 system to an Orange Pi serving as network serial server connected by USB to my printer.

I have an Anycubic I3 Mega running a Marlin firmware 1.1.9
I used a "socket://" url in octoprint,
The nw serial server is [rfc2217_server.py of the package pySerial](https://github.com/pyserial/pyserial/blob/master/examples/rfc2217_server.py). It was the only option I found to support the 250000 baud rate of the printer as I could not make any other raw network serial server working (socat or ser2net). The linux cp210x module for the USB serial converted does not support the TIOCGSERIAL ioctl and setserial does not work.  ("Cannot get serial info: Invalid argument"  error message)
AFAIK it's because termios2 is not supported by all the command line tools. pySerial does not have this limitation.

I had to add a "baudrate=250000" to the rfc2217_server.py file to force the correct serial speed. 
modify the [line](https://github.com/pyserial/pyserial/blob/0e7634747568547b8a7f9fd0c48ed74f16af4b23/examples/rfc2217_server.py#L137 ) to this: ser = serial.serial_for_url(args.SERIALPORT, baudrate=250000, do_not_open=True)
I may also have to remove this [ line ](https://github.com/pyserial/pyserial/blob/0e7634747568547b8a7f9fd0c48ed74f16af4b23/examples/rfc2217_server.py#L59) even though I doubt it will make any difference for G-code.

I was not able to use a rfc2217:// url as the current pySerial implementation does not support  [the reconfiguration of the write timeout value] (https://github.com/pyserial/pyserial/blob/0e7634747568547b8a7f9fd0c48ed74f16af4b23/serial/rfc2217.py#L507)
Octoprint set a timeout of 0 [ here ](https://github.com/OctoPrint/OctoPrint/blob/a811f191f2992b8e8f4a10ed337ce1f2421fd165/src/octoprint/util/comm.py#L3623) and it prevent any usage of a rfc2217 serial object.
I tried to comment out the line throwing the exception but... that's not enough.

I haven't tried to print something. I'll do it tomorrow and I hope the lack of flow control will not be an issue.

hellerbarde, 2021-04-15:
> I was not able to use a rfc2217:// url as the current pySerial implementation does not support  [ the reconfiguration of the write timeout value ]( [...]

@redge76 that's cool to hear and I'm curious how it'll work out for you. Would you be interested in connecting on discord (or whatever chat system) to talk about it. It would be very useful to have another person to exchange ideas and experiences before a more wide release. If you're up for it, write to me on my github email address.

redge76, 2021-04-15:
My tests were successful. I'm quite happy with this solution. 
Octoprint is running on my home server. Docker brings easy management and backup.
The performance is great. (not that it was really an issue when it was running on my small Orange Pi, but not plugins like Arc welder or ffmpeg to compress the timelapse run in seconds)
The cheapest SBC should be able to run the wifi to serial bridge. (the  [Orange Pi i96 256MB](https://fr.aliexpress.com/item/32819068354.html) seems a great option for only 6$)
It still need to test stability on very long prints. But I don't see why it would fail.