OctoPrint/OctoPrint #3071: Application keys don't work with access control disabled

cyounkins, 2019-03-04:
#### What were you doing?

1. Go through the setup wizard, disabling access control
2. Create an Application Key (in my case, this was for Cura)
3. Attempt to use new API key via something like `curl -H "X-Api-Key: d9dfabcdefa7960d9e36f1f13c0da3d3" http://pluto.lan:5000/api/connection`

#### What did you expect to happen?

I expected a JSON object response.

#### What happened instead?

"Invalid API key"

#### Did the same happen when running OctoPrint in safe mode?

Yes.

#### Version of OctoPrint

OctoPrint 1.3.10

#### Operating System running OctoPrint

Linux within custom docker environment

#### Printer model & used firmware incl. version

#### Browser and version of browser, operating system running browser

#### Link to octoprint.log

https://gist.github.com/cyounkins/6fda15b4b476775945214a0bd17bd90a

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

#### Other

Below I use the "API key", not application key.

```
$ curl -H "X-Api-Key: ABCD" http://pluto.lan:5000/api/users
{}

$ cat data/appkeys/keys.yaml
dummy:
- {api_key: d9dfabcdefa7960d9e36f1f13c0da3d3, app_id: Cura}
```

#### Possible cause

I believe either `get_user_for_apikey` is returning None or a reference to the dummy user which subsequently fails in `loginUser`. https://github.com/foosel/OctoPrint/blob/112a8b9c8bf7f5895f6a29dbe3d98ea4d9e0f400/src/octoprint/server/util/__init__.py#L74

cyounkins, 2019-03-05:
I re-ran the setup wizard and enabled access control, setting up the user 'admin'. I deleted the previous application key and created a new one. Now:

```
$ cat octoprint/basedir/data/appkeys/keys.yaml
admin:
- {api_key: ad6e95b44f443d258a913f68c9b7bfd7, app_id: Cura}
dummy: []

$ curl -H "X-Api-Key: ad6e95b44f443d258a913f68c9b7bfd7" http://pluto.lan:5000/api/users
{
  "users": [
    {
      "active": true,
      "admin": true,
      "apikey": null,
      "name": "admin",
      "settings": {},
      "user": true
    }
  ]
}
```
Enabling access control changed the user the application key was associated with and "fixed" the API access.

There's still a problem here. Should application keys be provisioned when access control is disabled? Should any key work when it's disabled? I do believe the behavior seen is problematic.

foosel, 2019-03-22:
Sorry, this somehow slipped through, taking a look now.

foosel, 2019-03-22:
API keys from the plugin should now be fixed when access control is disabled. See the above commit, pushed to `maintenance`, soon `devel` and to be part of 1.3.11.

However, since if you have access control disabled there are no users the users API in this case will always return an empty result, just as you observed.

cyounkins, 2019-03-24:
Confirmed fixed! Thank you.

commit 1d2ceefb (2019-03-22): 🐛 appkeys: Fix in case of disabled access control

Closes #3071
closed 2019-05-14 by foosel