OctoPrint/OctoPrint #3074: [Request] Update dependencies to recent versions (Especially scandir)

positron96, 2019-03-05:
**Is your feature request related to a problem? Please describe.**
I've tried to install Octoprint to an old netbook. I used compilation from source variant since I already had a working OS. Installation failed and I discovered that required lib scandir was the cause. 
Digging further, I discovered that version required in setup.py is scandir==1.3 (released in 2016) and it is provided only in source form on PyPI and requires C compiler. That required installation of a hundreed MBs of gcc toolchain and python-dev package for headers (the latter part was not obvious and I spent some time googling). 

The current version of scandir is 1.9 and binary archives are provided for that version on PyPI, so no gcc required (that would save a lot of time and disk space when your PC is not very fast)

**Describe the solution you'd like**
The proposal is to update library dependecies to more recent versions (or at least the ones that require compilation to fix this particular issue). I believe many of them will work without any changes in code. 
Apart from inconveniences during install, this would also cover some security holes in libraries.

foosel, 2019-03-22:
The above commit updates various dependencies (part of release preparations anyhow). Pushed to `maintenance`, soon `devel`, to be part of 1.3.11.

closed 2019-05-14 by foosel

positron96, 2019-05-16:
amazing!

commit 3d442202 (2019-03-22): ⬆ Updated various dependencies

Closes #3074