OctoPrint/OctoPrint #3075: [Request] Filament usage should take into account M605 duplication/mirroring commad

JanneMantyharju, 2019-03-05:
**Is your feature request related to a problem? Please describe.**
At least Marlin firmware supports M605 command on printers with two printing heads to duplicate or mirror on second tool what the first tool is printing. OctoPrint does not currently recognize this and thus the filament counter plugins can't take into account the filament usage on second tool

**Describe the solution you'd like**
When M605 S2, S4, S5 or S6 command is seen, filament usage of tool 0 should be added to tool 1 usage

foosel, 2019-03-06:
Where can one find a full documentation of this command and what the available parameters do exactly on what kind of firmware? [The RepRap wiki](https://reprap.org/wiki/G-code#M605:_Set_dual_x-carriage_movement_mode) doesn't go into a lot of detail.

JanneMantyharju, 2019-03-06:
Good question. I found this information from Marlin source code: 

> 
> // There are a few selectable movement modes for dual x-carriages using M605 S<mode>
> //    M605 S0: Full control mode. The slicer has full control over x-carriage movement
> //    M605 S1: Auto-park mode. The inactive head will auto park/unpark without slicer involvement
> //    M605 S2 [Xnnn] [Rmmm]: Duplication mode. The second extruder will duplicate the first with nnn
> //                         millimeters x-offset and an optional differential hotend temperature of
> //                         mmm degrees. E.g., with "M605 S2 X100 R2" the second extruder will duplicate
> //                         the first with a spacing of 100mm in the x direction and 2 degrees hotter.
> //
> //    Note: the X axis should be homed after changing dual x-carriage mode.
> //    M605 S3: Default mode
> //	  M605 S4: Duplication Mirror mode
> //    M605 S5 [Xnnn] [Rmmm]: Duplication Raft mode
> //	  M605 S6: Duplication Raft Mirror mode
> //
> //    Note: the X axis should be homed after changing dual x-carriage mode.

foosel, 2019-03-08:
That's sadly not very specific - as someone without access to a mirror setup these different modes tell me pretty much nothing and I'm not sure how to handle them with regards to filament usage calculation.

JanneMantyharju, 2019-03-09:
After M605 S2, S4, S5 or S6 are issued, from that command onwards second extruder does exactly what the first extruder does. Any other M605 SX command stops second extruder duplicating first extruder extrusion commands.

I'd like to see this feature, but it's of course possible to leave this on hold until the M605 becomes more widely used and standardized.

foosel, 2019-05-28:
Implemented in #3183, already merged to `maintenance`, soon `devel`, to be released with 1.3.12.

foosel, 2019-10-22:
1.3.12 has been released.

closed 2019-10-22 by foosel