OctoPrint/OctoPrint #3076: Can't close notification dialog on raspberry

WolfgangFahl, 2019-03-08:
#### What were you doing?
Trying to use octoprint in a browser directly on a raspberry Touch which also runs the server.

#### What did you expect to happen?
Being able to use the controls
#### What happened instead?
A notification dialog hides the controls. The button to close the notification dialog is not visible and not reachable via scrolling.
#### Did the same happen when running OctoPrint in safe mode?
N.A.

#### Version of OctoPrint
1.3.10

#### Operating System running OctoPrint
raspbian

#### Printer model & used firmware incl. version
N.A.

#### Browser and version of browser, operating system running browser
default raspbian browser
#### Link to octoprint.log
N.A.

#### Link to contents of terminal tab or serial.log
N.A.

#### Link to contents of Javascript console in the browser
N.A.

#### Screenshot(s)/video(s) showing the problem:

[image]

I have read the FAQ.

WolfgangFahl, 2019-03-08:
I worked around the issue by zooming the screen to 25%

foosel, 2019-03-08:
Duplicate of #2959 and already fixed on `maintenance`, fix will be released with 1.3.11 ,)

foosel, 2019-03-08:
As a side note, I'd really not recommend to use the same Raspy that you use for driving your printer (and running OctoPrint) to also run a full fledged desktop environment on and a browser displaying the UI in it.

OctoPrint was designed to do as much as possible on the client to remove load from the server which usually is a low powered single board computer. If you run the UI on the very same device you can overload it and negatively influence your print jobs.

closed 2019-03-08 by foosel

WolfgangFahl, 2019-03-08:
This was not for the print job itself but for the calibration tasks in advance of it. Having a touch screen right beside the printer is really handy for this. Would it be possible to separate just the control part of the ui to a less load demanding version?

foosel, 2019-03-08:
Sure, if someone writes it :) All the needed bits are there, API, plugin hooks, docs, everything.