OctoPrint/OctoPrint #3081: [Request] Show ALL extruder temperatures included in GCode

leofaoro, 2019-03-16:
Hi Gina
Almost had an accident today. I have a printer with 2 extruders, of which I only use one. I dismantled the other, but left the thermistor there hanging (bad idea in retrospect). Today I printed using a Simplify3D profile I hadn't used in a while, and turns out temperatures were enabled in both extruders. As I only had 1 extruder in Octoprint, I never realized both extruders warmed up. The loose thermistor promptly proceeded to melt part of the fan.

My request is therefore to perhaps implement a check: if user is using a printer with only 1 extruder configured, and commands from the GCode are being sent to the other extruder, there should be a warning of some sort. Does that make sense? Perhaps my case is rare (plus I should have removed the thermistor), but it's nice to have an extra safety layer.

thanks for considering
Leo

tedder, 2019-03-16:
Shouldn't your 3d controller (eg Marlin) detect the thermal runaway?

foosel, 2019-03-18:
If I understand your title correctly and you are requesting to have the graph always plot *all* temperatures reported back by the firmware, not just the ones matching the configured extruder count in the printer profile, then this is going to be tricky since the current implementation of the graph makes dynamic reporting tricky if not impossible (as I just noticed while implementing support for chamber temperature reporting). 

What I could do though - and I'm surprised I haven yet done that to be honest - is don't allow sending of set temperature commands for extruders that aren't contained in the printer profile.

leofaoro, 2019-03-18:
> Shouldn't your 3d controller (eg Marlin) detect the thermal runaway?

Hi @tedder yes, the controller detected it, but that was after it had already been up to temp and melted other components. I guess my take is that this should have been detected prior to that.

leofaoro, 2019-03-18:
> If I understand your title correctly and you are requesting to have the graph always plot _all_ temperatures reported back by the firmware, not just the ones matching the configured extruder count [...]

Yes, my request was just one possible solution, up for discussion on how to best handle it.

> What I could do though - and I'm surprised I haven yet done that to be honest - is don't allow sending of set temperature commands for extruders that aren't contained in the printer profile.

That would be great, that would have prevented this incident for sure.

> As a side note though: as @tedder said this should have been handled by your firmware, and the fact that it wasn't means either you don't have thermal runaway protection configured or its malfunctioning. In any case, this needs to be looked into and fixed asap since this is a fire risk in any case, regardless of dangling hotend parts or not.

The thermal runaway was detected and the printer shut down automatically, but it was after parts had been melted. Of course part of it is my fault too for not removing the unused thermistor and leaving it in the carriage. Lesson learned :)

Thanks so much for your (and other's) work on this project. I'm a Patreon supporter and this project helps me almost every day, can't imagine working without it.

Leo

cp2004, 2021-07-16:
OctoPrint does (and has for a while now) block commands to extruders either (a) not configured in the printer profile or (b) reported as invalid, this seems resolved.

closed 2021-07-16 by cp2004