OctoPrint/OctoPrint #3082: "wiki - Setup on a Raspberry Pi running Raspbian" page does not work

cliegh, 2019-03-19:
i want to read "wiki - Setup on a Raspberry Pi running Raspbian" 

but it does not work.

is that my personal problem?

tedder, 2019-03-19:
The link to it doesn't work? The instructions fail?

ntoff, 2019-03-19:
Looks like the wiki still links to discourse.octoprint, for me it auto-redirects to community.octoprint, in case it doesn't auto-redirect for you, try this https://community.octoprint.org/t/setting-up-octoprint-on-a-raspberry-pi-running-raspbian/2337

reloxx13, 2019-03-29:
i replaced all old urls with the new one now

Old:
https://discourse.octoprint.org/
New:
https://community.octoprint.org/

but it redirected correctly for me.

Pls close this if its solved for u @cliegh

closed 2019-04-03 by GitIssueBot