OctoPrint/OctoPrint #3083: Octoprint does not have internet access due to proxy server.

DimchikUA, 2019-03-19:
I connected the octopi to the Wi-Fi network. The network has a proxy server 192.168.100.101:3128, without this setting OctoPi will not connect to the Internet.
I set up a proxy for OctoPi using [this method](https://www.raspberrypi.org/documentation/configuration/use-a-proxy.md). The Internet has appeared and I can make  `apt-get sudo update`, but oktoprint does not have access to the Internet. I can not update or install new plugins. On top of that, the g-codes that I downloaded a minute ago are displayed as loaded a long time ago when the octoprint last had internet access.

#### Version of OctoPrint

OctoPrint version:  1.3.10

#### Operating System running OctoPrint

OctoPi version:  0.16.0

#### Link to octoprint.log

[octoprint.log](https://pastebin.com/raw/b23snmza)

#### Screenshot(s)/video(s) showing the problem:

[image]
[image]
[image]

How  I can set proxy settings for my octoprint?
I have read the FAQ.

foosel, 2019-03-19:
The proxy settings are probably being applied and used already, but the connectivity check isn't working and that also CANNOT be proxied (it requires a direct connection since we are testing actual connectivity into the net here, not whether HTTP requests will work). So just disable it and you should be fine.

> the g-codes that I downloaded a minute ago are displayed as loaded a long time ago when the octoprint last had internet access.

That sounds like a problem with system time. Pis normally use ntp for that on boot, so you'll need to find a different solution for that.

DimchikUA, 2019-03-19:
> So just disable it and you should be fine.

Unfortunately it did not help. Octoprint still does not have access to the Internet. See screenshots and log file.
I assume that oсtoprint works in a virtual environment and for python I have to set separate proxy settings. But I don’t know how to do it. I managed to find  [1](https://github.com/pypa/virtualenv/issues/1156), [2](https://stackoverflow.com/questions/21468550/pip-not-working-behind-firewall), [3](https://serverfault.com/questions/782726/pip-virtualenv-redhat-scl-proxy-not-working), but I couldn’t figure it out.
[image]
[octoprint.log](https://pastebin.com/raw/iLSwVh5b)

foosel, 2019-03-20:
Ah... You have to tell systemd to also apply these environment changes to the `octoprint` unit, otherwise it will be called without that stuff.

SSH into OctoPi. Then run `sudo nano /etc/systemd/system/octoprint.service`. You are now editing the file. Under `[Service]` add three new lines:

```
Environment=http_proxy=http://127.0.0.1:3128
Environment=https_proxy=http://127.0.0.1:3128
Environment=no_proxy=localhost,127.0.0.1
```

Make sure to substitute the proxy settings accordingly. The whole file should now look like this (with the correct proxy settings for your site of course):

```
[Unit]
Description=The snappy web interface for your 3D printer
After=network-online.target
Wants=network-online.target

[Service]
Type=simple
User=pi
Environment=http_proxy=http://127.0.0.1:3128
Environment=https_proxy=http://127.0.0.1:3128
Environment=no_proxy=localhost,127.0.0.1
ExecStart=/home/pi/oprint/bin/octoprint --host 127.0.0.1 --port 5000

[Install]
WantedBy=multi-user.target
```

Save, exit the editor. Then reload the unit: `sudo systemctl daemon-reload`. And finally restart OctoPrint: `sudo service octoprint restart`. It should now be using the proxy for all connections (besides the aforementioned connectivity check, so leave that off).

Note that this only applies to OctoPi 0.16 (and future versions using systemd).

DimchikUA, 2019-03-21:
Thank! It all worked!
The only thing that had to confuse me was that the octoprint.service file did not exist. I had to create it. Next, in the lines I had to write my proxy in two lines
```
Environment = http_proxy = http: //xxx.xxx.xxx.xxx: xxxx
Environment = https_proxy = http: //xxx.xxx.xxx.xxx: xxxx
```

The lines below do not need to be changed.
```
Environment = no_proxy = localhost, 127.0.0.1
ExecStart = / home / pi / oprint / bin / octoprint --host 127.0.0.1 --port 5000
```
Unfortunately, the problem has remained over time. I suppose that the proxy for NTP also needs to be configured somewhere ...

In any case, thanks for the help!

foosel, 2019-03-22:
> The only thing that had to confuse me was that the octoprint.service file did not exist.

🤔 Consider myself very confused now because I checked against an OctoPi 0.16.0 and it was there for me. Then I checked against [the OctoPi sources](https://github.com/guysoft/OctoPi/blob/devel/src/modules/octopi/filesystem/root/etc/init.d/octoprint) and there it still was an old init script. So now I really don't know why my image did have that file, I don't remember creating it myself.

Just creating the file worked for you though? Interesting... So I guess it'll prefer a systemd file if it can find one over the sysv init script. Still, I'll need to adjust the guide I posted on the forum then...

> [ntp issues]

Either you look into an NTP server running locally on your network or you'll take a look at something like [`tlsdate`](https://github.com/ioerror/tlsdate).

DimchikUA, 2019-03-25:
> Just creating the file worked for you though?
Yes, it worked. Now I can read octoprint news, install and update plugins.

foosel, 2019-05-27:
Forgot to close this since it's solved by the now documented systemd unit file.

closed 2019-05-27 by foosel