OctoPrint/OctoPrint #3084: Local Autologin TouchUI

Simur, 2019-03-22:
->

#### What were you doing?

Come and install a Waveshare touchscreen 7" C
You can not connect without a physical keyboard.
"Remember me" case don't work with a PC or the touchscreen.

#### What did you expect?
An automatic connection of the octoprint account.

#### What happened instead?
Arrived on the login page

#### Did the same thing happen when running OctoPrint in safe mode?

Yes.

#### OctoPrint version
OctoPrint 1.3.10 running on OctoPi 0.15.1

#### Browser and browser version, operating system running the browser

Chrome on PC, Chromium on Rpi.

#### Link to octoprint.log

https://gist.github.com/Simur/52fb1bfbf57a6c6053a1f76d789d20e8.js

Thanks you =)) 

[octoprint.log](https://github.com/foosel/OctoPrint/files/2996576/octoprint.log)

I read the FAQ.

foosel, 2019-03-22:
I'm not entirely sure what your problem is. Are you complaining about the forced login page in front of an instance now? You can disable the ForceLogin plugin to get rid of it. Are you saying that automatic login coming from a local network doesn't work? If so I need to know [if and what you configured for that](http://docs.octoprint.org/en/master/configuration/config_yaml.html#access-control) in order look into this - it will not work magically on its own.

"Remember me" on the force login page not working is a known bug (#2971) and will be fixed in 1.3.11.

tldr, please clarify what problem you are reporting here.

Simur, 2019-03-23:
Hi,
Thx for reply =) 

I can't autologin on local network.
My admin account on Octoprint is root. 
You will find config.yaml.

https://gist.github.com/Simur/73406833d20f75275a0323ccf89c5f0d

Thx

Simur, 2019-03-23:
New octoprint.log =
https://gist.github.com/Simur/bfb023546bac31897851f7984882e8b6

Simur, 2019-03-24:
Please, help me :( 
I try all ideas..

Simur, 2019-03-24:
Hello, you can close this post, i disable access control, and it work =) 
thx for your help 💯

foosel, 2019-03-25:
1. Since you shared your full config.yaml including several secret API credentials you really need to reset pushover and co now or anyone who comes across this file can happily abuse your credentials.

2. Your pi is reporting undervoltage. Your power supply is not supplying it with enough power to work properly. Fix that.

3. You provided a truncated log. Never truncate the log.

4. It looks like you are logging in via ipv6. OctoPrint 1.3.10 doesn't yet recognize the V6 `::1` as local address out of the box, you have to add that manually for stuff to work. 1.3.11 will correctly identify `::1` as a local address.

5. Closing this as a requested, but **reset your credentials and fix your power supply**.

closed 2019-03-25 by foosel