OctoPrint/OctoPrint #3087: [RC Feedback] Feedback on the 1.3.11rc1 Release Candidate

foosel, 2019-03-28:
Please provide *general* feedback on your experience with the 1.3.11rc1 Release Candidate here. An "All is working fine" is valuable feedback as well, because it tells me that people in fact are testing the RC and just not finding any problems. Thanks :)

**If you run into any obvious bugs not yet listed below the following line, please open a new ticket and follow ["How to file a bug report"](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report).**

---

### Currently known bugs

n/a

### Unreproduced issues

n/a

arhi, 2019-03-28:
- 3hr print succeeded
- 5hr print failed after successful 2h print time due to failed sensor wire (nothing to do with octoprint)

so for me 1.3.11rc1 so far so good

schnello, 2019-03-28:
Hmm. Ist my first time that i use a RC.
System Ubuntu 18.04.02
[octoprint.log](https://github.com/foosel/OctoPrint/files/3019553/octoprint.log)

Error:
2019-03-28 18:54:14,773 - tornado.access - ERROR - 500 GET / (192.168.1.30) 171.96ms
2019-03-28 18:54:14,946 - octoprint - ERROR - Exception on / [GET]
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python2.7/dist-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python2.7/dist-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/usr/local/lib/python2.7/dist-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python2.7/dist-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/usr/local/lib/python2.7/dist-packages/octoprint/server/views.py", line 412, in index
    response = default_view()
  File "/usr/local/lib/python2.7/dist-packages/octoprint/server/views.py", line 384, in default_view
    return preemptively_cached()
  File "/usr/local/lib/python2.7/dist-packages/octoprint/server/util/flask.py", line 913, in decorated_function
    return f(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/octoprint/server/util/flask.py", line 967, in decorated_function
    return f(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/octoprint/server/util/flask.py", line 700, in decorated_function
    rv = f_with_duration(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/octoprint/server/util/flask.py", line 669, in f_with_duration
    return f(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/octoprint/server/util/flask.py", line 922, in decorated_function
    rv = f(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/octoprint/server/util/flask.py", line 938, in decorated_function
    rv = f(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/octoprint/server/views.py", line 371, in make_default_ui
    r = make_response(render_template("index.jinja2", **render_kwargs))
  File "/usr/local/lib/python2.7/dist-packages/Flask-0.10.1-py2.7.egg/flask/templating.py", line 128, in render_template
    context, ctx.app)
  File "/usr/local/lib/python2.7/dist-packages/Flask-0.10.1-py2.7.egg/flask/templating.py", line 110, in _render
    rv = template.render(context)
  File "/usr/local/lib/python2.7/dist-packages/Jinja2-2.8.1-py2.7.egg/jinja2/environment.py", line 989, in render
    return self.environment.handle_exception(exc_info, True)
  File "/usr/local/lib/python2.7/dist-packages/Jinja2-2.8.1-py2.7.egg/jinja2/environment.py", line 754, in handle_exception
    reraise(exc_type, exc_value, tb)
  File "/home/nas/.local/lib/python2.7/site-packages/octoprint/templates/index.jinja2", line 16, in top-level template code
    {% include 'initscript.jinja2' %}
  File "/home/nas/.local/lib/python2.7/site-packages/octoprint/templates/initscript.jinja2", line 33, in top-level template code
    var SUPPORTED_FILETYPES = {{ supportedFiletypes|tojson }};
  File "/usr/local/lib/python2.7/dist-packages/Flask-0.10.1-py2.7.egg/flask/json.py", line 243, in tojson_filter
    return Markup(htmlsafe_dumps(obj, **kwargs))
  File "/usr/local/lib/python2.7/dist-packages/Flask-0.10.1-py2.7.egg/flask/json.py", line 186, in htmlsafe_dumps
    rv = dumps(obj, **kwargs) \
  File "/usr/local/lib/python2.7/dist-packages/Flask-0.10.1-py2.7.egg/flask/json.py", line 126, in dumps
    rv = _json.dumps(obj, **kwargs)
  File "/usr/lib/python2.7/json/__init__.py", line 251, in dumps
    sort_keys=sort_keys, **kw).encode(obj)
  File "/usr/lib/python2.7/json/encoder.py", line 209, in encode
    chunks = list(chunks)
  File "/usr/lib/python2.7/json/encoder.py", line 442, in _iterencode
    o = _default(o)
  File "/usr/local/lib/python2.7/dist-packages/octoprint/server/util/flask.py", line 1492, in default
    return flask.json.JSONEncoder.default(self, obj)
  File "/usr/local/lib/python2.7/dist-packages/Flask-0.10.1-py2.7.egg/flask/json.py", line 83, in default
    return _json.JSONEncoder.default(self, o)
  File "/usr/lib/python2.7/json/encoder.py", line 184, in default
    raise TypeError(repr(o) + " is not JSON serializable")
TypeError: Undefined is not JSON serializable
2019-03-28 18:54:14,946 - tornado.access - ERROR - 500 GET / (192.168.1.30) 172.21ms
2019-03-28 18:54:17,826 - tornado.access - WARNING - 409 GET /api/printer (192.168.1.30) 4.31ms

Full Log attached

foosel, 2019-03-28:
@schnello would be interesting to know if this problem reproduces in safemode for you

schnello, 2019-03-28:
@foosel 
Unfortunately, no difference

arhi, 2019-03-28:
another ~5h print successful :)

reloxx13, 2019-03-30:
so far no problems.

kazibole, 2019-03-30:
2 prints (4 hours + 16 hours), no problems.

reloxx13, 2019-03-30:
@foosel  pin this issue

nionio6915, 2019-03-31:
Print completes but temp data appears to be 'stuck' at last known value for bed and hotend. 

Bug report filed with logs and data.

schnello, 2019-04-01:
12 h print
2x 6h print and some smaller printer.
All fine (Klipper FW)

foosel, 2019-04-01:
> @foosel pin this issue

I forgot that this was now possible, thanks for the reminder. Done.

> Print completes but temp data appears to be 'stuck' at last known value for bed and hotend.
> 
> Bug report filed with logs and data.

Err... where? I haven't gotten a ticket.

foosel, 2019-04-02:
@nionio6915 I'm still waiting for the bug report - you never filed it here :confused:

gege2b, 2019-04-03:
I did some prints and as long as I can tell, no problem spotted

foosel, 2019-04-04:
1.3.11rc2 was just released, ticket is at https://github.com/foosel/OctoPrint/issues/3103. Closing this.

closed 2019-04-04 by foosel