OctoPrint/OctoPrint #3093: [1.3.1.1rc1] Webcam Windows not shown anymore

schnello, 2019-03-29:
System: Ubuntu 18.04.02
FW: Klipper

I started from the scratch (fresh install of stable version and update to RC)
Webcam is shown in the config page. Test page is OK

[image]

But in the contrall tab there is no webcam windows anymore. Also testet with different browser (also incognito mode)

[image]

[logfile.txt](https://github.com/foosel/OctoPrint/files/3022075/logfile.txt)

[image]

foosel, 2019-03-29:
The settings screen visible behind your webcam test looks outdated. There should be a checkbox at the very top, above the stream URL:

[image]

That checkbox enables/disables webcam support. If your browser for some reason is still using an old cached version of the UI, that would explain a lot of things, including the JS error your other screenshot shows.

Can you do a hard reload?

foosel, 2019-03-29:
Also please explain how things are set up. You are saying you did an install from scratch, there might be some issues with a reverse proxy in the middle (if there is one) that could also interfere with caching.

Based on your `octoprint.log` I'm a bit doubtful that your browser is even hitting the server. So please also provide a screenshot of the contents of the Network tab in the browser developer console.

schnello, 2019-03-29:
A lot of work behind me I reinstall all. pip, pyhton ... and i delete all pyhton packages in /usr/local/lib...
Fresh install of Octoprint and it was sucessfully:
Webcam works.
[image]

I guess i know also the root cause. But this is a little bit difficult to describe it in englisch. But it was on the server and not in Octoprint.

closed 2019-03-29 by schnello