OctoPrint/OctoPrint #3097: undeclared sentry-sdk dependency in 1.3.11rc1

salfter, 2019-03-29:
#### What were you doing?

1. updated my Gentoo ebuild to pull 1.3.11rc1
2. added/removed dependencies according to what's in setup.py
3. ```emerge -1 octoprint```
4. ```octoprint -h```

#### What did you expect to happen?

see the usual help output from ```octoprint -h```

#### What happened instead?

```
Traceback (most recent call last):
  File "/usr/lib/python-exec/python2.7/octoprint", line 6, in <module>
    from pkg_resources import load_entry_point
  File "/usr/lib64/python2.7/site-packages/pkg_resources/__init__.py", line 3126, in <module>
    @_call_aside
  File "/usr/lib64/python2.7/site-packages/pkg_resources/__init__.py", line 3110, in _call_aside
    f(*args, **kwargs)
  File "/usr/lib64/python2.7/site-packages/pkg_resources/__init__.py", line 3139, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/usr/lib64/python2.7/site-packages/pkg_resources/__init__.py", line 583, in _build_master
    return cls._build_from_requirements(__requires__)
  File "/usr/lib64/python2.7/site-packages/pkg_resources/__init__.py", line 596, in _build_from_requirements
    dists = ws.resolve(reqs, Environment())
  File "/usr/lib64/python2.7/site-packages/pkg_resources/__init__.py", line 784, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'sentry-sdk==0.7.7' distribution was not found and is required by OctoPrint
```

#### Did the same happen when running OctoPrint in safe mode?

n/a...OctoPrint won't start

#### Version of OctoPrint

1.3.11rc1

#### Operating System running OctoPrint

Gentoo Linux

salfter, 2019-03-29:
One more undeclared dependency: typing.  Also, the futures dependency has been updated to a newer version, and that's undeclared.

foosel, 2019-03-29:
All of them are declared in https://github.com/foosel/OctoPrint/blob/rc/maintenance/setup.py, so something in your build must be off. Stuff also gets installed during a regular pip install as expected.

salfter, 2019-03-29:
Hmm...so I was looking in the wrong place (was working off the top-level setup.py).  Gentoo doesn't use pip to manage Python packages.

Everything's up-to-date now, with new Python ebuilds created as needed.  (Some of the newer versions are in Portage, too, instead of needing me to provide old versions in my overlay.)  Having sussed out all the updates on my desktop, I'm now letting the Raspberry Pi on my printer update to the new version.

closed 2019-03-29 by salfter