OctoPrint/OctoPrint #3099: Invalid character (null byte) within the GCode

ForsakenNGS, 2019-03-30:
#### What were you doing?

1. Load and slice my model in Slic3r PE (also had that bug before with Cura 3.6, see here: https://github.com/Ultimaker/Cura/issues/5450 )
2. Send and print the gcode to my Octoprint instance via the octoprint integration
3. Had it running for over 11 hours after which it got stuck repeating the following over and over:

Send: N1089770 G1 X97.967\x00Y104.550 E0.00915*124
Recv: Error:No Checksum with line number, Last Line: 1089769
Recv: Resend: 1089770
Recv: ok
Send: N1089771 G1 X98.141 Y104.550 E0.00151*86
Recv: Error:Line Number is not Last Line Number+1, Last Line: 1089769
Recv: Resend: 1089770
Recv: ok
Send: N1089772 G1 X98.141 Y103.392 E0.01007*89
Recv: Error:Line Number is not Last Line Number+1, Last Line: 1089769
Recv: Resend: 1089770
Recv: ok
Send: N1089773 M400*36
Recv: Error:Line Number is not Last Line Number+1, Last Line: 1089769
Recv: Resend: 1089770
Recv: ok

#### What did you expect to happen?
I expected the print to finish normally.

#### What happened instead?
The uploaded gcode file contained a null byte ("\x00") where a space should be which prevented the printer from progressing past that line. It will leave the hotend and heated bed on an just stay where it stopped forever until stopped.
Luckily I was around this time when it happened and managed to manually stop it, truncate the gcode manually and resume the print that way. Otherwise I would have wasted over 11 hours of printing.

#### Did the same happen when running OctoPrint in safe mode?
Since this issue only happens randomly from time to time I can't simply switch to safe mode and reupload the file. After my print is finished I can do a few test uploads and then check the uploaded gcode files for errors (null bytes)
What I can say for sure is it is not dependend on the hardware since I encountered this issue on two different installations of octoprint on different boards (one pi and one beagleboard) and sdcards (one 64gb sandisk ultra and one 8gb unbranded that came with my ender 3).

#### Version of OctoPrint

OctoPrint: 1.3.10 

#### Operating System running OctoPrint

Debian 9.8 running on a beagleboard currently. With the follwing Kernel:
Linux beagleboard 4.18.11-armv7-x11 #1 SMP PREEMPT Mon Oct 1 16:21:38 UTC 2018 armv7l GNU/Linux

Also experienced this issue on a raspberry pi b rev 2 running the (at that point - like 2 or 3 weeks ago) latest OctoPi image. Can't get any more details on that since I wiped the machine.

#### Printer model & used firmware incl. version
Creality Ender-3 with the following modifications:
E3D V6 Hotend replacement, Filament runout sensor soldered to pin 29, Manually deployed Z Probe in parallel to a regular Z-Endstop and the current 2.0.x Mariln firmware (see below)

Firmare is stock Marlin 2.0.x at the following commit: https://github.com/MarlinFirmware/Marlin/commit/09eb30dc05a727e3816ab5a407f02ada18700e23

#### Browser and version of browser, operating system running browser
Chromium Version 73.0.3683.86 (Official Build) Arch Linux (64-bit)

#### Link to octoprint.log
https://gist.github.com/ForsakenNGS/0af3d952010a51ddcd9d6621415c8c6a

#### Link to contents of terminal tab or serial.log
2019-03-30 15:45:25,736 - serial.log is currently not enabled, you can enable it via Settings > Serial Connection > Log communication to serial.log

#### Link to contents of Javascript console in the browser
Not applicable for this issue.

#### Screenshot(s)/video(s) showing the problem:
Not applicable for this issue.

#### Other information:
Attached is a zip file including
 - The corrupted gcode file from octoprint
 - The full octoprint.log for download
[OctoprintNullByte.zip](https://github.com/foosel/OctoPrint/files/3025709/OctoprintNullByte.zip)

#### Further steps for figuring the problem out:
 - I'm (of course) willing to provide any further feedback required to solve this issue
 - I will try (as soon as the resumed print is finished) to upload a whole bunch of big gcode files in the same way I did before and inspect the uploaded gcode files for the issue. (And report the results here) That way I will most likely be able to reproduce the issue without wasting huge amounts of time and filament.

I have read the FAQ.

ForsakenNGS, 2019-03-31:
### Update: Reproducing the Issue
I have made some experiments to reproduce the issue. For that I sliced a model that I want to print in Slic3r PE and sent it to my octoprint instance. First I did that with 50% scale (resulting in a ~32MB gcode file) which was successful about 15-20 time in a row. After that I changed the scale to 100%  (resulting in a ~150MB gcode file) and continued testing with that. After another 8-9 uploads I encountered the described issue again. In the octoprint.log I can see nothing except the heartbeat. (See attached file)
[octoprint_upload.log](https://github.com/foosel/OctoPrint/files/3026169/octoprint_upload.log)

Now I knew that method made it possible for me to reproduce the problem I tried to figure out if the filesize is a factor and how frequently it occurs. So I continued uploading the 100% scale file and immediately encountered the problem again on the next upload. (On a different line, so yes the file definitly was changed)
And I continued my testing:
 - 2nd upload after first failure: Corrupted!
 - 3rd upload ...: Corrupted!
 - 4th upload ...: Okay!
 - 5th upload ...: Corrupted!
 - 6th upload ...: Okay!
 - 7th upload ...: Corrupted!
 - 8th upload ...: Corrupted!

So after the first failure reuploading the file over and over fail >50% of the time. The corrupted lines (which was always **exactly one** per file) were:
"G1 X104.171 Y132.319\x00E0.00333"
"G1 X54.293\x00Y139.374 E0.00324"
"G1 X54.293\x00Y139.374 E0.00324"
"G1 X111.057 Y112.918\x00E0.00159"
"G1 X111.057 Y112.918\x00E0.00159"
"G1 X173.569 Y144.825\x00E0.00319"
"G1 X109.703 Y114.361\x00E0.00439"
"G1 X87.609\x00Y88.609 E0.00368"
What stands out here is it looks like the space after the G1 never seems to be affected.

Next I tested uploading the gcode with a different filename each time, so it will create a new file each time and not overwrite the existing one:
 - 1st upload: Corrputed!
 - 2nd upload: Okay!
 - 3rd upload: Corrupted!
 - 4th upload: Okay!
 - 5th upload: Okay!
So that doesn't seem to make any difference.

The octoprint.log file still only contained further heartbeats. I also checked the systemlog which did not contain anything (like errors writing to the sdcard).

foosel, 2019-03-31:
At this point I'd take OctoPrint out of the equation and check if it isn't Slic3r generating corrupt files first. Otherwise we might be hunting down the rabbit hole here.

ForsakenNGS, 2019-03-31:
> At this point I'd take OctoPrint out of the equation and check if it isn't Slic3r generating corrupt files first. Otherwise we might be hunting down the rabbit hole here.

No it is not. Like described in the issue this happend on Cura first (see the linked issue in the Cura Github). I just used slic3r for testing since it has the option to send the gcode to the printer without starting the print right away. The issue definitly related to octprint.

foosel, 2019-03-31:
Ah, I somehow didn't see that. I read through the ticket and only saw repeated mentions of uploading directly from Slic3r PE.

[OctoPrint writes the data it receives from the upload to disk as it receives it from the multipart upload](https://github.com/foosel/OctoPrint/blob/master/src/octoprint/server/util/tornado.py#L380). Unless you have any kind of post-processing plugins installed it will not touch the file contents in any way but [simply move the file to its destination after the upload has completed](https://github.com/foosel/OctoPrint/blob/master/src/octoprint/filemanager/util.py#L58-L64). So I'm not entirely sure how a 0-byte could end up in there due to code. It would have to be in the received data already.

ForsakenNGS, 2019-03-31:
> Are you able to sniff the data and see if it goes over the line like that?

I will check that with wireshark and report back.
I would have assumed a hardware issue, but since I completely swapped the hardware and also despite it only corrupts spaces and never the first one this seems quite unlikely to me.

ForsakenNGS, 2019-03-31:
I sent countless (we're talking 100+) gcode files this evening while having wireshark running, without encountering the issue even once. I will retry tomorrow and update the issue accordingly.
I did neither restart octoprint nor my computer since I was able to reilably reproduce the issue before, so thats not the difference here. Also I tried 50+ big files (~120MB) and another 50+ smaller files (~25MB).

foosel, 2019-04-01:
That's really really weird O_O

ForsakenNGS, 2019-04-01:
> That's really really weird O_O

So I wrote a script for uploading via curl and then checking the gcode and ran it the whole evening (~3200 uploads) with again no issues at all. I'm starting to question if my pc (that I am slicing with) maybe has some weird software (linux side) or hardware (memory maybe?) glitches. Still it seems odd to me that the corruption isnt more random (only affecting spaces, but never the first one) and also doesn't affect any other programs. I'm really puzzled about this.
I will definitly run that script a few more times at different times with reboots in between, under cpu/memory load etc. just to check all that could be responsible.

ForsakenNGS, 2019-04-03:
So, another update. It happened again during a regular print last night, but this time things are a bit different. This morning I was greeted with a (still heating) printer stuck mid-print with the messages:
Send: N413395 G1\x00X91.567 Y112.231 E0.00236*66
Recv: Error:Line Number is not Last Line Number+1, Last Line: 413394
Recv: Resend: 413395
Recv: ok
Send: N413396 M108*16
Recv: Error:No Checksum with line number, Last Line: 413394
Recv: Resend: 413395
Recv: ok

Two things different here:
 - The "\x00" for a change is where the first space should be (which so far never happened on the many many times I had this issue occur)
 - The corruption this time is NOT present on the sdcard that octoprint stores the gcode on.

Honestly that denies any ideas I had in mind that could be causing this. To prevent this happening to me again I always checked the uploaded gcode for the "\x00" character before hitting print (like this time as well), but even that does not seem to help.
I'm honestly just out of ideas, what could be causing this. 
Two broken boards? Two broken sdcards? Two different issues resulting in almost identical symptoms? Cosmic radiation? A curse maybe?
Seriously, I'd be glad about any suggestions about what to check. I just want a reliable "print" button on my slicer :/

foosel, 2019-04-05:
I'm sadly a bit out of suggestions at the moment. If the file on the Pi's SD card is fine, the only explanation I have is something going really wrong while reading from it. And I'm really just reading from it.

The only thing that I can look into here from OctoPrint's side of things (for 1.3.12 though, 1.3.11 is already in feature freeze mode) is try to detect repeated resends of the same line over and over again, and if that is detected cancel the print and shut off everything. That's on the TODO list anyhow and I'll give it a higher priority. 

What brand are your SD cards? If they are the same (and maybe even bought at the same time) I'd go for a different brand and see if that solves things. But other than that I'm at a loss.

ForsakenNGS, 2019-04-05:
> I'm sadly a bit out of suggestions at the moment. If the file on the Pi's SD card is fine, the only explanation I have is something going really wrong while reading from it. And I'm really just [...]

That would already be a lot better. Having the printer standing still while keeping bed and hotend heated for hours is really the most concerning part about that.

> What brand are your SD cards? If they are the same (and maybe even bought at the same time) I'd go for a different brand and see if that solves things. But other than that I'm at a loss.

It's both Sandisk cards. The first one was a 64GB one, the current one is 8GB. I will give it another shot with a different 8GB card which came with my Ender-3. I will also try to use a new RaspberryPi board  from work next week, after trying out the third sdcard with the current hardware for a bit.

foosel, 2019-04-05:
In case that an additional data point helps: I think I've never used Sandisk in my Pis. I pretty much exclusively use Transcend Class 10 cards.

ForsakenNGS, 2019-06-07:
It's been a while, I swaped the hardware with a relatively new RaspberryPi like mentioned before and have been running OctoPi on it for about 3 weeks. At least 30 prints in I did not see that issue again. So I guess either both the other boards have issues or there's some hardware incompatibilities with the older boards:
- Raspberry Pi B Rev 2
- BeagleBoard-xM
Getting a new board is not the solution I would have wished for, but at this point it's not worth the time for me to investigate it further. *closed*

closed 2019-06-07 by ForsakenNGS