OctoPrint/OctoPrint #3100: "Could not initialize environment detector: " Critical error on service start

Technoguyfication, 2019-03-31:
#### What were you doing?

I updated OctoPrint using the update feature in the browser. It was working fine beforehand.

#### What did you expect to happen?

I expected the server to start back up again afterwards.

#### What happened instead?

I got the "server not running" message in my browser. The octoprint service is not running.

#### Did the same happen when running OctoPrint in safe mode?

Same issue in safe mode

#### Version of OctoPrint

1.13.10

#### Operating System running OctoPrint

OctoPi Raspi 3B+

#### Printer model & used firmware incl. version

#### Browser and version of browser, operating system running browser

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

Non-safe mode log: https://pastebin.com/gqgJ7Y5y
Safe mode log: https://pastebin.com/MiKh8154

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

foosel, 2019-04-01:
I'm honestly at a loss here since the exception that is apparently the cause of this doesn't even have a message attached (otherwise there would be text after the `:`) and the initialization code that must have failed here is about as simple as it gets.

Could it be that something in your installation has become corrupted? Were there any errors during your update (please provide `~/.octoprint/logs/plugin_softwareupdate_console.log`? Have you ever just pulled power from your Pi without shutting it down first? Do you feel comfortable pulling from a development branch? If so I could push a branch with some more logging. Finally, the problem might also just vanish/fix itself by a re-install (`~/oprint/bin/pip install --force-reinstall https://get.octoprint.org/latest`).

For the record, in the last 24h there were ~24k instances running 1.3.10 that reported back via the usage tracking plugin, so whatever you ran into here, it looks rare and possibly caused by your environment.

Technoguyfication, 2019-04-01:
This actually happened the last time I updated too (couple months ago), I decided to download the latest image and reinstall which fixed it both times. It’s probably something screwy with my SD card, since I’m using the unbranded card that came with my printer. Just wanted to check if it was a known issue. Thanks!

closed 2019-04-01 by Technoguyfication

foosel, 2019-04-01:
I've added a line to the startup routine anyhow that at least should shed some more light on this if it happens again. See the commit above. Will be part of 1.3.11 (and 1.3.11rc2)