OctoPrint/OctoPrint #3101: [Request]  Add `tags` as optional element to the tuple returned from  `protocol_scripts_hook`

kennethjiang, 2019-04-02:
Currently the GCodes returned as tuples from `protocol_scripts_hook` hooks are not tagged with the plugin's identifier.

This creates a problem for plugins that use both `octoprint.comm.protocol.gcode.queuing` and `protocol_scripts_hook` hooks, because there is no easy way to identify what gcodes were inserted/modified by the plugin.

The best solution in my opinion is to add `tags` as optional element to the tuple returned from  `protocol_scripts_hook`, so that the plugin has a way to flag the GCodes and treat them differently in `octoprint.comm.protocol.gcode.queuing` hook.

An alternative solution is to add a tag such as  `plugin:plugin_identifer` automatically to the GCodes returned from plugin's `protocol_scripts_hook` hook.

*Additional context: [my plugin](https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective) would insert GCodes to retract and z-lift after pausing a print. This involves saving the current absolute/relative mode and switching back to it. This is impossible to do reliably without knowing which mode-switch GCodes (G90/G91) were from my own plugin, which were from Gcode file or other plugins.*

foosel, 2019-05-08:
Implemented by the above commit. On `maintenance`, soon `devel`, will be part of 1.3.12.

foosel, 2019-10-22:
1.3.12 has been released.

closed 2019-10-22 by foosel