OctoPrint/OctoPrint #3102: [Request] timelapse tar/zip support

rfinnie, 2019-04-04:
As an alternative to ffmpeg movies from timelapses, it would be nice to have an option to take the snapshot images and simply put them into a tarball or zip file. Thanks!

ManuelMcLure, 2019-04-04:
You can implement this by changing the "Path to ffmpeg" entry in the settings with the path to a script that does whatever you want with the timelapse stuff. I have myself changed this to a shell script that calls a python script on a different server on my network to render the timelapse on that server instead of on my RPi, but there's no limit to what you could do with this.
For example, my script looks like:
```
#!/bin/bash
echo "$@" >/tmp/render_opts.txt
fixedargs=''
for var in "$@"; do
        printf -v __ '%q' "${var}"
        fixedargs="${fixedargs} ${__}"
done
ssh manuel@legend "/var/space/OctoPi/renderTimelapse.py ${fixedargs}"
if [ $? -eq 0 ] ; then
        exit 0
else
        exit 1
fi
```

rfinnie, 2019-04-05:
Thanks.  As the generated command line is very ffmpeg-specific, I ended up doing this to both tar the jpgs up and render them as normal:

```
#!/bin/sh

cd /home/pi/.octoprint/timelapse/tmp
BASE="$(ls -1t *.jpg | head -n 1 | perl -pe 's/-\d+\.jpg//g')"
tar zcvf "../${BASE}.tar.gz" "${BASE}"-*.jpg

exec /usr/bin/avconv "$@"
```

Ideally, I think a good suggestion for the future would be an option to "render" to zip (a basic configuration for the average user), and an option to "render" to "general-purpose" script, with the following arguments passed to the script:

* Base filename
* Path to timelapse directory
* Path to tmp directory with the images

After the script finishes, OctoPrint could scan ${TIMELAPSE_DIR}/${BASE}.* and add those to the UI file list.

Oninaig, 2021-11-15:
> You can implement this by changing the "Path to ffmpeg" entry in the settings with the path to a script that does whatever you want with the timelapse stuff. I have myself changed this to a shell [...]

I know this is an old issue but can you share that script you are using to render the timelapse remotely?

ManuelMcLure, 2021-11-16:
```
#!/usr/bin/python
# renderTimelapse.py - render an OctoPrint timelapse
# massages the input parameters to take into account changes in layout
import sys
import re
import subprocess

newarglist = ['/usr/bin/ffmpeg'];

for arg in sys.argv[1:]:
    if arg == 'mpeg2video':
        arg = 'h264'
    elif arg == 'vob':
        arg = 'mp4'
    else:
        arg = re.sub('movie=\S*',
            'movie=/var/space/OctoPi/watermark.png', arg)
        arg = re.sub('\.mpg$', '.mp4', arg);

    newarglist.append(arg)

sys.exit(subprocess.call(newarglist))
```