OctoPrint/OctoPrint #3109: 'ascii' codec can't encode character u'\xbb' in position 34

p-r-o-m-e-t-h-e-u-s, 2019-04-05:
Trying to add a new system command.  The config.yml looks like:

```
system:
  actions:
  - action: test
    command: sudo ls
    name: Test
```

When I try to execute the action from the UI, I get the following error:

> Command for custom:test failed: 'ascii' codec can't encode character u'\xbb' in position 34: ordinal not in range(128)

! sudo ls - is just an useless example, not other command seems to work (same error)

foosel, 2019-04-05:
If you don't share any logs (and all the other stuff you ignored) with me, this cannot be looked into.

p-r-o-m-e-t-h-e-u-s, 2019-04-06:
Unfortunately, even set to DEBUG level - the octoprint.log doesn't provide any helpful informations. 
```
2019-04-06 09:14:28,443 - octoprint.server.util.flask - DEBUG - App sessions after cleanup: {}
2019-04-06 09:14:28,463 - octoprint.server.util.flask - DEBUG - App sessions after cleanup: {}
2019-04-06 09:14:28,469 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.178.20) 12.56ms
2019-04-06 09:14:34,359 - octoprint.server.api.system - INFO - Performing command for custom:light
2019-04-06 09:14:34,444 - octoprint.server.api.system - WARNING - Command for custom:light failed: 'ascii' codec can't encode character u'\xbb' in position 34: ordinal not in range(128)
2019-04-06 09:14:34,460 - tornado.access - ERROR - 500 POST /api/system/commands/custom/light (2a02:778:11f:e101:7406:fbc7:4885:1846) 120.82ms
2019-04-06 09:14:41,008 - octoprint.server.util.flask - DEBUG - App sessions after cleanup: {}
2019-04-06 09:14:41,018 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.178.20) 23.06ms
2019-04-06 09:14:41,037 - octoprint.server.util.flask - DEBUG - App sessions after cleanup: {}
2019-04-06 09:14:41,066 - octoprint.server.util.flask - DEBUG - App sessions after cleanup: {}
2
```

I traced it down to the 'sudo' as the problem:

This works fine:
```
system:
  actions:
  - action: test
    command: ls
    name: Test
```

This not:
```
system:
  actions:
  - action: test
    command: sudo ls
    name: Test
```

This may be easily reproducible. Anything other helpful I can provide?

tedder, 2019-04-06:
It's much more information, actually, as it shows it's coming via `octoprint.server.api.system` and is followed by a 500 POST. There's also a weird 409 error before that. Before I post the rest of what I've seen please post the startup logs in octoprint. For instance, mine look like this:
```
2019-04-06 05:13:36,709 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2019-04-06 05:13:36,727 - octoprint.server - INFO - --- Log roll over detected ---------------------------------------------------
2019-04-06 05:13:36,728 - octoprint.server - INFO - OctoPrint 1.3.10
2019-04-06 05:13:36,734 - octoprint.plugin.core - INFO - 21 plugin(s) registered with the system:
|  Action Command Prompt Support (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/action_command_prompt
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/announcements
...
2019-04-06 05:13:36,765 - octoprint.environment - INFO - Detected environment is Python 2.7.13 under Linux (linux2). Details:
|  hardware:
|    cores: 4
...
|    virtualenv: /home/pi/oprint
2019-04-06 05:13:36,765 - octoprint.server - INFO - ------------------------------------------------------------------------------

```

I've elided mine because they aren't relevant to the ticket, but please include ALL of yours from the first set of `--------------`s through to the last set.

p-r-o-m-e-t-h-e-u-s, 2019-04-06:
Full log provided here [octoprint.log](https://github.com/foosel/OctoPrint/files/3050966/octoprint.log)

tedder, 2019-04-06:
There are a couple of problems, part of it is just that Octoprint isn't returning good error information.

Problems aside from that:
- you can't really sudo as a non-human on most unix systems, including octopi, unless you've made changes to the config.
- sudo or sispmctl is returning `»` it for some reason

Can you show what it looks like when you run that `sudo sispmctl` command in the terminal? Specifically, this verbatim command:
```
sudo /usr/local/bin/sispmctl -d 0 -t 2
```

@foosel, I think the codec complaint is not the actual error, [I think it's just coming from `str(e)`](https://github.com/foosel/OctoPrint/blob/1.3.10/src/octoprint/server/api/system.py#L131).

p-r-o-m-e-t-h-e-u-s, 2019-04-06:
I  guess you are right... The service is running as pi user and the sudo command asks for a password. I should add the privilege for the command to /etc/sudoers.d to avoid the interactive terminal.

The error `ascii' codec can't encode character u'\xbb' in position 34` is not a very descriptive error message for this issue. A meaningful error message could help people to understand the problem.

OutsourcedGuru, 2019-04-07:
On the offending computer, you might try to run this:

```
LC_ALL=C find . -name '*[! -~]*'
```

In theory, this will report all filenames within the current working directory which contains non-ASCII characters in them.

foosel, 2019-04-08:
> The error `ascii' codec can't encode character u'\xbb' in position 34` is not a very descriptive error message for this issue.

That's because it's not an error message for this issue but rather an error generated while trying to provide a meaningful message for this issue due to an unforeseen encoding problem.

foosel, 2019-04-08:
Fixed by the above commit, pushed to `maintenance`, soon `devel`, to be rolled out with 1.3.**12**

commit 43e2f795 (2019-04-08): 🐛 Fix a unicode issue

Fixes #3109
closed 2019-10-22 by foosel