OctoPrint/OctoPrint #3110: M118 not received

p-r-o-m-e-t-h-e-u-s, 2019-04-05:
I've the following addition to Marlin 2.0 / CUSTOM_USER_MENUS:

 ```
  #define USER_DESC_1 "Light on/off"
  #define USER_GCODE_1 "M118 A1 action:light"
```

When execute the menu from my LCD Display, the M118 does not seems to be received (no entry in the serial log or web terminal).

I would expect to receive the command and some defined "Action Command Definition" is been triggered. But the action command is never executed.

```
actioncommands:
    command_definitions:
    - action: reset
      command: sudo shutdown now -r
      enabled: true
      type: system
    - action: light
      command: sudo sispmctl -d 0 -t 2
      enabled: true
      type: system
```

foosel, 2019-04-05:
If you don't share any logs (and all the other stuff you ignored) with me, this cannot be looked into.

Your reproduction steps are also not helping me to even understand your problem. You have configured something in Marlin that should trigger something to be echo'd to the host but you don't see anything being echo'd? That isn't an OctoPrint issue, that's something you need to look at your firmware for.

And I have no idea what that yaml snippet is supposed to do, where you put it or what made you think that would work. OctoPrint doesn't allow configuring system commands for action commands by default. If you are using a separate plugin for that, you should rather ask the plugin author how to properly configure it. 

Finally your action command syntax is wrong, it isn't `action:foobar` but `//action:foobar`.

p-r-o-m-e-t-h-e-u-s, 2019-04-06:
I digged a little bit deeper and found out, that Marlin does not send the `M118` from other hosts to octoprint. So this seems to be a Marlin issue.

Sadly `M118` only works if the command was issued be octoprint.

Btw, as `M118 //action:foobar` also works for the action command plugin, `M118 A1 action:foobar` is the standard provided by Marlin (works also with action command plugin).

I will close the issue and try to fix it in the firmware.

closed 2019-04-06 by p-r-o-m-e-t-h-e-u-s