OctoPrint/OctoPrint #3112: PSU Control no longer works in 1.3.11rc2

burgedm, 2019-04-07:
PSU Control was working perfectly fine with 1.3.10 but when I upgraded to 1.3.11rc1 and then rc2 it no longer automatically turns off my printer.

The log file has an entry "Idle timeout reached after 5 minute(s). Turning heaters off prior to shutting off PSU" but it does not actually turn off the printer.
 
Nothing else is logged in the octopi logs or in the OS logs.

I'll reiterate that things were working perfectly fine until the upgrade - that's all that has changed.

tedder, 2019-04-07:
Please share your octoprint.log, from startup (plugin loading) through where you trigger the PSU control. You might need to increase your log level if you believe there's truly nothing in there.

burgedm, 2019-04-08:
Sure, once I get home.

I had it at debug level yesterday and there is no sign the octopi attempts to execute the command to power off the power supply.

This electronic message, as well as any transmitted files included in the electronic message, may contain privileged or confidential information and is intended solely for the use of the individual(s) or entity to which it is addressed. If you have received this electronic message in error please notify the sender immediately and delete the electronic message. Any unauthorized copying, disclosure or distribution of the electronic message is strictly forbidden. NAV CANADA accepts no liability for any damage caused by any virus and/or other malicious code transmitted by this electronic communication.

Le présent message électronique et tout fichier qui peut y être joint peuvent contenir des renseignements privilégiés ou confidentiels destinés à l’usage exclusif des personnes ou des organismes à qui ils s’adressent. Si vous avez reçu ce message électronique par erreur, veuillez en informer l’expéditeur immédiatement et supprimez le. Toute reproduction, divulgation ou distribution du présent message électronique est strictement interdite. NAV CANADA n’assume aucune responsabilité en cas de dommage causé par tout virus ou autre programme malveillant transmis par ce message électronique.

foosel, 2019-04-08:
I need all the data requested in the ticket template, otherwise I can't look into this. Also pinging @kantlivelong who's the author of the affected plugin.

burgedm, 2019-04-08:
Here's the log file, at 2019-04-09 00:45:50,849 the PSU is commanded ON, it's currently configured to turn OFF after two minutes of inactivity.

[octoprint.log](https://github.com/foosel/OctoPrint/files/3056619/octoprint.log)

tedder, 2019-04-09:
So PSU Control is turning it on properly. That works- as confirmed by the serial connection 5 seconds later. And after a few seconds we can see the polling:

```
2019-04-09 00:46:11,599 - octoprint.plugins.psucontrol - DEBUG - Sensing system command executed. PID=22587, Command=/home/pi/.octoprint/scripts/power-toggle.sh
2019-04-09 00:46:11,701 - octoprint.plugins.psucontrol - DEBUG - Sensing system command returned: 0
2019-04-09 00:46:11,702 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2019-04-09 00:46:16,738 - octoprint.plugins.psucontrol - DEBUG - Sensing system command executed. PID=22591, Command=/home/pi/.octoprint/scripts/power-toggle.sh
```

I think it goes to the plugin author at this point.

kantlivelong, 2019-04-09:
Hmm, will have a look. Trying to get my test env up but having issues.

foosel, 2019-04-09:
@kantlivelong I think it might be [this sentry issue](https://sentry.io/share/issue/58b96736121848c6a6e1a1cd9d04f676/). What I don't understand is why that didn't happen earlier because the target could always be `None` as far as I know.

foosel, 2019-04-09:
Dug a bit deeper and you have the same issue with printers that do not report a bed temperature. `target` (and `actual`) can and will stay `None` in these cases. 

A simple `None` check of both `actual` and `target` here would make sense and solve it. 

I'm a bit reluctant to modify OctoPrint here to default to `0` instead of `None` because a) that will change OctoPrint's behaviour compared to prior versions and b) it will no longer allow to distinguish between "value reported by firmware" and "no value reported by firmware".

kantlivelong, 2019-04-10:
Thanks @foosel !

Tested out good here. Merged and released 0.1.8.

foosel, 2019-04-10:
Perfect, closing this then.

burgedm, 2019-04-10:
Uh guys ...

The firmware is reporting the temperatures just fine.
It displays on the "Temperature Pane" gives me a proper graph, it displays in the NavBar plugin, it displays on the video stream from my camera when I'm in full screen and as I reported at the beginning ... it works fine with the previous version of Octopi (1.3.10).

foosel, 2019-04-10:
1.3.11 introduces parsing of chamber temperatures. The temperature data handed from OctoPrint to plugins thus contains an entry for a `chamber` heater. If your firmware doesn't report chamber data said data will be `None` though. And that is what the PSU Control's heater routine could not handle, neither for bed nor for chamber entries. 

So unless you are telling me that you have a heated chamber configured in your firmware and thus data for `T`, `B` and `C` in your `M105` outputs - this is the bug.

Of course, if you had fully filled out a ticket template as instructed in several places and even provided with during creation of a ticket, you'd also have shared a `serial.log` and we could have more easily verified this. As things are now we have to rely on reading code and running assumptions based on that, and your report combined with what's in the code and what's changed in 1.3.11 heavily points at the missing handling of `None` entries in the temperature data in the PSU Control plugin's auto shutdown code to be the culprit.

Please update the plugin and test. If it's still broken we can reevaluate this.

burgedm, 2019-04-10:
Now you're just being an ass so ... I'm done.

kantlivelong, 2019-04-10:
Perfect! If there's an issue be sure to open it up in the appropriate repo and fill out all required info.

closed 2019-04-10 by kantlivelong