OctoPrint/OctoPrint #3115: [1.3.11rc2] Serial terminal has extra line breaks

kazibole, 2019-04-09:
#### What were you doing?
Started a print from the printer's SD card via the printer's LCD panel.

#### What did you expect to happen?
Terminal shows communication without extra line breaks.

#### What happened instead?
Terminal shows communication with extra line breaks, random blank "Recv: " lines.

#### Did the same happen when running OctoPrint in safe mode?
Yes.

#### Version of OctoPrint
Occurs in 1.3.11rc2
Did not occur in 1.3.11rc1

#### Operating System running OctoPrint
OctoPi 0.15.1
OctoPi 0.16.0

#### Printer model & used firmware incl. version
Prusa MK3 firmware 3.7.0

#### Browser and version of browser, operating system running browser
Chrome 73.0.3683.86 (64-bit)
Windows 10 1809

#### Link to octoprint.log
octoprint_1.3.11rc1.log https://gist.github.com/kazibole/f8e61b45cbeba44e10ab7b95c352557c
octoprint_1.3.11rc2.log https://gist.github.com/kazibole/055ef8ab07b0c575b9d9783008827bcb
octoprint_1.3.11rc2_safe.log https://gist.github.com/kazibole/ec267c6019694e9e89596e563b5e383a

#### Link to contents of terminal tab or serial.log
serial_1.3.11rc1.log https://gist.github.com/kazibole/b18f730713c044cb144ba7bef61fb24c
serial_1.3.11rc2.log https://gist.github.com/kazibole/0319511f8755888d1f94e719c1a75643
serial_1.3.11rc2_safe.log https://gist.github.com/kazibole/3fec066a4d1f63961a3f452168bb0ccb

#### Link to contents of Javascript console in the browser
Did not capture.

#### Screenshot(s)/video(s) showing the problem:
[image]

I have read the FAQ.

foosel, 2019-04-09:
Thanks for the thorough report! That was actually an easy fix and an obvious reason. Fixed by the above commit for 1.3.11rc3.

foosel, 2019-04-11:
1.3.11rc3 is out which should fix this.

commit d6576306 (2019-04-09): 🐛 Add missing rstrip

Fixes #3115
closed 2019-04-11 by foosel