OctoPrint/OctoPrint #3117: Connecting to Virtual Printer causes TypeError

timothyhollabaugh, 2019-04-09:
#### What were you doing?

I was trying to connect to the virtual printer through the VIRTUAL serial port.

#### What did you expect to happen?

Octoprint to connect to the virtual printer

#### What happened instead?

Octoprint popped up the following error:
```
There was an error while trying to connect to your printer. Error: Connection error, see Terminal tab
```

In the terminal tab, the following error was present:
```
Changing monitoring state from "Offline" to "Error: Connection error, see Terminal tab"
Unexpected error while connecting to serial port: VIRTUAL TypeError: 'unsupported operand type(s) for +: 'dict' and 'str'' @ comm.py:_openSerial:2478 (hook virtual_printer)
```

In the terminal Octoprint was run from, Octoprint logged:
```
2019-04-09 17:52:24,873 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Error: Connection error, see Terminal tab"
2019-04-09 17:52:24,877 - octoprint.util.comm - ERROR - Unexpected error while connecting to serial port: VIRTUAL TypeError: 'unsupported operand type(s) for +: 'dict' and 'str'' @ comm.py:_openSerial:2478 (hook virtual_printer)
Traceback (most recent call last):
  File "/home/tim/Projects/3dprinter/octoprint/octoprint/venv/lib/python2.7/site-packages/OctoPrint-1.3.10-py2.7.egg/octoprint/util/comm.py", line 2478, in _openSerial
    serial_obj = factory(self, self._port, self._baudrate, settings().getFloat(["serial", "timeout", "connection"]))
  File "/home/tim/Projects/3dprinter/octoprint/octoprint/venv/lib/python2.7/site-packages/OctoPrint-1.3.10-py2.7.egg/octoprint/plugins/virtual_printer/__init__.py", line 33, in virtual_printer_factory
    read_timeout=float(read_timeout))
  File "/home/tim/Projects/3dprinter/octoprint/octoprint/venv/lib/python2.7/site-packages/OctoPrint-1.3.10-py2.7.egg/octoprint/plugins/virtual_printer/virtual.py", line 61, in __init__
    self._send(item + "\n")
TypeError: unsupported operand type(s) for +: 'dict' and 'str'
```

#### Did the same happen when running OctoPrint in safe mode?

Yes

#### Version of OctoPrint

1.3.10 from Github master branch.
Also occurs on latest, last, and devel-rc from pip

#### Operating System running OctoPrint

Antergos Linux

#### Printer model & used firmware incl. version

The virtual printer

#### Browser and version of browser, operating system running browser

Firefox Nightly

#### Link to octoprint.log

https://gist.github.com/timothyhollabaugh/d06fbacb71a0df896ce8521f81dd8b1c

#### Link to contents of terminal tab or serial.log

```
Changing monitoring state from "Offline" to "Error: Connection error, see Terminal tab"
Unexpected error while connecting to serial port: VIRTUAL TypeError: 'unsupported operand type(s) for +: 'dict' and 'str'' @ comm.py:_openSerial:2478 (hook virtual_printer)
```

#### Link to contents of Javascript console in the browser
N/A

#### Screenshot(s)/video(s) showing the problem:
N/A

I have read the FAQ.

foosel, 2019-04-10:
Looks like a misconfiguration of `devel.virtualPrinter.resetLines`. Do you have that adjusted in your `config.yaml` in any way? If so, please share.

timothyhollabaugh, 2019-04-10:
Here is my `config.yaml`:

```
accessControl:
  salt: si2llKewwTMCE1GXvXegYNPaRAILLxyS
api:
  key: D8F72AC7BBCD4197889E4036B6ACA561
devel:
  virtualPrinter:
    ambientTemperature: 21.3
    brokenM29: true
    capabilities:
      AUTOREPORT_TEMP: true
    commandBuffer: 4
    echoOnM117: true
    enabled: true
    extendedSdFileList: false
    firmwareName: Virtual Marlin 1.0
    forceChecksum: false
    hasBed: true
    includeCurrentToolInTemps: true
    includeFilenameInOpened: true
    m115FormatString: 'FIRMWARE_NAME: {firmware_name} PROTOCOL_VERSION:1.0'
    m115ReportCapabilites: false
    numExtruders: 1
    okAfterResend: false
    okBeforeCommandOutput: false
    okFormatString: ok
    okWithLinenumber: false
    pinnedExtruders: null
    preparedOks: []
    repetierStyleResends: false
    repetierStyleTargetTemperature: false
    resetLines:
    - start
    - Marlin: Virtual Marlin!
    - "\x80"
    - SD card ok
    rxBuffer: 64
    sendBusy: false
    sendWait: false
    sharedNozzle: false
    simulateReset: true
    smoothieTemperatureReporting: false
    supportF: false
    supportM112: true
    throttle: 0.01
    waitInterval: 1
plugins:
  announcements:
    _config_version: 1
    channels:
      _blog:
        read_until: 1551867600
      _important:
        read_until: 1521111600
      _octopi:
        read_until: 1549966200
      _plugins:
        read_until: 1554163200
      _releases:
        read_until: 1554390000
  discovery:
    upnpUuid: e4cb9127-f24a-46a7-bf8c-89101c81d3a7
  softwareupdate:
    _config_version: 6
  tracking:
    enabled: true
    unique_id: 9ce50cfe-516f-43af-8333-8c921332452b
printerProfiles:
  default: _default
serial:
  autoconnect: true
  baudrate: 115200
  port: VIRTUAL
server:
  firstRun: false
  onlineCheck:
    enabled: true
  pluginBlacklist:
    enabled: true
  secretKey: qRw4FFawCusTo66rLciLEqu9LVZMX2ha
  seenWizards:
    corewizard: 3
    cura: null
    tracking: null
```
Now that I look at it, it looks like the ` - Marlin: Virtual Marlin!` line is begin interpreted as a dictionary. I went ahead and tried quoting all of the `resetLines` items, but then it got stuck on this error:
```
Traceback (most recent call last):
  File "/home/tim/Projects/3dprinter/octoprint/octoprint/venv/lib/python2.7/site-packages/OctoPrint-1.3.10-py2.7.egg/octoprint/util/comm.py", line 2605, in _readline
    ret = self._serial.readline()
  File "/home/tim/Projects/3dprinter/octoprint/octoprint/venv/lib/python2.7/site-packages/OctoPrint-1.3.10-py2.7.egg/octoprint/plugins/virtual_printer/virtual.py", line 1342, in readline
    self._seriallog.info(">>> {}".format(line.strip()))
UnicodeEncodeError: 'ascii' codec can't encode character u'\x80' in position 0: ordinal not in range(128)
```
So it does not like the `"\x80"`. I took out that line, and now it works great! It looks like this is mainly a docs issue, since the development settings example contains both of the problematic lines.

Thanks for the hint!

foosel, 2019-04-10:
I'm surprised about the latter error since I thought that was only introduced in 1.3.11rc1 🤔 Odd. Maybe that's been in there a bit longer. It will be fixed in 1.3.11rc3 though.

But you are right, the example "Marlin" line there is missing quotation marks in the docs. Will fix that.

closed 2019-04-11 by foosel

foosel, 2019-04-11:
Docs should be fixed now.

commit 03e328fd (2019-04-10): 📝 Fix a documentation issue

Fixes #3117