OctoPrint/OctoPrint #3118: GCode Viewer: part shown in the top right corner

guestisp, 2019-04-10:
Octoprint 1.3.10 on Octopi 0.16.0

During a print (the print is properly centered on the bed), the gcode visualizer is always showing the part starting from the top right corner. The center of the part is set as the top right corner

[image]

foosel, 2019-04-11:
Thank you for reporting this!

Sadly your report lacks some of the information that is needed in order to look into this issue:

  * the browser with which you are seeing this (underlying OS and full version please)
  * unmodified full length [`octoprint.log`](https://discourse.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299/1)
  * contents of the terminal tab, or even better a [`serial.log` (please enable that first though, it's not being written by default)](https://discourse.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299/1)
  * the contents of [your JS error console](https://webmasters.stackexchange.com/questions/8525/how-to-open-the-javascript-console-in-different-browsers)
  * since this looks like an issue related to specific GCODE file properties, one or more GCODE files with which to reproduce this
  * since this might be related to your configured printer profile, screenshots of how you've set that up.

(If you need help with locating this stuff, please take a look [into the Contribution Guidelines](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#what-should-i-include-in-a-bug-report) which explains everything - and if anything is still unclear then just ask)

Could you please provide that so this issue can be analyzed and ideally reproduced and fixed? Thank you!

guestisp, 2019-04-11:
the printer profile! I've totally forgot it.
Ok, i had the printer profile set as origin from the center. that's wrong.

I think we can consider this issue solved. it was my mistake. Sorry for the noise.

closed 2019-04-11 by guestisp