OctoPrint/OctoPrint #3119: socat, pyserial, and error: (22, 'Invalid argument')

cevich, 2019-04-10:
#### What were you doing?

Loosely following [this manual install guide ](https://community.octoprint.org/t/setting-up-octoprint-on-a-raspberry-pi-running-raspbian/2337/1) inside a Fedora 29 docker container.

```
# dnf update -y
# dnf install -y PyYAML curl dnf-plugins-core findutils gcc \
git libffi-devel libselinux-python nmap-ncat openssl-devel \
patch python-devel python-pycurl python-simplejson \
python2-dnf python2-simplejson python2-virtualenv \
python3-devel python3-pycurl python3-requests python3-simplejson \
python3-virtualenv redhat-rpm-config rsync sshpass socat \
wget which
# useradd octoprint
# su - octoprint
$ virtualenv --python=python2.7 venv
$ source venv/bin/activate
$ pip install --force --upgrade pip
$ pip install https://github.com/foosel/OctoPrint/archive/1.3.10.zip
```

My printer is network connected, (not USB) so I followed [the smoothieware guide to wire a socket through to a pty device](http://smoothieware.org/octoprint).  I confirmed this was functional inside my container using the command `echo 'G28' | socat STDIO tcp:$SMOOTHIE_IP:23` (having the correct value set for `$SMOOTHIE_IP`).  Indeed my printer went through it's homing routine.

So (in another terminal, as root) I started up the `socat` listener with `-D -d -d -d -d` options (Logs I/O to stderr).  Then, as the `octoprint` user, I ran `octoprint serve`.

I accessed the web page, and performed the basic configuration.  Then I set the printer's "serial" device to the `pty` provided by `socat`, and hit connect.

#### What did you expect to happen?

Octoprint to connect and be able to communicate with my printer.

#### What happened instead?

Got this error:

```
2019-04-10 16:35:26,340 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Error: Connection error, see Terminal tab"
2019-04-10 16:35:26,341 - octoprint.util.comm - ERROR - Unexpected error while connecting to serial port: /home/octoprint/pty error: '(22, 'Invalid argument')' @ comm.py:_openSerial:2478 (hook default)
Traceback (most recent call last):
  File "/home/octoprint/venv/lib/python2.7/site-packages/octoprint/util/comm.py", line 2478, in _openSerial
    serial_obj = factory(self, self._port, self._baudrate, settings().getFloat(["serial", "timeout", "connection"]))
  File "/home/octoprint/venv/lib/python2.7/site-packages/octoprint/util/comm.py", line 2462, in default
    parity=serial.PARITY_ODD)
  File "/home/octoprint/venv/lib/python2.7/site-packages/serial/serialutil.py", line 240, in __init__
    self.open()
  File "/home/octoprint/venv/lib/python2.7/site-packages/serial/serialposix.py", line 272, in open
    self._reconfigure_port(force_update=True)
  File "/home/octoprint/venv/lib/python2.7/site-packages/serial/serialposix.py", line 438, in _reconfigure_port
    [iflag, oflag, cflag, lflag, ispeed, ospeed, cc])
error: (22, 'Invalid argument')
```

#### Did the same happen when running OctoPrint in safe mode?

Yes

#### Version of OctoPrint

1.3.10

#### Operating System running OctoPrint

Fedora 29

#### Printer model & used firmware incl. version

Tiertime Cetus Extended MkII (stripped down "Up!" printer), but modded w/ tinyfab.xyz board running `Smoothieware`.  I can connect to the Smoothieware web interface and control the printer just fine.

#### Debugging:

Spent hours searching issues related to `pyserial` bugs, got a few hits WRT the version I have installed (`3.4`).  Got a number of hits regarding using it in similar situations, trying to modify `cflag`.  In particular, attempting to set `PARITY_ODD` (or even) when this is not supported on a pty device.  In fact, in my years of experience using serial devices, I've never actually encountered one that used anything but no parity :grin:

I reset `serialposix.py` back to it's stock version, and went after ``.../site-packages/octoprint/util/comm.py`` instead.  Found around line 2459 the code sets up the serial-port for odd-parity, but then always immediately changes it to no-parity (i.e. what I want).  So I patched this:

```diff
--- a
+++ b
@@ -2459,15 +2459,14 @@
 				                           baudrates[0],
 				                           timeout=read_timeout,
 				                           write_timeout=10000,
-				                           parity=serial.PARITY_ODD)
+				                           parity=serial.PARITY_NONE)
 			else:
 				serial_obj = serial.Serial(str(port),
 				                           baudrate,
 				                           timeout=read_timeout,
 				                           write_timeout=10000,
-				                           parity=serial.PARITY_ODD)
+				                           parity=serial.PARITY_NONE)
 			serial_obj.close()
-			serial_obj.parity = serial.PARITY_NONE
 			serial_obj.open()
 
 			return BufferedReadlineWrapper(serial_obj)

```

Restarted `octoprint serve`, connected to my printer and *behold* it works perfectly.  However, I know this isn't a proper fix.  So I provide all the above detail in the hopes that this might be resolved in a future version of Octoprint by someone who knows the codebase better :smile: 

(I saw #3036 which touches this code, but obviously my version doesn't have that yet)

cevich, 2019-04-10:
@GitIssueBot but I did read the FAQ and I checked existing issues silly bot!.

I think defaulting to *odd* parity is a bug, since the (almost) universal default is *no* parity.  The code even forces it this way after setup.  Is there a good reason not to use ``serial.PARITY_NONE`` as the default?

foosel, 2019-04-11:
> Is there a good reason not to use `serial.PARITY_NONE` as the default?

#673 comes to mind, which introduced this toggling behaviour. Which in turn is based on http://forum.arduino.cc/index.php?topic=91291.0.

cevich, 2019-04-11:
Thanks for responding :smile: 

Interesting...is there no end to the oddities of serial ports?  So nobody actually knows why the open/close/open thing works...we'll call it ***JFM***  I wonder if Octoprint could get by (device-support wise) with something like this (easy to document):

```diff
diff --git a/src/octoprint/util/comm.py b/src/octoprint/util/comm.py
index 99cb3c07..8bd6279c 100644
--- a/src/octoprint/util/comm.py
+++ b/src/octoprint/util/comm.py
@@ -2459,7 +2459,7 @@ class MachineCom(object):
                                                           baudrates[0],
                                                           timeout=read_timeout,
                                                           write_timeout=10000,
-                                                          parity=serial.PARITY_ODD)
+                                                          parity=serial.PARITY_NONE)
                        else:
                                serial_obj = serial.Serial(str(port),
                                                           baudrate,
```

Since (IIUC) in the odd (Arduino) parity case, it *also* coincides with setting a fixed baud rate.  I'm assuming the `0` baud setting equates to `AUTO` mode?

I haven't tried it, but with the above it should solve my issue with `socat`, and not (hopefully) break anybody else.  Unf.  I don't have different hardware to confirm this :frowning_face: 

Also (worse) after feeding some GCode through socat to my printer, it seems I fall into the "this might be really slow" category mentioned in the link above.  So slow in fact (over gigabit ethernet), the steppers were  :cry:  for faster Gcode, stuttering all the way :face_with_head_bandage:   

Anway...not sure if this would be useful to others trying to use `socat` or not. Seems I need to dig up an empty box to try running directly off USB-serial.  Hopefully that will get me up and printing without socat.

cevich, 2019-04-13:
Okay, confirmed, socat is causing the problem.  I got a VM up with the USB device shared to it, octoprint at the helm and there's no stuttering like before.  Since `socat` won't work for me anyway, even with this fix, I guess it's not really a fix.  Sorry to take up your time.

closed 2019-04-13 by cevich

foosel, 2019-04-14:
All good, thanks for testing and getting back to me!