OctoPrint/OctoPrint #3120: [RC Feedback] Feedback on the 1.3.11rc3 Release Candidate

foosel, 2019-04-11:
Please provide *general* feedback on your experience with the 1.3.11rc3 Release Candidate here. An "All is working fine" is valuable feedback as well, because it tells me that people in fact are testing the RC and just not finding any problems. Thanks :)

**If you run into any obvious bugs not yet listed below the following line, please open a new ticket and follow ["How to file a bug report"](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report).**

---

### Currently known issues

### Unreproduced issues

  * (✔) [#3111 - 1.3.11rc2 disconnect/connect issue](https://github.com/foosel/OctoPrint/issues/3111) - potential fix included in 1.3.11rc3, still awaiting validation of underlying theory (cannot reproduce locally)

### Unreproduced and information for further analysis missing

  * ❓ ["Print completes but temp data appears to be 'stuck' at last known value for bed and hotend."](https://github.com/foosel/OctoPrint/issues/3087#issuecomment-478369220) (1.3.11rc1)

### Plugin issues

b-morgan, 2019-04-11:
I have two OctoPi systems, a Pi 3B connected to my LulzBot TAZ 6 using OctoPi 0.14.0 and a Pi 3B+ running OctoPi 0.15.1 which just today got a virtual printer. Both systems were running 1.3.11rc2 and got the announcement for 1.3.11rc3. 

The OctoPi 0.15.1 system also got the software upgrade popup (or whatever its called) and successfully upgraded to 1.3.11rc3.

The OctoPi 0.14.0 system did NOT get the software upgrade popup. I've tried restarting OctoPrint and reboot system but neither offers the upgrade. This system was upgraded from 1.13.10 to 1.13.11rc1, then rc2, then back to 1.13.10, then upgraded to 1.13.11rc2.

Suggestions for how to get it back into a state where the updates will be offered automagically are welcome!

b-morgan, 2019-04-11:
I figured it out. I had to do a "Force check for updates".

kazibole, 2019-04-12:
21 hour SD card print completed successfully!

nionio6915, 2019-04-13:
Bug submitted - appearant API key mismatch allows files to be copied to system

zeroflow, 2019-04-15:
Installed on a Raspi 3B with OctoPi 0.15.1

* Flawless update
* Printing works
* Remove access via Printoid App works the same as before

kazibole, 2019-04-17:
I have done a few more 21 hour SD card prints, as well as a print from OctoPrint, without any issues.

JohnOCFII, 2019-04-22:
- Updated successfully from rc2 to rc3
- Successfully uploaded GCODE, connected to Prusa MK3 and completed print
- Print History plugin working normally
- OctoSlack plugin working normally
- Telegram plugin working normally

isbric, 2019-04-24:
- 1.3.10 -> 1.3.11rc3 
Fixes a G4 bug (dwell hangs)

foosel, 2019-04-24:
@isbric is that a "yay, something got fixed" or a "something broke"? Feedback's a bit too short ;)

foosel, 2019-05-03:
Short update (after I finally get around into maintenance again after an awesome 3D Meetup Sweden): It looks like we'll need another RC due to what looks like a deadlock when cancelling SD prints, see #3125. I'm trying to understand the scenario under which the deadlock can happen and how to prevent it. Will be a bit before a new RC.

FormerLurker, 2019-05-04:
I'm having an issue that I thought was related to my plugin, but I've verified that it occurs even with Octolapse disabled.  When I open the settings (wrench/spanner icon) the height of the modal body appears to be way too small.  This only seems to occur when I run the application from my development environment.  It does not happen when I load OctoPrint from my pi.

Here is a screencap of what I'm seeing:

[image]

If I zoom in so that the extents of the modal window extend to the edge of the browser, then the settings re-adjust and fill the browser window height:

[image]

I looked around in the CSS and noticed that if I change the following css:
```
.full-sized-box {
    position: absolute;
    bottom: 0;
    left: 0;
    right: 0;
    top: 0;
    padding: 15px;
}
```
to
```
.full-sized-box {
    position: relative;
    bottom: 0;
    left: 0;
    right: 0;
    top: 0;
    padding: 15px;
}
```
then it works as expected.  It's also worth pointing out that this issue exists even if I disable themeify.  Also, if I use ```position: relative``` in the full-sized-box class while running OctoPrint from the pi, it still renders correctly.

Ideas?

foosel, 2019-05-06:
@FormerLurker Are you sure you observed that on the RC? I noticed that too, on `maintenance` though, and fixed it there in b3173386ae1b5efdd13a4786bfdf7aaa9d6a510e

FormerLurker, 2019-05-06:
@foosel , you are correct.  Is my face red.  I must have switched to maintenance to test out a fix you applied and forgot about it.  As usual you are one step ahead of me :)

evanquinnalaska, 2019-05-06:
About 170+ hours of printing with RC3, So far the only trouble I have seen with it so far is that I am being throttled with this release where as I was not in RC2 or RC1. throttled=0x20000 Error. It hasn't caused any issues other then some basic lag with file transfers. Once rebooted the error goes away but returns after a day printing or being idle. Not sure if its RC3 specific or if its just time for a clean image after 10+ released / versions on this SD Card.

Average temp during or when Idle is 50-52c. CPU and Memory usage all seem well within normal levels idle and while in use. 
3b+, Heat Breaks installed on GPU / CPU, Using a 2.5A, 5V Raspberry Pi Power Supply, Samsung 32gb SD Ultra SD Card. 2x C270 webcams.

foosel, 2019-05-07:
@evanquinnalaska hm... 0x20000 indicates a past overheat issue. I'm not sure how any of the changes in RC3 compared to RC2 could cause the CPU to run hotter during bootup or something, and that's the only things that could happen here from OctoPrint's side, since I only use the `vcgencmd get_throttled` command as available on the Pi and parse the output instead of rolling my own detection code.

> Side Note may need a plugin update after the stable has been released: OctoPod seems to be having a Connection error with this release: Connection Error Details: The Operation couldn't Be completed. (StarScream.WSError error1.) http based authentication. Worked before RC3.

I have no insight into OctoPod so can't say what's needed here. Isn't that also more of a third party client rather than a plugin? 🤔 There shouldn't be any backwards incompatible API changes in RC3, 2 or 1.

foosel, 2019-05-13:
The possible deadlock issue tracked in #3125 is something I still haven't been able to reproduce locally, excessive code analysis also hasn't unearthed anything and it doesn't occur for the OP anymore under safe mode. So I'm going to make a jump here and decide that we'll move forward with 1.3.11.

I'll make some final tests and look into releasing stable 1.3.11 this week.

foosel, 2019-05-15:
1.3.11 was released yesterday, thanks everyone for the feedback 🤗

closed 2019-05-15 by foosel