OctoPrint/OctoPrint #3121: RC3- API key mismatch allows files to be copied

nionio6915, 2019-04-13:
#### What were you doing?

Installed new instance of RC3 to a new card.  PC has curl script to copy files as post slicing routine from KISS. 

Tried copying new gcode file to new instance and it allowed it through. I did not expect new install to allow copying with old key. 

purposefully obfuscated api key in script to verify, and verified. 

#### What did you expect to happen?
curl copy script to fail because of bad key

#### What happened instead?
octoprint allowed script to copy gcode files even with bogus key
script in zip file
[EV2.zip](https://github.com/foosel/OctoPrint/files/3075559/EV2.zip)

#### Did the same happen when running OctoPrint in safe mode?
TBD

#### Version of OctoPrint
1.3.11RC3

#### Operating System running OctoPrint
0.16.0

#### Printer model & used firmware incl. version
experimental homebuilt running marlin 2.0 n ramps/mega

#### Browser and version of browser, operating system running browser
Windows 10, Chrome

#### Link to octoprint.log
[api-err-logs.zip](https://github.com/foosel/OctoPrint/files/3075553/api-err-logs.zip)

#### Link to contents of terminal tab or serial.log
see above

#### Link to contents of Javascript console in the browser
NA

#### Screenshot(s)/video(s) showing the problem:
https://i.imgur.com/WbHOr2k.png
https://i.imgur.com/0dSJ2Km.png
https://i.imgur.com/EOPAndM.png

I have read the FAQ. 
Yes

kantlivelong, 2019-04-13:
Seems like it only happens with access control off. Not sure if that's really a bug being clients have full access to the UI anyway. If anything it should be made a little more clear that the key is useless with access control off.

kazibole, 2019-04-13:
This may be why: https://github.com/foosel/OctoPrint/issues/3071

foosel, 2019-04-13:
See  17a7e50.

Will still take a proper look at this though, after a couple of days off.

foosel, 2019-04-24:
The behaviour is correct. The old behaviour (always require an API key, even if access control is disabled) was kinda non-sensical and the whole UI API key stuff also caused way too much confusion. 

I've adjusted the docs accordingly, see the above commit. If RC3 happens to be the final RC for 1.3.11 I'll manually cherry pick this commit into the final version to make sure the docs are up to date. Currently still waiting for more info on a ticket to be able to judge that though.

For now, take a look [here](https://docs.octoprint.org/en/maintenance/api/general.html#authorization) and [here](https://docs.octoprint.org/en/maintenance/features/accesscontrol.html) for the updated documentation.

commit 0843c0fb (2019-04-24): 📝 Reflect API key changes in docs

Closes #3121
commit 0331a25a (2019-05-14): 📝 Reflect API key changes in docs

Closes #3121
closed 2019-05-14 by foosel