OctoPrint/OctoPrint #3122: no sufficient history is available

BlackWarriorNT, 2019-04-13:
When I try to start printing, I get the following error. It started a week ago. I did not change the firmware. Octoprint installed 2 months ago and did not update it either. This error should get out already ready and previously printed models. Printing may begin by pressing the “print” button again. Sometimes this action has to be repeated several times.
`Changing monitoring state from "Operational" to "Printing"
Send: N123 M117 ETA: -*99
Recv: ok 123
Send: N0 M110 N0*125
Recv: Error:Wrong checksum
Recv: 
Recv: Resend:124
Printer requested line 124 but no sufficient history is available, can't resend
Changing monitoring state from "Printing" to "Error: Printer requested line 124 but no sufficient history is available, can't resend"
Changing monitoring state from "Error: Printer requested line 124 but no sufficient history is available, can't resend" to "Offline (Error: Printer requested line 124 but no sufficient history is available, can't resend)"
Connection closed, closing down monitor
Closing down send loop`

3D printer HE3D (clone Prusa i3) with Repetier_0.91 firmware. Octoprint version 1.3.10. Cable that the printer is connected in good order.

tedder, 2019-04-13:
The bot is right, we need your `octoprint.log`, preferably from Octoprint starting, through connecting to the printer and on to starting a job.

You might want to enable the serial log and send that too.

BlackWarriorNT, 2019-04-14:
octoprint.log:
`2019-04-14 10:50:32,585 - octoprint.startup - INFO - ******************************************************************************
2019-04-14 10:50:32,588 - octoprint.startup - INFO - Starting OctoPrint 1.3.10
2019-04-14 10:50:32,590 - octoprint.startup - INFO - ******************************************************************************
2019-04-14 10:50:32,659 - octoprint.plugin.core - INFO - Loading plugins from /home/alisa/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugins, /home/alisa/.octoprint/plugins and installed plugin packages...
2019-04-14 10:50:32,679 - octoprint.util.connectivity_checker - INFO - Connectivity changed from offline to online
2019-04-14 10:50:32,681 - octoprint.plugin.core - INFO - Plugin Anonymous Usage Tracking is disabled.
2019-04-14 10:50:32,744 - octoprint.plugin.core - INFO - Plugin Software Update is disabled.
2019-04-14 10:50:33,452 - octoprint.plugin.core - INFO - Plugin Discovery is disabled.
2019-04-14 10:50:33,474 - octoprint.plugin.core - INFO - Plugin CuraEngine (<= 15.04) is disabled.
2019-04-14 10:50:33,532 - octoprint.plugin.core - INFO - Plugin Printer Safety Check is disabled.
2019-04-14 10:50:33,551 - octoprint.plugin.core - INFO - Plugin Announcement Plugin is disabled.
2019-04-14 10:50:33,558 - octoprint.plugin.core - INFO - Plugin Action Command Prompt Support is disabled.
2019-04-14 10:50:33,571 - octoprint.plugin.core - INFO - Plugin Pi Support Plugin did not pass check, not loading.
2019-04-14 10:50:33,643 - octoprint.plugin.core - INFO - Plugin Virtual Printer is disabled.
2019-04-14 10:50:33,710 - octoprint.plugin.core - INFO - Plugin RTMP Streamer (0.1.3) is disabled.
2019-04-14 10:50:33,737 - octoprint.plugin.core - INFO - Plugin YouTube Live (0.5.0) is disabled.
2019-04-14 10:50:33,773 - octoprint.plugin.core - INFO - Found 21 plugin(s) providing 11 mixin implementations, 16 hook handlers
2019-04-14 10:50:33,862 - octoprint.server.heartbeat - INFO - Starting server heartbeat, 900.0s interval
2019-04-14 10:50:33,892 - octoprint.server - INFO - Intermediary server started
2019-04-14 10:50:33,893 - octoprint.plugin.core - INFO - Loading plugins from /home/alisa/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugins, /home/alisa/.octoprint/plugins and installed plugin packages...
2019-04-14 10:50:33,908 - octoprint.plugin.core - INFO - Plugin Pi Support Plugin did not pass check, not loading.
2019-04-14 10:50:33,963 - octoprint.plugin.core - INFO - Found 21 plugin(s) providing 11 mixin implementations, 16 hook handlers
2019-04-14 10:50:33,998 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/alisa/.octoprint/uploads...
2019-04-14 10:50:34,641 - octoprint.filemanager.storage - INFO - ... file metadata for /home/alisa/.octoprint/uploads initialized successfully.
2019-04-14 10:50:35,975 - octoprint.util.pip - INFO - Using "/home/alisa/OctoPrint/venv/bin/python -m pip" as command to invoke pip
2019-04-14 10:50:37,282 - octoprint.util.pip - INFO - Version of pip is 1.5.4
2019-04-14 10:50:37,283 - octoprint.util.pip - INFO - pip installs to /home/alisa/OctoPrint/venv/lib/python2.7/site-packages (writable -> yes), --user flag needed -> no, virtual env -> yes
2019-04-14 10:50:37,284 - octoprint.util.pip - INFO - ==> pip ok -> yes
2019-04-14 10:50:37,291 - octoprint.plugin.core - INFO - Initialized 11 plugin implementation(s)
2019-04-14 10:50:37,296 - octoprint.plugin.core - INFO - 21 plugin(s) registered with the system:
| !Action Command Prompt Support (bundled) = /home/alisa/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugins/action_command_prompt
| !Announcement Plugin (bundled) = /home/alisa/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugins/announcements
| !Anonymous Usage Tracking (bundled) = /home/alisa/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugins/tracking
|  Application Keys Plugin (bundled) = /home/alisa/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugins/appkeys
|  Backup & Restore (bundled) = /home/alisa/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugins/backup
|  Core Wizard (bundled) = /home/alisa/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugins/corewizard
| !CuraEngine (<= 15.04) (bundled) = /home/alisa/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugins/cura
| !Discovery (bundled) = /home/alisa/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugins/discovery
|  Display ETA (1.0.2) = /home/alisa/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_eta
|  EEPROM Repetier Editor Plugin (0.1.2) = /home/alisa/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_eeprom_repetier
|  Emergency Stop Button (0.1.2) = /home/alisa/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_estop
|  Force Login (bundled) = /home/alisa/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugins/forcelogin
|  Logging (bundled) = /home/alisa/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugins/logging
|  Plugin Manager (bundled) = /home/alisa/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugins/pluginmanager
| !Printer Safety Check (bundled) = /home/alisa/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugins/printer_safety_check
| !RTMP Streamer (0.1.3) = /home/alisa/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_rtmpstreamer
| !Software Update (bundled) = /home/alisa/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugins/softwareupdate
[... cut ...]
2019-04-14 10:52:33,671 - Recv: ok 1
2019-04-14 10:52:33,675 - Send: N2 M190 S70.000000*115
2019-04-14 10:52:33,704 - Recv: Error:Wrong checksum
2019-04-14 10:52:33,705 - Recv: 
2019-04-14 10:52:33,706 - Recv: Resend:2
2019-04-14 10:52:33,711 - Recv: ok
2019-04-14 10:52:33,712 - Recv: ok 2
2019-04-14 10:52:33,713 - Send: N2 M190 S70.000000*115
2019-04-14 10:52:33,718 - Send: N3 M109 S210.000000*70
2019-04-14 10:52:33,738 - Recv: TargetBed:70
2019-04-14 10:52:34,750 - Recv: T:22.22 /0 B:24.17 /70 B@:255 @:0
`

foosel, 2019-04-14:
Try to reproduce this issue in [safe mode](http://docs.octoprint.org/en/master/features/safemode.html), as instructed by the issue template and contribution guidelines. It is very much possible that this is caused by the Display ETA plugin sending a `:` in its generated `M117` message which some firmwares don't like and then cause these kinds of issues.

BlackWarriorNT, 2019-04-21:
Thanx. After removing the ETA plugin, everything worked fine.

closed 2019-04-21 by BlackWarriorNT