OctoPrint/OctoPrint #3123: printer reset when I was checking webcam feed.

napalmpt, 2019-04-14:
#### What were you doing?

opened web interface to check webcam feed like I always do in my phone (android, firefox) without problem.

#### What did you expect to happen?
show webcam

#### What happened instead?
show webcam, but print was stopped

#### Did the same happen when running OctoPrint in safe mode?

Did not test. 

#### Version of OctoPrint
OctoPrint 1.3.10

#### Operating System running OctoPrint

octopi_version: 0.15.1

#### Printer model & used firmware incl. version

Ender 3 Marlin TH3D U1.R2.9b

#### Browser and version of browser, operating system running browser
firefox 66.0.2 android oreo 8.0

#### Link to octoprint.log

https://pastebin.com/U4rNyLwF
printer stopped at time: 2019-04-14 04:18:02

#### Link to contents of terminal tab or serial.log
was not enabled

tedder, 2019-04-14:
Can you reproduce the bug, or it just happened once?

napalmpt, 2019-04-14:
It happened again just now!
I didnt open the webinterface this time. it was stopped and still heating the bed and hotend.
on the webinterface it say's: 
State: Offline (Error: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:2605)

napalmpt, 2019-04-14:
I edited the gcode and put it on the sdcardto continue the print and disconected the usb cable from the raspberry. I will see how it goes

tedder, 2019-04-14:
My two suggestions are to upgrade to a Pie and to ensure your USB cable is high quality, probably replacing it.

napalmpt, 2019-04-14:
Ok thanks, I will test another cable afterwards.
I forgot to put here the terminal tab output:
`Send: N89984 G1 X162.621 Y152.308 E9915.23948*91Recv: okSend: N89985 G1 X163.461 Y151.454 E9915.29805*82Recv: okSend: N89986 G1 X164.39 Y150.309 E9915.37014*102Unexpected error while reading serial port, please consult octoprint.log for details: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:2605Please see https://faq.octoprint.org/serialerror for possible reasons of this.Changing monitoring state from "Printing" to "Offline (Error: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:2605)"Connection closed, closing down monitor`

foosel, 2019-04-14:
Check the FAQ please, this exact message has its own entry that covers things to try *at length*.

closed 2019-04-23 by foosel