OctoPrint/OctoPrint #3124: [Request] Holistic logging level

OutsourcedGuru, 2019-04-15:
#### What were you doing?
Since I spend every day writing a plugin for OctoPrint, I'm very interested in the logs and monitor them with a `tail -f` much of the time. Since this is a lot of information to sift through, I wanted to lower the amount of logging seen and only focus on my own debugging messages.

OctoPrint -> Settings -> Logging -> octoprint -> WARNING

This resulted in a quieter log but I still notice a lot of other INFO-level logging. If I did want to quiet the world (except for my plugin of course), I would need to manually adjust **65** of these individually within the interface.

I'll probably go ahead and do this but it seems a bit much.

## What would be nice
I think I'm looking for a button/option called "Everything Except..." I would like to only see my own development's logging while I'm doing this and nothing else of the INFO-level and below. Maybe that's an asterisk entry in that pull-down at the top to toggle everything to WARNING level and then also allow me to select my own plugin and mark that to INFO... or something.

#### What did you expect to happen?
I was kind of hoping that the `octoprint.log` would just not have anything but my own logging in it.

#### What happened instead?
A variety of logging still occurs, for example...

```
2019-04-15 11:42:20,306 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2019-04-15 11:42:20,317 - octoprint.util.comm - INFO - Changing monitoring state from "Detecting baudrate" to "Operational"
2019-04-15 11:42:20,651 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from disk, was still valid
2019-04-15 11:42:20,936 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
```
#### Did the same happen when running OctoPrint in safe mode?
Yes

#### Version of OctoPrint
1.3.10

#### Operating System running OctoPrint
Raspbian (Linux) Stretch Lite from 1.15.1

#### Printer model & used firmware incl. version
Robo board, Marlin Marlin_1.1.6_RoboVersion_1.2.4 Robo C2

#### Browser and version of browser, operating system running browser
Safari 12.0.2, macOS Mojave 10.14.2

#### Link to octoprint.log
```
2019-04-15 11:42:10,437 - octoprint.startup - INFO - ******************************************************************************
2019-04-15 11:42:10,439 - octoprint.startup - INFO - Starting OctoPrint 1.3.10
2019-04-15 11:42:10,440 - octoprint.startup - INFO - ******************************************************************************
2019-04-15 11:42:10,486 - octoprint.startup - INFO - Blacklist processing done, adding 1 blacklisted plugin versions: roomtemp (any)
2019-04-15 11:42:10,527 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2019-04-15 11:42:11,448 - octoprint.plugin.core - INFO - Plugin Anonymous Usage Tracking is disabled.
2019-04-15 11:42:11,747 - octoprint.plugin.core - INFO - Plugin CuraEngine (<= 15.04) is disabled.
2019-04-15 11:42:11,757 - octoprint.plugin.core - INFO - Plugin Force Login is disabled.
2019-04-15 11:42:11,856 - octoprint.plugin.core - INFO - Plugin Pi Support Plugin is disabled.
2019-04-15 11:42:12,578 - octoprint.plugin.core - INFO - Found 16 plugin(s) providing 12 mixin implementations, 13 hook handlers
2019-04-15 11:42:12,692 - octoprint.server.heartbeat - INFO - Starting server heartbeat, 900.0s interval
2019-04-15 11:42:12,722 - octoprint.server - INFO - Intermediary server started
2019-04-15 11:42:12,723 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2019-04-15 11:42:13,386 - octoprint.plugin.core - INFO - Found 16 plugin(s) providing 12 mixin implementations, 13 hook handlers
2019-04-15 11:42:13,428 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octoprint/uploads...
2019-04-15 11:42:14,189 - octoprint.filemanager.storage - INFO - ... file metadata for /home/pi/.octoprint/uploads initialized successfully.
2019-04-15 11:42:14,258 - octoprint.plugins.softwareupdate - INFO - Loaded version cache from disk
2019-04-15 11:42:16,430 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python -m pip" as command to invoke pip
2019-04-15 11:42:18,577 - octoprint.util.pip - INFO - Version of pip is 9.0.3
2019-04-15 11:42:18,579 - octoprint.util.pip - INFO - pip installs to /home/pi/oprint/lib/python2.7/site-packages (writable -> yes), --user flag needed -> no, virtual env -> yes
2019-04-15 11:42:18,579 - octoprint.util.pip - INFO - ==> pip ok -> yes
2019-04-15 11:42:18,586 - octoprint.plugin.core - INFO - Initialized 12 plugin implementation(s)
2019-04-15 11:42:18,596 - octoprint.plugin.core - INFO - 16 plugin(s) registered with the system:
|  Action Command Prompt Support (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/action_command_prompt
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/announcements
| !Anonymous Usage Tracking (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/tracking
|  Application Keys Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/appkeys
|  Backup & Restore (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/backup
|  Core Wizard (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/corewizard
[... 38 lines cut ...]
2019-04-15 11:42:19,231 - kivy - INFO - [Python      ] v2.7.13 (default, Nov 24 2017, 17:33:09) 
[GCC 6.3.0 20170516]
2019-04-15 11:42:20,306 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2019-04-15 11:42:20,317 - octoprint.util.comm - INFO - Changing monitoring state from "Detecting baudrate" to "Operational"
2019-04-15 11:42:20,651 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from disk, was still valid
2019-04-15 11:42:20,936 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2019-04-15 11:42:21,028 - octoprint.server.util.sockjs - INFO - New connection from client: 2600:8801:9905:b400:197:8eb6:8699:a5fa
2019-04-15 11:42:21,311 - octoprint.util.comm - INFO - Printer reports firmware name "Marlin Marlin_1.1.6_RoboVersion_1.2.4 Robo C2"
2019-04-15 11:42:21,316 - octoprint.plugins.pluginmanager - INFO - Loaded notice data from disk, was still valid
2019-04-15 11:42:21,321 - octoprint.util.comm - INFO - Firmware states that it supports temperature autoreporting
2019-04-15 11:42:21,360 - octoprint.util.comm - INFO - Firmware states that it supports emergency GCODEs M108 and M410 to be sent without waiting for an acknowledgement first
2019-04-15 11:42:21,893 - octoprint.server.util.flask - INFO - Passively logging in user michael from 2600:8801:9905:b400:197:8eb6:8699:a5fa
2019-04-15 11:42:24,798 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python -m pip" as command to invoke pip
2019-04-15 11:42:24,800 - octoprint.util.pip - INFO - pip installs to /home/pi/oprint/lib/python2.7/site-packages (writable -> yes), --user flag needed -> no, virtual env -> yes
2019-04-15 11:42:24,825 - octoprint.util.pip - INFO - ==> pip ok -> yes
2019-04-15 11:43:00,125 - octoprint.server.util.sockjs - INFO - User michael logged in on the socket from client 2600:8801:9905:b400:197:8eb6:8699:a5fa
```

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

OutsourcedGuru, 2019-04-15:
For what it's worth, I made these edits and it's now reasonably quiet—it's not as bad as having to edit all 65 of them.

[image]

```
2019-04-15 12:06:36,081 - octoprint.plugins.pluginmanager - INFO - Loaded notice data from disk, was still valid
2019-04-15 12:08:22,901 - octoprint.plugins.logging - INFO - Setting logger octoprint.util.comm level to 30
2019-04-15 12:08:22,903 - octoprint.plugins.logging - INFO - Setting logger octoprint.plugins.oneclickmetal level to 20
2019-04-15 12:08:22,904 - octoprint.plugins.logging - INFO - Setting logger octoprint.server.util.flask level to 30
2019-04-15 12:08:22,906 - octoprint.plugins.logging - INFO - Setting logger octoprint.plugins.discovery level to 30
2019-04-15 12:08:22,907 - octoprint.plugins.logging - INFO - Setting logger octoprint level to 30
2019-04-15 12:08:22,908 - octoprint.plugins.logging - INFO - Setting logger octoprint.plugins.softwareupdate level to 30
2019-04-15 12:08:22,910 - octoprint.plugins.logging - INFO - Setting logger octoprint.plugins.pluginmanager level to 30
2019-04-15 12:08:22,911 - octoprint.plugins.logging - INFO - Setting logger octoprint.util.pip level to 30
2019-04-15 12:08:22,913 - octoprint.plugins.logging - INFO - Setting logger octoprint.server.heartbeat level to 30
2019-04-15 12:09:19,243 - kivy - INFO - [Logger      ] Record log in /home/pi/.kivy/logs/kivy_19-04-15_7.txt
2019-04-15 12:09:19,242 - kivy - INFO - [Kivy        ] v1.10.1
2019-04-15 12:09:19,247 - kivy - INFO - [Python      ] v2.7.13 (default, Nov 24 2017, 17:33:09) 
[GCC 6.3.0 20170516]
```

jneilliii, 2019-04-15:
In some cases it might be better to put your specific plugin logs in its own file. Would make it easier for you. Here's an example...

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/blob/master/octoprint_tplinksmartplug/__init__.py

tedder, 2019-04-15:
You can also pipe it:
```
tail -f   logfile | egrep "octoprint.plugin.myplugin|ERROR|Warning"
```

OutsourcedGuru, 2019-04-16:
Yeah, that's a good idea, actually. (Both are.)

foosel, 2019-11-05:
Providing a UI option to limit logging this much feels like it's just inviting fiddling around by people who definitely should not be fiddling around with these things and get us severely stripped log files unsuitable for providing support as a consequence. So... I really don't want to do that.

You have full control over the whole logging subsystem through `logging.yaml` and can also do the two aforementioned things. I think that needs to suffice to not give people more ammo to shoot themselves into their collective feet with.

closed 2019-11-05 by foosel