OctoPrint/OctoPrint #3125: [1.3.11rc3] freeze when stop print

devildant, 2019-04-16:
Hello,

I create this problem because it is 2 times that I encounter the problem with the RC2 and RC3
I can not produce it easily, it's pretty random :(

here is the problem : 
when I cancel the impression from the interface, it happens that the print continues, at this time the octoprint does not answer anymore, I am forced to restart from the command line because it becomes unreachable.

in the logs I see this:
`2019-04-16 16:51:32,796 - octoprint.util.comm - INFO - Changing monitoring state from "Printing from SD" to "Cancelling"
2019-04-16 16:51:32,802 - octoprint.plugins.XXX- INFO - Print cancelling because : nothing
2019-04-16 16:51:32,804 - octoprint.printer.standard.job - INFO - Print job cancelled - origin: sdcard, path: xyzcal~1.gco, owner: _api, user: devildant
2019-04-16 16:51:32,819 - octoprint.plugins.shutdownprinter - INFO - Print cancelled
2019-04-16 16:51:32,905 - octoprint.util.comm - INFO - Pausing print job due to command M25
`

after this log nothing is logged, and octoprint does not work anymore

(i use SD printing)

devildant, 2019-04-16:
Sorry @foosel , i'm a magnet for randomly  bug :(

reloxx13, 2019-04-22:
canceled like 30 test prints today, no problems.

its normal that the print continues tll the buffer is cleaned. after that the printer and octoprint works fine again.

check ur cancel gcode.

devildant, 2019-04-22:
hello, it's very very random, this is produced 3 times for me since the release of the RC2, it had never happened before the RC2.
I did not test the RC1.
I will continue my test, if I have more information, in the log / usage CPU, I will update this issue.

for the cancel gcode (octoprint settings) :
```
; disable motors
M84

;disable all heaters
{% snippet 'disable_hotends' %}
{% snippet 'disable_bed' %}
;disable fan
M106 S0
```

however, when I encounter the problem, it does not go so far.
it's just a complete freeze of octoprint and the printer is not notified for stopping printing

foosel, 2019-04-23:
Can you please provide more details and full logs? 

I'd need a *full* `octoprint.log` and a `serial.log`, and I'll also need more information on what specifically you are doing - from your few lines of log up there it sounds like you are cancelling via the *printer's controller*, not from OctoPrint? I'm completely unclear on the exact steps involved...

Also please make sure it isn't caused by some third party plugin you have installed - judging by those few log lines you have quite some that get activated on a print cancel.

devildant, 2019-04-23:
Hello @foosel,
the cancel has been done octoprint side and not printer side directly.
(even if sometimes it happens to me to do printer side, but I avoid, because sometimes there are failures, and the status is not transmitted correctly to octoprint.)

the octoprint.log : 
```
2019-04-15 18:50:50,152 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2019-04-15 18:50:50,157 - octoprint.server - INFO - --- Log roll over detected ---------------------------------------------------
2019-04-15 18:50:50,158 - octoprint.server - INFO - OctoPrint 1.3.11rc3
2019-04-15 18:50:50,168 - octoprint.plugin.core - INFO - 25 plugin(s) registered with the system:
|  Action Command Prompt Support (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/action_command_prompt
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/announcements
|  Anonymous Usage Tracking (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/tracking
|  Application Keys Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/appkeys
|  Auth Docker (1.0.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_authdocker
|  Backup & Restore (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/backup
|  Core Wizard (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/corewizard
|  Custom Control Editor (0.2.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_customControl
|  DeleteAfterPrint Plugin (1.3.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_DeleteAfterPrint
|  Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/discovery
|  Dragon Order (0.1.3) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_dragon_order
|  EEPROM Marlin Editor Plugin (1.2.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_eeprom_marlin
|  Enclosure Plugin (4.12) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_enclosure
|  Error Tracking (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/errortracking
|  Firmware Updater (1.2.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_firmwareupdater
| !Force Login (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/forcelogin
|  Logging (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/logging
|  Pi Support Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pi_support
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pluginmanager
|  Printer Safety Check (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/printer_safety_check
|  sd Switch Fast Copy (1.0.2) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_sdswitchfastcopy
|  Shutdown Printer (1.0.2) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_shutdownprinter
|  Software Update (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/softwareupdate
|  TP-Link Smartplug (0.9.13) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_tplinksmartplug
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/virtual_printer
2019-04-15 18:50:50,195 - octoprint.environment - INFO - Detected environment is Python 2.7.13 under Linux (linux2). Details:
|  hardware:
|    cores: 4
|    freq: 1200.0
|    ram: 918192128
|  os:
|    id: linux
|    platform: linux2
|  plugins:
|    pi_support:
|      model: Raspberry Pi 3 Model B Rev 1.2
|      octopi_version: 0.15.1
|  python:
|    pip: 9.0.3
|    version: 2.7.13
|    virtualenv: /home/pi/oprint
2019-04-15 18:50:50,196 - octoprint.server - INFO - ------------------------------------------------------------------------------
2019-04-15 19:05:50,154 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2019-04-15 19:20:50,158 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2019-04-15 19:35:50,162 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2019-04-15 19:50:50,166 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2019-04-15 20:05:50,169 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2019-04-15 20:20:50,173 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2019-04-15 20:35:50,176 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2019-04-15 20:50:50,181 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2019-04-15 21:05:50,183 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2019-04-15 21:20:50,188 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2019-04-15 21:35:50,191 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2019-04-15 21:50:50,196 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2019-04-15 22:05:50,199 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2019-04-15 22:20:50,203 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2019-04-15 22:35:50,206 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2019-04-15 22:50:50,210 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2019-04-15 23:05:50,213 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2019-04-15 23:20:50,216 - octoprint.server.heartbeat - INFO - Server heartbeat <3
[... 122 lines cut ...]
2019-04-16 16:40:55,623 - octoprint.plugins.shutdownprinter - INFO - Shutdown aborted.
[... cut ...]
2019-04-16 16:51:32,905 - octoprint.util.comm - INFO - Pausing print job due to command M25
```
the plugin octoprint.plugins.sdswitchfastcopy  is still in DEV, it is the plugin that I create to replace the system of copy on the sd of the printer :)

on the other hand I do not have a serial log before the 22/04 :( I did not think about the save mb)
as soon as it happens I will save them.
ps all the plugin that log the cancel status are mine, but it's just an event recovery (sdswitchfastcopy and shutdownprinter)

foosel, 2019-04-23:
Have you only have that happen while printing from SD? Or has it also happened when printing via OctoPrint directly?

From the looks of that log it's never reaching `Operational` state again, probably due to something in the printer communication being wonky. For that I'd need a `serial.log` to dig deeper.

And is that really all in the log? Nothing that follows after?

devildant, 2019-04-23:
I have not been printing from USB for a long time, I could have tried, I'll do tests.
otherwise for octoprint log, I have nothing else after, as octoprint was completely freeze (web page no response), I restarted the service (i waited 10min before restart).
I'm going to do some cancellation tests to reproduce, but that happens a little when that wants.
this is 3 times in all that I have the problem, but I have a lot of difficulty to reproduce it.
I'm sorry I forgot to save the serial.log file. I hope to reproduce quickly to transmit you the missing log

foosel, 2019-04-24:
> I restarted the service (i waited 10min before restart).

How did you restart it? Could you still SSH into the machine? Was the service still running? Since there are not even heartbeat messages anymore, either it crashed hard, or it ran into some kind of deadlock. In any case, this is very strange.

> I hope to reproduce quickly to transmit you the missing log

❤

devildant, 2019-04-24:
yes, I performed a restart service under SSH. on this side pi worked, not slow in SSH.
Hmm, I did not think to check if the service was still launched....
I really have to reproduce, the problem, so I could give you all the missing info.
I will do a maximum of test, I should have more time in the weekend.
hoping that I can quickly reproduce the problem.
I blame myself for not having thought about the serial.log and the status of the service, rrrr.

foosel, 2019-04-24:
> I blame myself for not having thought about the serial.log and the status of the service, rrrr.

Don't beat yourself up over that, it happens :) Btw, even the contents of the terminal tab help in such cases, it doesn't always have to be a full blown `serial.log`. So if you don't have that enabled at the point of encountering an issue, copy the terminal tab contents. Even if you can't reproduce the issue with enabled `serial.log` there's at least the last 300 lines of printer communication to work with then.

devildant, 2019-04-24:
ok it's noted, I'm going to work hard to reproduce the problem :)

devildant, 2019-04-27:
hello @foosel ,
the problem is reproduced,I am testing the prower loss, and after a recovery the problem is reproduced (the other 3 times it's was with normal use without power loss)
cmd ps -ef : 
```
pi       14319     1 14 11:51 ?        00:14:23 /home/pi/oprint/bin/python2 /home/pi/oprint/bin/octoprint serve --host=0.0.0.0 --port=5000
```
it's ok octoprint is run but no response.

log file :
[serial (16).log](https://github.com/foosel/OctoPrint/files/3123918/serial.16.log)
[octoprint - 2019-04-27T154138.086.log](https://github.com/foosel/OctoPrint/files/3123919/octoprint.-.2019-04-27T154138.086.log)

devildant, 2019-04-27:
note for test recovery i turn off printer power (the pi was not affected)

foosel, 2019-05-02:
Thank you for the logs. Could you please clarify what you mean with "testing the power loss" and "after recovery"? Is that a firmware feature here? 

Could you also please point out what parts of the `serial.log` refer to what situation? I'm trying to make sense of it but it's a bit confusing at the moment.

foosel, 2019-05-02:
Also, just to make sure it isn't caused by one of your plugins, please run in safe mode and make sure the problem persists there. I don't actually think it's a plugin problem, but it would be really bad if I now spent a day or more looking into this and then it turned out to be one after all.

foosel, 2019-05-02:
Looking at this a bit closer it might be a dead lock issue between `sendingLock` and `jobLock` due the `M25` now causing a call to `setPause` even for SD prints (which is a change compared to 1.3.10).

Right now I'm unable to see how there could be a deadlock here though, but that is about the only thing that might cause the described symptoms.

devildant, 2019-05-02:
Hello @foosel, yes I was testing the power loss recovery of marlin (with manual turnoff /on printer)
for the serial.log file, this is the end of the file in the last lines.
as part of my test I tested different command, to see if I could manage the recovery from octoprint (I plan to create a plugin to manage the recovery of printing after error or loss of power).

I recovered the log just after the server crash, so that is for octoprint.log and serial.log, we must aim for the last lines.

```
2019-04-27 13:29:06,122 - Connection closed, closing down monitor
2019-04-27 13:29:06,126 - Changing monitoring state from "Printing from SD" to "Offline"
2019-04-27 13:29:23,341 - Changing monitoring state from "Offline" to "Detecting serial port"
2019-04-27 13:29:23,361 - Serial port list: ['/dev/ttyUSB0']
2019-04-27 13:29:23,362 - Connecting to: /dev/ttyUSB0
2019-04-27 13:29:23,371 - Changing monitoring state from "Detecting serial port" to "Opening serial port"
2019-04-27 13:29:23,375 - Connected to: Serial<id=0x6ed007b0, open=True>(port='/dev/ttyUSB0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
2019-04-27 13:29:23,376 - Starting baud rate detection...
2019-04-27 13:29:23,376 - Changing monitoring state from "Opening serial port" to "Detecting baudrate"
2019-04-27 13:29:24,384 - Trying baudrate: 115200
2019-04-27 13:29:24,398 - Send: N0 M110 N0*125
2019-04-27 13:29:24,466 - Recv: start
2019-04-27 13:29:24,476 - Changing monitoring state from "Detecting baudrate" to "Operational"
2019-04-27 13:29:24,502 - Recv: echo: External Reset
2019-04-27 13:29:24,511 - Send: N0 M110 N0*125
2019-04-27 13:29:24,523 - Recv: Marlin TH3D U1.R2.7
2019-04-27 13:29:24,529 - Recv: 
2019-04-27 13:29:24,536 - Recv: echo: Last Updated:  | Author: (TH3D)
2019-04-27 13:29:24,543 - Recv: echo:Compiled: Apr 27 2019
2019-04-27 13:29:24,552 - Recv: echo: Free Memory: 1289  PlannerBufferBytes: 1232
2019-04-27 13:29:24,569 - Recv: echo:V55 stored settings retrieved (655 bytes; crc 17969)
2019-04-27 13:29:24,580 - Recv: echo:  G21    ; (mm)
2019-04-27 13:29:24,587 - Recv: echo:  M149 C ; Units in Celsius
2019-04-27 13:29:24,595 - Recv: 
2019-04-27 13:29:24,599 - Recv: echo:Steps per unit:
2019-04-27 13:29:24,602 - Recv: echo:  M92 X80.00 Y80.00 Z400.00 E415.00
2019-04-27 13:29:24,605 - Recv: echo:Maximum feedrates (units/s):
2019-04-27 13:29:24,609 - Recv: echo:  M203 X500.00 Y500.00 Z15.00 E50.00
2019-04-27 13:29:24,612 - Recv: echo:Maximum Acceleration (units/s2):
2019-04-27 13:29:24,616 - Recv: echo:  M201 X1000 Y300 Z100 E5000
2019-04-27 13:29:24,619 - Recv: echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
2019-04-27 13:29:24,624 - Recv: echo:  M204 P250.00 R1000.00 T250.00
2019-04-27 13:29:24,627 - Recv: echo:Advanced: Q<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
2019-04-27 13:29:24,629 - Recv: echo:  M205 Q25000 S0.00 T0.00 X11.00 Y11.00 Z0.40 E5.00
2019-04-27 13:29:24,632 - Recv: echo:Home offset:
2019-04-27 13:29:24,636 - Recv: echo:  M206 X0.00 Y0.00 Z0.00
2019-04-27 13:29:24,640 - Recv: echo:Auto Bed Leveling:
2019-04-27 13:29:24,643 - Recv: echo:  M420 S0 Z0.00
2019-04-27 13:29:24,647 - Recv: echo:  G29 W I0 J0 Z0.20000
2019-04-27 13:29:24,651 - Recv: echo:  G29 W I1 J0 Z0.06400
2019-04-27 13:29:24,654 - Recv: echo:  G29 W I2 J0 Z0.11100
2019-04-27 13:29:24,657 - Recv: echo:  G29 W I0 J1 Z0.19200
2019-04-27 13:29:24,661 - Recv: echo:  G29 W I1 J1 Z0.02800
2019-04-27 13:29:24,663 - Recv: echo:  G29 W I2 J1 Z0.11250
2019-04-27 13:29:24,668 - Recv: echo:  G29 W I0 J2 Z0.27050
2019-04-27 13:29:24,671 - Recv: echo:  G29 W I1 J2 Z0.07050
2019-04-27 13:29:24,675 - Recv: echo:  G29 W I2 J2 Z0.13300
2019-04-27 13:29:24,679 - Recv: echo:Material heatup parameters:
2019-04-27 13:29:24,682 - Recv: echo:  M145 S0 H200 B60 F0
2019-04-27 13:29:24,685 - Recv: echo:  M145 S1 H240 B100 F0
2019-04-27 13:29:24,689 - Recv: echo:PID settings:
2019-04-27 13:29:24,693 - Recv: echo:  M301 P23.87 I4.35 D32.75
2019-04-27 13:29:24,696 - Recv: echo:  M304 P61.56 I3.81 D248.39
2019-04-27 13:29:24,700 - Recv: echo:Z-Probe Offset (mm):
2019-04-27 13:29:24,704 - Recv: echo:  M851 Z-1.27
2019-04-27 13:29:24,707 - Recv: echo:Filament load/unload lengths:
2019-04-27 13:29:24,711 - Recv: echo:  M603 L60.00 U120.00
2019-04-27 13:29:29,195 - Recv: echo:SD card ok
2019-04-27 13:29:29,216 - Recv: ok
2019-04-27 13:29:29,225 - Send: N1 M115*39
2019-04-27 13:29:29,288 - Recv: FIRMWARE_NAME:Marlin TH3D U1.R2.7 SOURCE_CODE_URL: PROTOCOL_VERSION:1.0 MACHINE_TYPE:CR-10S EXTRUDER_COUNT:1 UUID:
2019-04-27 13:29:29,295 - Recv: Cap:SERIAL_XON_XOFF:0
2019-04-27 13:29:29,310 - Recv: Cap:EEPROM:1
[... 248 lines cut ...]
2019-04-27 13:31:11,368 - Recv: SD printing byte 1643/444353
2019-04-27 13:31:11,372 - Recv: ok
2019-04-27 13:31:11,627 - Send: M27
2019-04-27 13:31:11,735 - Recv:  T:224.82 /225.00 B:74.94 /75.00 @:83 B@:30
2019-04-27 13:31:12,205 - Recv: SD printing byte 1852/444353
2019-04-27 13:31:12,218 - Recv: ok
2019-04-27 13:31:12,629 - Send: M27
2019-04-27 13:31:13,223 - Recv: SD printing byte 2298/444353
2019-04-27 13:31:13,229 - Recv: ok
2019-04-27 13:31:13,630 - Send: M27
2019-04-27 13:31:13,733 - Recv:  T:228.06 /225.00 B:74.77 /75.00 @:27 B@:37
2019-04-27 13:31:14,245 - Recv: SD printing byte 2743/444353
2019-04-27 13:31:14,252 - Recv: ok
2019-04-27 13:31:14,632 - Send: M27
[... cut ...]
2019-04-27 13:31:15,280 - Recv: SD printing byte 3191/444353
2019-04-27 13:31:15,288 - Recv: ok
```

for the plugins, I can try, but I have not made a new installation since I switched to RC.
on the other hand, the problem seems to arise that very very rarely.

However, I realized that the pi was sometimes underfed and I found the cause of the problem, a USB hub become defective.
I took it off yesterday. I will try to reproduce the problem.
even if I do not remember seeing the icon on octoprint at the time of the crash.

devildant, 2019-05-02:
> Looking at this a bit closer it might be a dead lock issue between `sendingLock` and `jobLock` due the `M25` now causing a call to `setPause` even for SD prints (which is a change compared to [...]

I forgot to call it back, but the problem occurs every time, when I cancel from the octoprint interface for printing via SD.
at the moment the information is never transmitted to the printer, it continues to print, and octoprint is frozen

foosel, 2019-05-03:
What about safe mode? I cannot reproduce this at all so far. I can imagine there might be some deadlock in there, but looking at intensely I'm not able to spot it. Reproduction would help, but I can't seem to get that.

devildant, 2019-05-03:
Hello foosel, I will try in safe mode.
however before I would like to reproduce it without my usb hub (failing), because maybe a under power supply cause at the time of cancellation, could explain the bug encountered.
because even for me this bug is very very difficult to reproduce.
and I'm still looking for the conditions of this bug
but I will not forget to test in safe mode, I promise you :)

devildant, 2019-05-03:
I also tried to analyze the logs.
[image]
in the case of the bug, the M25 command never seems to arrive serial.log

I looked in the logs the cases that works
[image]
M25 correctly send to the printer

devildant, 2019-05-03:
would there be any reason in the code that would prevent sending the order to the printer?
because finally, if the command is not sent but octoprint wait are back it is normal that it blocks, right?

foosel, 2019-05-06:
> in the case of the bug, the M25 command never seems to arrive serial.log

That's what I figured, which is why my initial suspicion was a deadlock. But I went over the whole code several times and can't see any place where that makes sense, so before I continue to spend hours on hunting for that I'd prefer to fully rule out and third party plugin issues and also hardware issues.

> would there be any reason in the code that would prevent sending the order to the printer?

If it hasn't received back an `ok`, or if it's deadlocked.

> because finally, if the command is not sent but octoprint wait are back it is normal that it blocks, right?

It would be normal that it doesn't continue sending stuff to the printer, but it would *not* be normal that you can't even reach the server anymore (which I understand is happening for you). So - stuck comms, yes, but still with a usable web interface and anything that doesn't involve talking to the printer still working completely fine. Also a server heartbeat getting logged to the log every couple of minutes.

foosel, 2019-05-06:
I'd also really like to get to the bottom of this until Wednesday, because this is currently blocking the release of 1.3.11 (or 1.3.11rc4), so the earlier you can test safe mode, the better.

devildant, 2019-05-06:
hello @foosel, I tried to reproduce this week end without my hub defective, and with a 3a power supply, and I have not yet succeeded in reproducing the problem.
but hey it is very difficult to produce, the 3 times or it happened to me it was by chance.
I'll continue the tests tonight, I'll make you a return tomorrow

jubaleth, 2019-05-06:
> I'll continue the tests tonight   

In safe-mode please ;)

devildant, 2019-05-06:
I attract this kind of bug, shitty life ^^

devildant, 2019-05-06:
> > I'll continue the tests tonight
> 
> In safe-mode please ;)

yes I will try, but the only plugin that touches the event (status printer change) are my plugin in dev, my plugin shutdown printer, and the plugin enclosure, but this is only the receipt of event.

what annoys me the most is not to find the logic of prodution of the bug

kazibole, 2019-05-06:
Just to add another data point. I started and cancelled 2 SD card prints (started from the printer and cancelled from OctoPrint), and it behaved well.

devildant, 2019-05-06:
me also most of the time it works without worry, but 1/1000 i encounter the problem.
Currently, I am in safe mode, no errors encountered at the moment.
but it was also the case for this weekend in normal mode.
it makes me crazy ^^

devildant, 2019-05-11:
hello @foosel, I have not reproduced for the moment in safe mode, I will test again with my plugins.
I do not want to block the RC4 anymore (or the release 1.3.11).
what I propose is to leave for the moment this open issue, the time that I do more test, if I have new I will update the subject.
in the meantime I think you can switch to the RC4 or the final release.

foosel, 2019-05-13:
Ok... will move ahead with 1.3.11 stable then but leave this open. I have a feeling it might really be something in your plugin(s).

devildant, 2019-05-13:
I ask myself a question, I installed a USB UPS with a nut server.
I wonder if this could not be the cause of my problem.
I will test with the unplugged UPS

devildant, 2019-05-13:
because looking at the log, the serial.log never gets the M25 command.
I produced a similar bug during a development test.
I ran a subprocess that did not complete (umount command) and octoprint crashed

devildant, 2019-05-13:
I added threads in my plugin because I execute shell commands (mount, umount, copy, delete)
However, none of these actions are performed during the cancellation, but in case of doubt ..., because I managed to generate the problem with a sub-process (which not stopped, deadlock).
```
p = subprocess.Popen(["sudo",  "umount", path])
p.communicate()
p.wait()
```
according to the documentation with the function communicate the deadlocks should not happen.
and during the ocotprint crash, I did not see any of my orders blocking too long.

I will test with a limited number of plugins.

Jartza, 2019-05-15:
Yesterday I updated to Octoprint 1.3.11, I'm using Klipper firmware. I can't cancel the SD prints anymore, behavior looks similar to the issues reported in this thread. Unfortunately it seems that I didn't have serial.log enabled. This is fully reproducible on my environment, happens all the time.

foosel, 2019-05-15:
@Jartza please enable `serial.log`, reproduce, then provide the full `octoprint.log` and `serial.log`.

Jartza, 2019-05-15:
> @Jartza please enable `serial.log`, reproduce, then provide the full `octoprint.log` and `serial.log`.

I will try, funnily it failed for me already 5 times today, but after enabling serial log, I can't reproduce... will try again.

foosel, 2019-05-16:
> after enabling serial log, I can't reproduce...

Ewwww.... that sounds like it might be a race condition and that is very worrying.

devildant, 2019-05-16:
Hello,
me on my side, I still can not isolate the problem.
what I notice is that the M25 command does not arrive at the printer.
9/10 the cancel works, but 1/10 octoprint crash, however the serial log seems to continue to be filled.
it makes me think of a problem that I met some years ago on a synology server.
At the time it was the log rotation system that randomly crashed php-fpm.
I can not help but find similarity with the problem encountered.
i use marlin 1.1.9 (TH3D 2.7) and "OctoPrint 1.3.11 running on OctoPi 0.15.1"

devildant, 2019-06-15:
Hello,

the problem is produced again, this time:
- pause print
- cancel print
as in my previous logs, nothing after the "octoprint.util.comm - INFO - Pause print job due to command M25"

best regards

devildant, 2019-06-27:
Hello,
the bug continues to occur randomly.
by dint of looking at the log, I notice one thing.
During this accident, I noticed that the M27 command was sent a few milliseconds before.
if the M27 command responds exactly when sending the M25 command, can it not create an inconsistency that would crash / freeze octoprint ? 
this will explain the fact that it is very hard to reproduce the problem

devildant, 2019-06-27:
I also remarked remarkably, during the development of my plugin, that the threads were not written in the logs.
I try to give all the tracks that come to me :(

aschor, 2019-09-22:
WOW, I'M NOT THE ONLY ONE !  I'm setting up a new printer with klipper, and get this almost everytime I cancel a print, and nothing in the logs ... I'm printing with virtual sdcard, and the last message I get is something like "your printer reported an error, print will be cancelled. Message : 'sd busy'"  (but I'm already cancelling the print so double canceling ? oO).

Then I have to /etc/init.d/octoprint restart, cause I get no answer from the webserver (but it is still there as a process)

foosel, 2019-09-23:
@aschor can you please share your logs (`octoprint.log` and `serial.log`) anyhow?

aschor, 2019-09-23:
@foosel probably not before tomorrow, if I can reproduce it. But I'm pretty sure it almost always happens when I cancel during the pre-heat of the printer  (M109 or M190), where the printer actually does not answer to any Gcode command.  MAYBE I hit multiple times the "cancel" button because the temperature continues to rise...

aschor, 2019-09-24:
@foosel here it is : a server freeze when cancelling a (retraction) test print : it did NOT cancel, the file finished while octoprint webui was (and still is) unresponsive : https://gofile.io/?c=01ADZo

foosel, 2019-09-25:
What this shows is that once again the `M27` came just 10ms before the switch to cancelling state. So maybe @devildant is on to something here.

The only thing that could explain the server hanging so it doesn't even response to anything anymore would be some kind of deadlock 🤔

foosel, 2019-09-25:
Good news. I tracked it down I think and am looking at a fairly obvious deadlock now, if the conditions and timing are just right. Problem is that right now I'm not 100% sure how to solve it, but I'm sure I'll figure it out.

[image]

foosel, 2019-09-26:
Should be fixed by the above commit. Funnily enough this is now both a confirmed and unreproduced bug - I couldn't get it to reproduce even once, but based on the code it should be this. I've pushed the fix to the `maintenance` branch. Should another RC for 1.3.12 still be needed (unlikely based on current feedback though) I'll see about merging into that release, otherwise it'll have to wait until the next maintenance release.

It would be great though if those of you who *can* more or less reliably reproduce it would give the `maintenance` branch with the fix a test and see if the issue is solved indeed. It should but without being able to reproduce it myself the solution is purely theoretical at this point.

devildant, 2019-09-26:
hello @foosel , I do not have the time to test yet, however if I find a moment, I'll make you a return.
however, it may take some time since I do not systematically reproduce the bug (which always happens a few ms after the M27)
in any case the "trigger" seems to me to be a good track
thanks for the work :)

aschor, 2019-10-08:
hello @foosel, I tested multiple times to cancel prints with the maintenance branch, and I did NOT get any lockup, I think that's fixed now 👍 

However ... I have to cancel the print 2 times : 
- first time, the print stops, then I get an error and I cannot launch a new print, octoprint still thinks it is printing : "Error reported by printer
Your printer's firmware reported an error. Due to that the ongoing print job will be cancelled. Reported error: SD busy"

- I hit a second time the cancel button, and THEN I get de "print" button enabled again, and I can continue. (my firmware is klipper)

foosel, 2019-10-08:
@aschor would you have a `serial.log` for that by any chance? Without one I don't think I'll be able to debug that.

aschor, 2019-10-08:
[serial.log](https://github.com/foosel/OctoPrint/files/3703405/serial.log)

here you are, just cancelled a print (twice). As I see it I jump from "Cancelling" to "Operational" then for no reason from "Operational" to "Starting print from SD" (that twice).

on my 2nd cancellation, I stay at "operational"

foosel, 2019-10-09:
Ah. That looks like a different issue. That's more a problem with the SD print detection code in place for when people start an SD print from the printer controller causing a problem here. I'll have to take a look at that, but it merits a new ticket.

foosel, 2019-10-09:
This is now tracked in #3301

foosel, 2020-03-04:
1.4.0 has been released.

closed 2020-03-04 by foosel