OctoPrint/OctoPrint #3128: Gcode logs

Simur, 2019-04-22:
Bonjour à tous ! 

J'aimerais développer un plugins pour un module PCB qu'un ami à conçu.
C'est un module de power loss recovery. 
Du coup mon idée serait ceci : 

Le module envoie 6V lorsque le 220V est présent. Lorsque le 220V disparaît, le module envoie 0V.
Le module alimentera l'imprimante via une banque de condensateur pour parquer la tête et reffroidir tant que c'est possible. 
Il me faut donc un plugin qui : 
-Détection perte du courant via GPIO.
-Envoie Gcode prioritaire pour parquer la tête.
-Log de la dernière ligne Gcode effectuée.
-Si courant reviens avant que la banque de condensateur soit vide, reprise d'impression.
Merci pour votre aide ! 

-------------
Hello everyone !

I would like to develop a plugins for a PCB module that a friend has designed.

It is a module of power loss recovery.
So my idea would be this:

The module sends 6V when the 220V is present. When the 220V disappears, the module sends 0V.
The module will power the printer via a capacitor bank to park the head and cool as much as possible.
So I need a plugin that:
-Detection power loss via GPIO.
-Send priority Gcode to park the head.
-Log of the last line Gcode performed.
-If 220V come back before the capacitor bank is empty, resume printing.

Thanks for your help !

foosel, 2019-04-23:
For questions or other kind of help or support, please get in touch on the [community forum](https://community.octoprint.org). The bug tracker is reserved for bugs, feature requests and brainstorming tickets. See also the [Contribution Guidelines](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md). Thank you! :)

closed 2019-04-23 by foosel

Simur, 2019-04-23:
Ok, sorry for inconvenience =)