OctoPrint/OctoPrint #3130: Environment versions too high (?)

danielvandenberg95, 2019-04-23:
#### What were you doing?

1. Go to 'settings'
2. Click on 'software update'
3. See error
[image]

#### What did you expect to happen?
Be able to update

#### What happened instead?
I get the givenwarning

#### Did the same happen when running OctoPrint in safe mode?
Software update is not available in safe mode.

#### Version of OctoPrint
1.3.10

#### Operating System running OctoPrint
Distributor ID: Debian
Description:    Debian GNU/Linux testing (buster)
Release:        testing
Codename:       buster

#### Printer model & used firmware incl. version
Anet AM8 (but doesn't seem relevant)

#### Browser and version of browser, operating system running browser
Any

#### Link to octoprint.log
https://gist.github.com/danielvandenberg95/d23fa69786f62d41b894c152c9cd66d0

#### Link to contents of terminal tab or serial.log
Not relevant

#### Link to contents of Javascript console in the browser
Not relevant

#### Screenshot(s)/video(s) showing the problem:
[image]

I have most definitely read the FAQ.

danielvandenberg95, 2019-04-23:
Okay, what did I do wrong...

foosel, 2019-04-24:
Your Python version reports as "2.7.15+" which isn't a proper PEP440 compliant Python version. For some reason someone maintaining the Debian Python packages decided to add a + there where it doesn't belong. That is already tracked in #3049 and a workaround exists in 1.3.11 (including the RCs). Debian also fixed their versioning. 

Two options:

  * update your Python package: `apt update && apt install python2.7` and then update/redo the virtual environment
  * or manually install 1.3.11rc3: `/home/octoprint/octoprint/OctoPrint/venv/bin/pip install https://get.octoprint.org/maintenance-rc`

> Okay, what did I do wrong...

Looks like you figured it out. The bot isn't capable of natural language processing and doesn't like changes to its check phrase ;) And yes, I wish that thing wasn't needed too, but too many people just never provided me the info that I need to look into bugs so that a bot became necessary.

closed 2019-04-24 by foosel