OctoPrint/OctoPrint #3132: Login failure, race condition, when using apache for reverse proxy

smoofra, 2019-04-24:
#### What were you doing?

Logging into octoprint.

I'm using a reverse HTTP proxy, with the apache config settings from here:
https://community.octoprint.org/t/reverse-proxy-configuration-examples/1107

1. Go to https://mysever/octoprint
2. Click on 'log in'
3. Sometimes it logs in, sometimes it just brings you back to a black login page.

#### What did you expect to happen?

I expected to go to the main octoprint UI page.

#### What happened instead?

It went back to the login page.  Sometimes.  Sometimes it works.

#### Did the same happen when running OctoPrint in safe mode?

yes.

#### Version of OctoPrint

reproduces in 1.3.10 and also  on the latest master (03e328fd1392161c44a7170dc086e6bcfca8b200)

#### Operating System running OctoPrint

ubuntu 18.10 

#### Printer model & used firmware incl. version

n/a

#### Browser and version of browser, operating system running browser

Chrome Version 73.0.3683.103 (Official Build) (64-bit)
macOS.   
also reproduces with Safari

#### Link to octoprint.log

https://odin.elder-gods.org/paste/octoprint-2019-04-23.log

#### Link to contents of terminal tab or serial.log

WebSocket connection to 'wss://XXXXX/octoprint/sockjs/XXX/XXXXX/websocket' failed: Error during WebSocket handshake: Unexpected response code: 400

That's the only message in the console)

#### Screenshot(s)/video(s) showing the problem:

none

I have read the FAQ.

foosel, 2019-04-24:
Can you please share your exact apache config and the exact URL you call it under? Since there are multiple examples there, I'd like to see which one exactly you've chosen and also if there are any typos in there. The double slash `404` in the log might be relevant.

Also please provide a screenshot of the network tab of your browser's dev console.

smoofra, 2019-04-24:
url is `https://odin.elder-gods.org/inky/`   Apache config is:

```
Redirect 301 /inky /inky/

<Location "/inky/">
  RequestHeader set X-SCRIPT-NAME "/inky/"
  RequestHeader set X-SCHEME https
  #RequestHeader set X-FORWARDED-HOST "odin.elder-gods.org"

  ProxyPass  "http://localhost:5000/"
  ProxyPassReverse "http://localhost:5000/"

  RewriteEngine on

  RewriteCond %{HTTP:UPGRADE} =websocket [NC]
  RewriteRule /inky/(.*) ws://localhost:5000/$1 [P,L]

  RewriteCond %{HTTP:UPGRADE} !=websocket [NC]
  RewriteRule /inky/(.*) http://localhost:5000/$1 [P,L]

  AuthName inky
  AuthType Basic
  AuthBasicProvider file
  AuthUserFile /etc/apache2/passwords
  Require user lawrence_danna
</Location>
```

CraXyOW3, 2019-04-28:
I just had this problem, tho i do not have Apache, I am using nginx and what i had to do was adding.
`proxy_set_header Upgrade websocket;`
in my sockjs location.

Perhaps this can point you in the right direction.

smoofra, 2019-04-28:
I spent hours messing with the Apache settings the day I created this issue.   Most of the time it was just impossible to log in.  At one point it would log in maybe 1 in 10 times.      Now I can't reproduce it anymore.     Oh well.   This level of frustration is pretty typical when dealing with Apache.

foosel, 2019-05-03:
Sorry, only came around to taking another look now (3d printing event...)

> Also please provide a screenshot of the network tab of your browser's dev console.

I'm still lacking this piece of information.

foosel, 2019-05-03:
... For some reason my browser only now showed me the reply.

@smoofra Should be close this since you cannot reproduce it anymore?

foosel, 2019-05-09:
@smoofra ping? :)

smoofra, 2019-05-29:
Yea I can't reproduce anymore.  I don't know WTF was going on.   If I see it again I'll try to boil it down to a docker image or something that can be shared.

closed 2019-05-29 by smoofra

foosel, 2019-05-31:
Maybe it was just Gremlins 🤷‍♀️