OctoPrint/OctoPrint #3135: Page loads slowly or not at all

seamusmcgoon4d, 2019-04-30:
For about a month now, when trying to login to any of my three Octoprint devices I either can't login at all or it takes about five minutes for the main page to load.

What logs can I check or look at for more help?

I've tried checking the CPU usage on the Raspberry Pi's

```
  PID USER      PR  NI    VIRT    RES    SHR S  %CPU %MEM     TIME+ COMMAND                                                                 
  314 avahi     20   0    6792   3444   2808 S   1.0  0.4  12:25.01 avahi-daemon                                                            
13845 pi        20   0    8312   3292   2788 R   0.7  0.4   0:04.40 top                                                                     
  397 pi        20   0  295312  63404  10372 S   0.3  7.1  60:52.11 octoprint                                                               
13368 root      20   0       0      0      0 I   0.3  0.0   0:06.65 kworker/u8:1                                                            
    1 root      20   0   28148   6160   4856 S   0.0  0.7   0:04.59 systemd     
```

tedder, 2019-04-30:
Try `curl`ing the page to remove some variables. Also look on Chrome's do tools network console to see if it is a specific request.

Which Pi models? Post an octoprint.log from one of them showing the startup through your web request.

WillBro, 2019-05-11:
@seamusmcgoon4d I was unable to receive a response from the server. I have a suggestion though as I've only been using OctoPrint for a short time I'm not sure how valid a suggestion this is.

What worked for myself was to navigate to `/home/pi/.octoprint/data` and remove the file `preemptive_cache_config.yaml`. I was then able to access the page.

closed 2019-05-15 by GitIssueBot