OctoPrint/OctoPrint #3138: Octoprint 1.3.11rc3 - Can't Upload STLs

strikerfired, 2019-05-03:
Preface: used the printer a week ago and printed existing gcode on the server, after upgrading to RC3.  That print functioned as expected

####Problem
Attempting to upload a .STL to be sliced for print, using the file interface on the left side of the screen below the connection settings.  I AM able to upload files I manually sliced in CURA.

The upload dialogue wouldn't let me select STL files, so I attempted a drag-and-drop into the file control, however I received an immediate error message.

This has been replicated with any STL file on my system.

#### What did you expect to happen?
I expected the file to upload and be prompted to slice

#### What happened instead?
Error message indicating failure OR complete inability to select an STL from the file browser 

#### Did the same happen when running OctoPrint in safe mode?
Yes
(I'm aware of a few plugin issues on my configuration; as evident in the logs)

#### Version of OctoPrint
OctoPrint 1.3.11rc3 running on OctoPi 0.15.1

#### Printer model & used firmware incl. version
Anycubic Chiron 1.2.7

#### Browser and version of browser, operating system running browser
OSX/Safari 12.0.3 (only browser I've used, never had a problem before)

#### Link to octoprint.log
https://gist.github.com/strikerfired/43c2f5f83bda9c1e544b0ea87938b2d5

#### Link to contents of terminal tab or serial.log
https://gist.github.com/strikerfired/e7ac58a48b57b3b2265cbbfe6681cb6b

#### Screenshot(s)/video(s) showing the problem:
https://imgur.com/a/x5WxcLE

I have read the FAQ.

foosel, 2019-05-03:
Did you install the CuraLegacy plugin from the repository? As the changelog indicates, the CuraEngine plugin was unbundled from OctoPrint in 1.3.11 and in order to slice (and hence also in order to be allowed to upload STL files) you need to have a slicer plugin installed and active.

strikerfired, 2019-05-03:
Apologies, I must have missed that in the change log.  Plugin installed, functions as intended.

closed 2019-05-03 by strikerfired