OctoPrint/OctoPrint #3139: OctoPrintFlaskResponse doesn't behave correctly for generator

kennethjiang, 2019-05-03:
#### What were you doing?

I am implementing a new API in my plugin that implements `BlueprintPlugin`. This new API is supposed to return a Response with generator so the response is actually a stream.

That API is tested ok when it's running inside a separate Flask server. However, when I moved the same code inside my plugin, the endpoint never sends HTTP response headers back.

The relevant code.

```
        @octoprint.plugin.BlueprintPlugin.route("/mjpeg", methods=["GET"])
        def get_mjpeg(self):
                from octoprint.server.util.flask import OctoPrintFlaskResponse
                boundary='herebedragons'
                return OctoPrintFlaskResponse(self.mjpeg_generator(boundary), mimetype='multipart/x-mixed-replace;boundary=%s' % boundary)

```

I believe OctoPrintFlaskResponse somehow mishandled generator. Even if I return a `flask.Response`, OctoPrint would wrap it into a `OctoPrintFlaskResponse` before handing it back to Flask and results in the same behavior.

I attached the test plugin so that you can reproduce the problem.[test_plugin.zip](https://github.com/foosel/OctoPrint/files/3143726/test_plugin.zip)

Steps to reproduce:

1. Download the zip file and  `unzip test_plugin.zip`

2. `cd OctoPrint-Test && /home/pi/oprint/bin/python2 ./setup.py develop && sudo service octoprint restart`

3. Test the API directly in Flask

```
cd octoprint_test && /home/pi/oprint/bin/python2 __init__.py
```

4. Use `curl` to test it.

```
pi@octopi2:/tmp $ curl -v http://localhost:8080/mjpeg >/dev/null
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0*   Trying ::1...
* TCP_NODELAY set
* connect to ::1 port 8080 failed: Connection refused
*   Trying 127.0.0.1...
* TCP_NODELAY set
* Connected to localhost (127.0.0.1) port 8080 (#0)
> GET /mjpeg HTTP/1.1
> Host: localhost:8080
> User-Agent: curl/7.52.1
> Accept: */*
>
* HTTP 1.0, assume close after body
< HTTP/1.0 200 OK
< Content-Type: multipart/x-mixed-replace;boundary=herebedragons
< Connection: close
< Server: Werkzeug/0.8.3 Python/2.7.13
< Date: Fri, 03 May 2019 22:25:00 GMT
<
{ [8192 bytes data]
100  124M    0  124M    0     0  15.4M      0 --:--:--  0:00:08 --:--:-- 21.0M^C
```

The response was correct.

5. Now test the equivalent API in OctoPrint:

```
pi@octopi2:/tmp $ curl -v http://localhost/plugin/test/mjpeg >/dev/null
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0*   Trying ::1...
* TCP_NODELAY set
* Connected to localhost (::1) port 80 (#0)
> GET /plugin/test/mjpeg HTTP/1.1
> Host: localhost
> User-Agent: curl/7.52.1
> Accept: */*
>
  0     0    0     0    0     0      0      0 --:--:--  0:00:09 --:--:--     0
```

Note that no response headers were ever returned. The for loop in generator was actually executed by somehow the output never goes to Response and soon OctoPrint will run out of memory 

#### What did you expect to happen?

The API `http://localhost/plugin/test/mjpeg` running in OctoPrint behaves the same as the API directly runs in Flask.

#### What happened instead?

The API `http://localhost/plugin/test/mjpeg` running in OctoPrint  fails to return

#### Did the same happen when running OctoPrint in safe mode?

Not relevant. Safe mode doesn't load the plugin.

#### Version of OctoPrint

1.3.10

#### Operating System running OctoPrint

OctoPi 0.16.0

kennethjiang, 2019-05-03:
@foosel I skipped some of the fields that are not relevant since this is an issue with plugin development. But let me know if you still want me to fill them out.

foosel, 2019-05-09:
So, here's the problem. Flask runs inside Tornado as a WSGI application, meaning it runs fully synchronous - no streaming support. In fact, trying to stream will block the single threaded server and make any other requests fail.

If you need to stream data, you'll have to skip Flask and implement a Tornado route.

kennethjiang, 2019-05-11:
Thanks @foosel for looking into this issue.

Is there a way to specify the number of "workers" (I'm borrowing the concept from `gunicorn` since I'm not familiar with `tornado`)? Being able to handle only 1 request at a time seems to be very limiting for a web server.

kennethjiang, 2019-05-11:
Nevermind. I think I figured out a way to work around this problem. Thank you @foosel  for looking into this issue for me.

closed 2019-05-11 by kennethjiang