OctoPrint/OctoPrint #3140: Make restore work over filesystems.

stone, 2019-05-04:
#### What were you doing?

Restoring a backup, the backup plugin are using os.rename which does not
work over filesystems (cross partitions, disks, etc):
https://github.com/foosel/OctoPrint/blob/03e328fd1392161c44a7170dc086e6bcfca8b200/src/octoprint/plugins/backup/__init__.py#L673

https://docs.python.org/2/library/os.html#os.rename

> The operation may fail on some Unix flavors if src and dst are on different filesystems

Simple fix is to use shutil.move instead, which works across filesystems (it is already imported)
`shutil.move(temporary_path, final_path)`

https://docs.python.org/2/library/shutil.html#shutil.move

> If the destination is on the current filesystem, then os.rename() is used. Otherwise, src is copied (using shutil.copy2()) to dst and then removed.

#### What did you expect to happen?

Backup restored.

#### What happened instead?

```
(oprint) pi@octopi:/ $ octoprint plugins backup:restore bkp.zip
Initializing settings & plugin subsystem...
Unpacking backup to /tmp/tmpu09sMf...
Unpacked
Renaming /home/pi/.octoprint to /home/pi/.octoprint.bck...
Removing temporary unpacked folder
Error while running restore
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/backup/__init__.py", line 823, in _restore_backup
    os.rename(basedir, basedir_backup)
OSError: [Errno 18] Invalid cross-device link
Removing temporary zip
Restoring from /tmp/tmpT4RsYn failed
```

#### Version of OctoPrint
octoprint, version 1.3.10

#### Operating System running OctoPrint
```
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 9.4 (stretch)
Release:	9.4
Codename:	stretch
```

#### Link to octoprint.log
https://gist.github.com/stone/839508841c72f39d6f2d76bfee24cb34

tedder, 2019-05-04:
FWIW, this is fixed in `devel`, which will become 1.4.0.

tedder, 2019-05-04:
See #2992 (which should probably be closed too).

foosel, 2019-05-06:
It's actually even fixed in `maintenance` in 1.3.11rc3 and this is a duplicate of #2992 (which is still open since 1.3.11 hasn't yet been released as stable, but the ticket is tagged as solved).

closed 2019-05-06 by foosel