OctoPrint/OctoPrint #3142: Auto login - logout does not work

seba-78, 2019-05-06:
#### What were you doing?

In octoprint 1.3.11RC3 I could finally make the auto login working. It did not work with 1.3.10.

My problem is that I cannot logout and login with a different user (switch users), because I am immediately logged in again with the user specified in the config.yaml . However, for maintenance I would like to login with an administrator account. This is not possible with this behaviour.

#### What did you expect to happen?

select logout and then be directed to the login page

#### What happened instead?

immediately logged in again with the user specified in the config.yaml 

#### Did the same happen when running OctoPrint in safe mode?

yes, it is the same bahaviour

#### Version of OctoPrint

1.3.11rc3

#### Operating System running OctoPrint

OctoPi 0.13.0

#### Printer model & used firmware incl. version

Ultimaker 2+

#### Browser and version of browser, operating system running browser

Firefox 60.5.2esr

#### Link to octoprint.log

https://gist.github.com/seba-78/99e9c528840823622ccdde46aa2038d4

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

https://gist.github.com/seba-78/99e9c528840823622ccdde46aa2038d4

I have read the FAQ.

foosel, 2019-05-07:
Just checked against `master`/1.3.10 and this is not a regression in 1.3.11. Hence scheduling this for 1.3.12.

foosel, 2019-05-07:
Solved via the above commit, already on `maintenance`, soon `devel`, to be released with 1.3.12.

commit e1e52130 (2019-05-07): 🐛 Only do autologin if not actively logged out

Fixes #3142
closed 2019-10-22 by foosel