OctoPrint/OctoPrint #3145: Chunking Upload Problem with Flask

FormerLurker, 2019-05-09:
#### What were you doing?
I implemented a jquery.fileupload.js upload, and noticed the 100KB upload limit per request, and instead of increasing the upload size via the bodysize_hook, I decided chunking upload was the way to go.  I added a ```maxChunkSize: 100000``` option to the initialization of the fileupload wildget, and noticed that flask was not receiving the entire file.  At first I thought (expected actually) that I needed to implement the chunking process myself, but (as we discussed already), OctoPrint attempts to handle this before my handler gets called.  I only receive the first chunk.

I figured perhaps the bodysize_hook was still restricting the upload, so I implemented it and increased the max size to 5mb.  I still only receive the first 100000 bytes, however.

##### Steps to reproduce:
1.  Create a fileupload widget.
2.  Add the following option: ```maxChunkSize: 100000```
3.  Create a handler for the upload plugin.
4.  Upload a file whose size is greater than the maxChunkSize and see that only the first chunk is received.
5.  Add a bodysize_hook and increase the max size to something greater than your upload file size and observe the same behavior.

#### What did you expect to happen?
I expected to either be passed a 'files' object within flask to handle chunking myself, or for the entire upload to be handled by the OctoPrint wrapper.

#### What happened instead?
Only the first chunk was received.

#### Did the same happen when running OctoPrint in safe mode?
I cannot run this in safe mode since I'm testing a plugin.

#### Version of OctoPrint
OctoPrint 1.3.11rc3 

#### Operating System running OctoPrint
Windows 10

#### Printer model & used firmware incl. version
virtual printer

#### Browser and version of browser, operating system running browser
Version 74.0.3729.131 (Official Build) (64-bit)
Windows 10.0.17763 Build 17763

#### Link to octoprint.log
The log is not helpful here, since nothing related to the file upload appears to be logged.  Let me know if I'm in error here and I'll do whatever is necessary to give you a proper log.

#### Link to contents of terminal tab or serial.log
NA - this is not related to gcode or serial processing

#### Link to contents of Javascript console in the browser
No console information is logged during the upload except for what I log in my plugin, which I have already described.  I could give you the entire request if that helps.  Let me know

#### Screenshot(s)/video(s) showing the problem:
NA, not a UI issue

I have read the FAQ.

foosel, 2019-05-10:
What would be glorious is if you could share a minimal viable plugin with which to reproduce this, to make sure I'm not making a mistake in reproduction.

As already said on Discord though, that definitely sounds like a bug in OctoPrint's file upload wrapper that needs to be fixed. And probably not a regression in 1.3.11 since I didn't touch that stuff in a long time.

FormerLurker, 2019-05-10:
Here you go:

[OctoPrint-Mvp.zip](https://github.com/foosel/OctoPrint/files/3167583/OctoPrint-Mvp.zip)

After you install, go to MVP Plugin tab and upload a text file.  It must be text because I populate the response with json.dumps with the following code:
```
return json.dumps({"data":text_data}), 200, {'ContentType': 'application/json'}
```

The behavior is slightly different than what I was seeing in my plugin for whatever reason, probably because of my error handling.  However, if you pick a file > 100KB it will upload in multiple chunks (look at the network tab in the browser debugger).  It WILL process each chunk, but they are sent one at a time, and there doesn't appear to be any indication that there are more chunks available.  The javascript will return success after all chunks are uploaded.

I'm less sure this is a bug, but since the flask behavior is custom I'm not sure exactly what to expect.

foosel, 2019-05-13:
Now look at that. I thought we were talking about [Chunked transfer encoding](https://en.wikipedia.org/wiki/Chunked_transfer_encoding) which the server definitely should support. This however looks completely different - multiple requests to the same endpoint, each with a multipart payload containing a bit of data. The server currently doesn't support that. I'm not sure if it can be easily made to support it either to be honest since we currently handle file uploads in a request context without state tracking. The underlying [`UploadStorageFallbackHandler`](https://github.com/foosel/OctoPrint/blob/03e328fd1392161c44a7170dc086e6bcfca8b200/src/octoprint/server/util/tornado.py#L72) would need to be extended to handle this kind of upload mechanism, build up the temporary file and only pass it through to flask once the final chunk has been received. So tornado would see the individual chunk requests but flask wouldn't. That way it would be handled transparently and also wouldn't cause issues with the existing upload handling. [Maybe something similar to this wrapped into `UploadStorageFallbackHandler`](https://codecalamity.com/uploading-large-files-by-chunking-featuring-python-flask-and-dropzone-js/).

Is there any particular reason why you go that route? The browser should actually already stream the upload just fine using chunked transfer encoding, without a need for manual chunking. The only advantage that might give you is resuming file uploads, but is that really of concern here?

Reclassifying this as a request - if there's a real need for this kind of functionality I'd be happy to accept pull requests. I fear it might be a while until I can look into this.

FormerLurker, 2019-05-13:
Foosel,

I wanted to use it primarily to bypass the need to bypass a predefined body size limit.  The only other fileupload I have was submitted via a pull request, and included chunking upload as well.

It's not a big deal, so don't sweat it!  If I ever have time I'll take a look at that UploadStorageFallbackHandler and see if it can be implemented.  It's not a priority.  I only submitted the ticket because you thought it might be a bug.

Thanks!