OctoPrint/OctoPrint #3148: Difference in speed when using different webcam URL

sfjuocekr, 2019-05-12:
After asking myself why the webcam stream was dreadfully slow, I tested the following:

Change URL from: webcam/?action=stream
To: http://x.x.x.x:8080/?action=stream

The first, default, URL gives an increasing delay in the stream.
The second, streams with the speed it should be.

Now I have not taken a look at the underlying code, but I assumed "webcam" renames to the HTTP-REFERRER?

If so, why does this happen?

foosel, 2019-05-13:
For questions or other kind of help or support, please get in touch on the [community forum](https://community.octoprint.org). The bug tracker is reserved for bugs, feature requests and brainstorming tickets. See also the [Contribution Guidelines](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md). Thank you! :)

closed 2019-05-13 by foosel

sfjuocekr, 2019-06-18:
I'm pretty sure this is a bug, I've been able to replicate it easily by just changing the stream URL on multiple devices.

foosel, 2019-06-18:
If it's a bug, it's not a bug in OctoPrint since OctoPrint only *embeds* the webcam, it doesn't control nor serve it itself.

sfjuocekr, 2019-06-25:
But it does some mangling on the default URL, which in my case causes the problem.

The webcam stream as served by mjpg_streamer works as it should, it works with a manually entered URL and not with the "default" one.