OctoPrint/OctoPrint #3150: Since Update to OctoPrint 1.3.11 auf OctoPi 0.16.0 the Telegram Plugin wont work!

xenon2008, 2019-05-14:
#### What were you doing?

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

#### What did you expect to happen?
Since i have updated today from 1.3.10 to OctoPrint 1.3.11 auf OctoPi 0.16.0 the Telegram Plugin wont work.

#### What happened instead?
For example, if I restart the RPi, I always get a message in Telegram with:

>  ERROR: I was not able to format the Notification for 'PrinterStart' properly. Please open your OctoPrint settings for Telegram Notifications and check message settings for 'PrinterStart'.
#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPrint
 OctoPrint 1.3.11 auf OctoPi 0.16.0  

#### Operating System running OctoPrint
RPI 3b v1.2

#### Printer model & used firmware incl. version
Ender 3 Pro

#### Browser and version of browser, operating system running browser
WIndows 10 x64, with alle Browsers.

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

tedder, 2019-05-14:
Assuming Octoprint starts up, what does that PrinterStart setting look like?

Also, your octoprint.log is required for any assistance to be possible.

xenon2008, 2019-05-15:
The Error also is when i Shutdown the Octopi (as Example)
So it is not only an Error when Octoprint starts!

Here is my Octoprint.log
[octoprint.log](https://github.com/foosel/OctoPrint/files/3180928/octoprint.log)

Hope this helps..

Best Regards

tedder, 2019-05-15:
You missed answering this:
> Assuming Octoprint starts up, what does that PrinterStart setting look like?

The error is clear, it's either the upgrade is causing the plugin to fail or you have it misconfigured.
```
2019-05-15 05:33:11,151 - octoprint.server.api - ERROR - Error calling SimpleApiPlugin telegram
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/__init__.py", line 68, in pluginData
    response = api_plugin.on_api_get(request)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_telegram/__init__.py", line 858, in on_api_get
    'bind_cmd':[k for k, v in self.tcmd.commandDict.iteritems() if 'bind_none' not in v ],
AttributeError: 'NoneType' object has no attribute 'commandDict'
2019-05-15 05:33:11,171 - tornado.access - ERROR - 500 GET /api/plugin/telegram?bindings=true (::ffff:10.242.2.2) 32.01ms

```

It's up to [the creator/maintainer of the telegram plugin](https://github.com/fabianonline/OctoPrint-Telegram) to fix it, not an problem with OctoPrint itself.

xenon2008, 2019-05-15:
Sry my English is not so Good....

PrinterStart Settings is Set to Default:
{emo:rocket} Hello. I'm online and ready to receive your commands.

I have default Settings in the Plugin Options....
OK, so i must post it to the creator ;)

Thanks!

foosel, 2019-05-15:
As already pointed out, that is a problem with the plugin, not OctoPrint. The RC was out for the better part of two months.

closed 2019-05-15 by foosel