OctoPrint/OctoPrint #3151: Update from OctoPrint 1.3.10 to OctoPrint 1.3.11 failed

JustAnother1, 2019-05-14:
#### What were you doing?

I booted my RaspberryPi3 and it informed me that the new stable version is available. I clicked the update button and the update started. After running for some time I got a red notification that the update has failed.

#### What did you expect to happen?

I expected the update to succeed.

#### What happened instead?

The update failed.

#### Version of OctoPrint

OctoPrint 1.3.10 auf OctoPi 0.13.0 

#### Operating System running OctoPrint

OctoPrint 1.3.10 auf OctoPi 0.13.0 

#### Printer model & used firmware incl. version

Klipper Firmware on Kossel Printer

#### Browser and version of browser, operating system running browser

Firefox on Linux

#### Link to octoprint.log
https://gist.github.com/JustAnother1/ec759e38253632115d161d9430bd19d4

#### Link to contents of terminal tab or serial.log

also in the gist

I have read the FAQ.

JustAnother1, 2019-05-14:
This is probably the issue, but don't know why that happens:

OSError: [Errno 2] Datei oder Verzeichnis nicht gefunden: '/home/pi/oprint/local/lib/python2.7/site-packages/semantic_version-2.4.2-py2.7.egg'

Martin-Dolan, 2019-05-15:
Mine is just sitting there saying "updating" and has been for the last 20 minutes...

tedder, 2019-05-15:
And your logs, @Martin-Dolan ?

Martin-Dolan, 2019-05-15:
Here you go.

After a couple of reboots and retries it finally worked. hope it went ok.  Anyway it's at least printing again.

Here are the log files which should contain all the data from that. Hope it is a help to you. 
Let me know if you need anything else.

Cheers Marty

[Octoprint error.zip](https://github.com/foosel/OctoPrint/files/3180746/Octoprint.error.zip)

tedder, 2019-05-15:
Looks like you had- and still have- issues connecting to the package repositories. Do you have an "adult filter" or pihole or anything running?
```
2019-05-15 02:06:26,102 ! ReadTimeoutError: HTTPSConnectionPool(host='codeload.github.com', port=443): Read timed out.
```

For more info, I'd be curious what you get from these:
```
nslookup codeload.github.com
host codeload.github.com
curl -D - https://codeload.github.com/
```

foosel, 2019-05-15:
I can't look into update problems of any kind without 

  * `octoprint.log`
  * `plugin_softwareupdate_console.log`

from *everyone*.

[See here on where to find those](https://community.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/).

I also can't fix inherent network issues like not being able to connect to `codeload.github.com` or `pypi.org`, you need to solve them on your local network.

Martin-Dolan, 2019-05-15:
@tedder 
This is what I got:

marty@hassio:~$ nslookup codeload.github.com
Server:		127.0.0.53
Address:	127.0.0.53#53

Non-authoritative answer:
Name:	codeload.github.com
Address: 192.30.255.121

marty@hassio:~$ host codeload.github.com
codeload.github.com has address 192.30.255.121
marty@hassio:~$ curl -D - https://codeload.github.com/
HTTP/1.1 301 Moved Permanently
Content-Length: 0
Location: https://github.com/
Access-Control-Allow-Origin: https://render.githubusercontent.com
Content-Security-Policy: default-src 'none'; style-src 'unsafe-inline'; sandbox
Strict-Transport-Security: max-age=31536000
Vary: Authorization,Accept-Encoding
X-Content-Type-Options: nosniff
X-Frame-Options: deny
X-XSS-Protection: 1; mode=block
Date: Wed, 15 May 2019 10:19:14 GMT
X-GitHub-Request-Id: 8B76:1C7E:29C9A:55075:5CDBE7A2

foosel, 2019-05-15:
@Martin-Dolan You need to run these commands off the machine that is running OctoPrint or that information is meaningless. This looks like hassio, not OctoPi or some other common installation environment. Do you have OctoPrint manually installed on hassio? Your logs claim it's running on OctoPi, so you need to run those commands *on that*.

JustAnother1, 2019-05-15:
Good News! I tried the update again an this time it worked!

@foosel I understand that you can not fix the Internet for everyone.I also understand that you need the log files. Therefore I provided them.

But at least my log file shows a file not found error on a local file. That is not obviously caused by an internet issue. I suppose an earlier step responsible to gather the file from the internet silently failed. If that could be fixed that would be great.

I understand that this is an open source software that I get for free. Therefore I don't demand a fix. And I only reported this issue as I assumed that it was something in OctoPrint and I wanted to help fix this so that others have a better experience.

My internet connection is a very reliable (not something wireless mobile or experimental ) I also don't mess with my DNS server, no pihole,... But I'm long enough on the internet to know that even the servers with the highest reliability sometimes have a bad day. So maybe (not as a bug report, but rather as a feature request) it would be good if instead of relying on everybody else to function perfectly it would be better if OctoPrint could deal with others failing once in a while.

Thank you all that helped me and @Martin-Dolan resolve this. I will now continue to enjoy octoprint. Thank you for that !

foosel, 2019-05-15:
@JustAnother1 The thing is that in order to understand *why* a file that should really just be there as it's part of a third party dependency that is defined in OctoPrint's requirements isn't there, I need the update log. And I didn't realize that was part of the gist already (you pasted it as octoprint.log, I opened it and only saw the octoprint.log). Apologies for that.

This actually looks like a known issue when coming from old egg installs to new wheel installs which OctoPrint even includes a workaround for (if it sees that `egg` message it will perform a reinstall). Alas, that workaround couldn't trigger here thanks to your german locale - it's sadly not possible to add a parser for "File not found" in every language on the earth and the only thing I can run on here is the human readable error message. A manual reinstall in that case was indeed the solution. Just remembered that there was even a heads-up about this issue in [the release notes of 1.3.9](https://github.com/foosel/OctoPrint/releases/tag/1.3.9). 1.3.10 added the mentioned workaround but - again - not for anything but `en` locales. So we can consider this a known and unsolvable issue for non-en-locale installs that have seen quite a number of updates.

The network comment was targeted at the "me toos" that started popping up here and elsewhere.

JustAnother1, 2019-05-15:
Thank you for the explanation. Parsing for every language makes no sense. Reinstalling regardless of the error message (or if locale != `en` ) could be an option. But I also understand if you don't want to do that to not hide other issues.

Thanks again!

closed 2019-05-15 by JustAnother1