OctoPrint/OctoPrint #3152: [Request] Upload 'n Print! checkbox

HenryAlfredo, 2019-05-15:
First of all thanks for this software i really like using it.

My request is to add a checkbox which toggles to immediately start printing as soon as the .gcode upload has finished. 

Maybe somewhere near this text: _Hint: You can also drag and drop files on this page to upload them_

In the long run this will save some time, especially for people printing many design iterations daily.

foosel, 2019-05-15:
I'm not gonna add something like this to core since it can be dangerous (uncleared build plate). However it's something that someone could easily add through a plugin.

HenryAlfredo, 2019-05-15:
Yes i can understand that, had my fair share of collisions. Thanks again for the software and the new label.

dovecode, 2019-05-28:
@HenryAlfredo what I found useful was the [Autoselect plugin](https://github.com/OctoPrint/OctoPrint-Autoselect). It loads the last uploaded file, so all you need to do is click Print after uploading.

evanquinnalaska, 2019-05-31:
If you use Slic3r PE it has it avaliable out of the box :)
[image]

LazeMSS, 2021-03-31:
Like this: https://plugins.octoprint.org/plugins/AutostartPrint/

closed 2021-03-31 by LazeMSS