OctoPrint/OctoPrint #3158: ERROR - Error while saving config.yaml! After update 1.3.11

zotes, 2019-05-15:
Hello, I have updated to the latest version of octoprint 1.3.11 and when I have restarted the service the octoprint instance does not work. 
I looked at the log and I think there is a problem with the file "config.yaml".
Next I leave you the log and the "config.yaml".
OS: Debian 9
OctoPrint 1.3.11

I hope your help!
Thank You!

`2019-05-15 18:32:24,047 - octoprint.startup - INFO - ******************************************************************************
2019-05-15 18:32:24,051 - octoprint.startup - INFO - Starting OctoPrint 1.3.11
2019-05-15 18:32:24,052 - octoprint.startup - INFO - ******************************************************************************
2019-05-15 18:32:24,108 - octoprint.startup - INFO - Blacklist processing done, adding 1 blacklisted plugin versions: roomtemp (any)
2019-05-15 18:32:24,152 - octoprint.plugin.core - INFO - Loading plugins from /home/impresor4/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugins, /home/impresor4/.octoprint/plugins and installed plugin packages...
2019-05-15 18:32:25,270 - octoprint.plugin.core - INFO - Plugin Pi Support Plugin did not pass check, not loading.
2019-05-15 18:32:27,104 - octoprint.plugins.octoprint.plugins.discovery - INFO - pybonjour is not installed, Zeroconf Discovery won't be available
2019-05-15 18:32:27,240 - octoprint.settings - ERROR - Error while saving config.yaml!
Traceback (most recent call last):
  File "/home/impresor4/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/settings.py", line 1368, in save
    self._dirty = False
  File "/usr/lib/python2.7/contextlib.py", line 24, in __exit__
    self.gen.next()
  File "/home/impresor4/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/util/__init__.py", line 914, in atomic_write
    shutil.move(temp_config.name, filename)
  File "/usr/lib/python2.7/shutil.py", line 302, in move
    copy2(src, real_dst)
  File "/usr/lib/python2.7/shutil.py", line 130, in copy2
    copyfile(src, dst)
  File "/usr/lib/python2.7/shutil.py", line 83, in copyfile
    with open(dst, 'wb') as fdst:
IOError: [Errno 13] Permission denied: '/home/impresor4/.octoprint/config.yaml'
2019-05-15 18:32:27,240 - octoprint.plugin.core - ERROR - There was an error while enabling plugin errortracking
Traceback (most recent call last):
  File "/home/impresor4/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 1061, in enable_plugin
    plugin.enable()
  File "/home/impresor4/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugins/errortracking/__init__.py", line 200, in __plugin_enable__
    _enable_errortracking()
  File "/home/impresor4/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugins/errortracking/__init__.py", line 127, in _enable_errortracking
    s.save()
  File "/home/impresor4/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/settings.py", line 1368, in save
    self._dirty = False
  File "/usr/lib/python2.7/contextlib.py", line 24, in __exit__
    self.gen.next()
  File "/home/impresor4/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/util/__init__.py", line 914, in atomic_write
    shutil.move(temp_config.name, filename)
  File "/usr/lib/python2.7/shutil.py", line 302, in move
    copy2(src, real_dst)
  File "/usr/lib/python2.7/shutil.py", line 130, in copy2
    copyfile(src, dst)
  File "/usr/lib/python2.7/shutil.py", line 83, in copyfile
    with open(dst, 'wb') as fdst:
IOError: [Errno 13] Permission denied: '/home/impresor4/.octoprint/config.yaml'
2019-05-15 18:32:27,247 - octoprint.plugin.core - INFO - Found 24 plugin(s) providing 23 mixin implementations, 41 hook handlers
2019-05-15 18:32:27,488 - octoprint.settings - ERROR - Error while saving config.yaml!
Traceback (most recent call last):
  File "/home/impresor4/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/settings.py", line 1368, in save
    self._dirty = False
  File "/usr/lib/python2.7/contextlib.py", line 24, in __exit__
    self.gen.next()
  File "/home/impresor4/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/util/__init__.py", line 914, in atomic_write
    shutil.move(temp_config.name, filename)
  File "/usr/lib/python2.7/shutil.py", line 302, in move
    copy2(src, real_dst)
  File "/usr/lib/python2.7/shutil.py", line 130, in copy2
    copyfile(src, dst)
  File "/usr/lib/python2.7/shutil.py", line 83, in copyfile
    with open(dst, 'wb') as fdst:
IOError: [Errno 13] Permission denied: '/home/impresor4/.octoprint/config.yaml'
2019-05-15 18:32:27,489 - octoprint.cli.server - ERROR - Uncaught exception
Traceback (most recent call last):
[... 187 lines cut ...]
  - /dev/tty3DP*
  autoconnect: true
  baudrate: 115200
  disconnectOnErrors: false
  logPositionOnCancel: true
server:
  commands:
    serverRestartCommand: sudo /etc/init.d/octoprint4 restart
    systemRestartCommand: sudo shutdown -r -t 0
    systemShutdownCommand: sudo shutdown -h -t 0
  firstRun: false
  onlineCheck:
    enabled: false
  pluginBlacklist:
    enabled: true
  secretKey: 3qt4TiFFreuuEXDPtSQOX0DpZNoZ1GzK
  seenWizards:
    corewizard: 3
    cura: null
    telegram: 1
    tracking: null
webcam:
  ffmpeg: /usr/bin/ffmpeg
  snapshot: ./wcam/?action=snapshot
  stream: ./wcam/?action=stream`

zotes, 2019-05-15:
What were you doing?
I have updated to the latest version of octoprint 1.3.11 and when I have restarted the service the octoprint instance does not work.
I looked at the log and I think there is a problem with the file "config.yaml".

What did you expect to happen?
OctoPrint serve works

What happened instead?
OctoPrint serve doest work

Did the same happen when running OctoPrint in safe mode?
yes, it is the same bahaviour

Version of OctoPrint
1.3.11

Operating System running OctoPrint
Debian 9

Printer model & used firmware incl. version
Makerbot 2

Browser and version of browser, operating system running browser
Chrome Versión 74.0.3729.131 (Build oficial) (64 bits)

Link to octoprint.log and config.yaml
https://gist.github.com/zotes/9e02701ba93a82fa550582652ace1fcb

I have read the FAQ.

tedder, 2019-05-15:
Can you fix the `octoprint.log` so it appears with newlines and whitespace? If you look at the gist you'll see it is all smooshed onto one line, and things like exceptions lost all indentation.

tedder, 2019-05-15:
Also, give the output of this from ssh/terminal:
```
ls -al /home/impresor4/.octoprint/
```

DanielJoyce, 2019-05-15:
Very last line of log:

Permission denied: '/home/impresor4/.octoprint/config.yaml'

What are the directory permissions and what user is octoprint running as?

foosel, 2019-05-16:
Looks like a misconfigured custom install. If OctoPrint can't write to the configuration file due to a permission error, it obviously can't save the configuration file.

zotes, 2019-05-16:
OK. Thank you! The problem was that the owner of the file "config.yaml" was root instead of the user "impresor4".

closed 2019-05-16 by zotes