OctoPrint/OctoPrint #3159: [Request] Two button: G1 speed override button and G0 speed override button

DimchikUA, 2019-05-15:
Good day!
I want to offer some improvement for Octoprint. I think this will be useful to many users.
Octoprint has a remarkable function - to redefine a given speed of movement with the slider and the button.
[image]

The problem is that the button will redefine everything, and the speed of the working stroke G1 and the speed of idle movement G0. It is not good. It so happens that you need to increase or decrease the printing speed G1 (G2, G3, G5), and with it the idle speed of the head G0 changes, and this is not necessary. It happens and vice versa. The G1 print speed is good, but a lot of short, quick G0 moves and the printer start to shake violently. In this case, it is necessary to reduce only the fast speed G0, but not the printing speed G1.

**Describe the solution you'd like**
I propose to make two such buttons, one of which will control the speed of fast movement G0, and the other the speed of working stroke G1, etc. I think it should not be difficult.
Something similar is implemented in the program Repetier Host.
[image]
This is convenient, and if you implement it in Octoprint - it will be wonderful!

P.S.
Sorry Google translator, I tried to correct his mistakes :)

foosel, 2019-05-16:
> The problem is that the button will redefine everything, and the speed of the working stroke G1 and the speed of idle movement G0.

That slider/button uses [`M221`](https://reprap.org/wiki/G-code#M220:_Set_speed_factor_override_percentage) in order to set a factor override that is evaluated by your printer's firmware. Your printer's firmware doesn't distinguish between `G0` and `G1` here so OctoPrint can't either.

Side note: I don't see what you are requesting here in the screenshot from Repetier Host. The sliders present there look like the exact same functionality as OctoPrint provides: Look to the right in the "Tool" section of the Control panel and note the "Flowrate" slider right next to the "Feedrate" one.

closed 2019-05-16 by foosel

DimchikUA, 2019-05-16:
Oh how it is. I thought oktopint analyzes the sent G-code and replaces the values of F according to the coefficient.
Now I understand. I apologize for the open topic and time taken from you.

foosel, 2019-05-16:
No worries! Thank you for taking the time to think about how to improve OctoPrint and also for understanding why that particular approach isn't feasible :)