OctoPrint/OctoPrint #3161: Plugin list not correctly resized in Plugin Manager

yet-another-average-joe, 2019-05-16:
Sorry if this has already been reported, I searched for, but did'nt find anything...

#### What were you doing?

1. in the browser, open OctoPrint settings/Plugin Manager
2. Zoom the browser to 90% or more (ctrl+wheel)
3. The listbox containing the installed plugins is not fully visible ; buttons are hidden

#### What did you expect to happen?

The listbox and the buttons should be visible

#### What happened instead?

The listbox is partially visible, and the buttons are hidden (see uploaded pic below). This "frame" or its layout (listbox + buttons) is not correctly resized

**Workaround** : zoom the browser to 80% or less for Firefox, 75% or less for Slimjet

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPrint

1.3.11

maybe it existed before, but I did'nt notice...

#### Operating System running OctoPrint

Windows 10

#### Printer model & used firmware incl. version

#### Browser and version of browser, operating system running browser

Firefox 66.0.5 64bit
Slimjet 22.0.4.0 64bit

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

Slimjet, Zoom 90% +

[image]

Slimjet Zoom 80%

[image]

Slimjet Zoom 75%

[image]

Firefox 80%

[image]

tested on 2 OctoPrint installations, one on a "real" server (Pi 3B+), the other on a printerless one (dedicated to testings, virtual printer, Pi3B).

foosel, 2019-05-27:
Ah yeah, the size should probably rather adapt than be hard coded to avoid double scrollbars. Will look into this for 1.3.12.

foosel, 2019-05-27:
Solved for now via the above commit by reducing the height to 400px instead of 500px, which translates to the size of the paginated table that used to be there with 5 items height. 

Dynamic size adjustment would still be better, but so far I don't have a good idea how to do that ideally only with CSS (I'd rather prefer to avoid having to implement dynamic resizing in JS here).

foosel, 2019-10-22:
1.3.12 has been released.

closed 2019-10-22 by foosel