OctoPrint/OctoPrint #3162: [REQUEST] Keep dialogs open while a button is not clicked

yet-another-average-joe, 2019-05-16:
Hi,

I wanted to ask for this feature for a long time. It would be more usefull if TouchUI is installed.

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

It is related to the interface default behaviour. When clicking outside a dialog window / box, it closes. This is sometimes frustrating in a browser, it can be irritating on a touchscreen because of the lack of precision (approximative touch)

But maybe it is not possible...

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

An option to keep the dialogs open as long as close / cancel / validate / etc. is not clicked / touched / taped

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

I could shoot a short video if neeeded

foosel, 2019-05-27:
This has been implemented in form of a new optional setting:

[image]

and will be part of 1.3.12. Already available on `maintenance` & soon `devel`.

foosel, 2019-10-22:
1.3.12 has been released.

closed 2019-10-22 by foosel

yet-another-average-joe, 2019-10-22:
Many thanks for this option !