OctoPrint/OctoPrint #3165: pip installs old version

Patschke, 2019-05-17:
#### What were you doing?

Installing OctoPrint. 

In the first place I wanted to update from 1.3.10 to 1.3.11, but it failed for no good reason (however, I am not able to specify the state my 1.3.10 installation was in, so I am unable to create a proper bug report fot that). After a while I decided to simply uninstall/install OctoPrint.

Steps: 
1. pip install https://get.octoprint.org/latest

Plus some previous steps along the lines of apt and pip update and upgrade. Should not matter. 

#### What did you expect to happen?

Install the latest version of OctoPrint (1.3.11)

#### What happened instead?

Installed OctoPrint version 1.3.9

#### Did the same happen when running OctoPrint in safe mode?

Not required.

#### Version of OctoPrint

Ah, well. You know, that is the issue ;-)

#### Operating System running OctoPrint

Raspbian

#### Printer model & used firmware incl. version

Not relevant.

#### Browser and version of browser, operating system running browser

Not relevant.

Some more not relevant questions. BlahBla.

If you need any additional information, just ask. After the installation upgrading from 1.3.9 to 1.3.11 worked fine.

tedder, 2019-05-17:
Where's the pip log/console log?

Patschke, 2019-05-17:
I noticed that I pip installed an old version in the GUI, which I opend after rebooting, as I wanted to check if the autostart works properly. As I am not having --log enabled when installing stuff, I don't have either pip log or console log. I am not aware of any log files pip produced silently? There were nothing special in the output, just usual installation of dependencies and octoprint.

Patschke, 2019-05-17:
If you cannot reproduce this issue, I should be able to create another venv and install a copy of octoprint in there.

foosel, 2019-05-20:
Looks like pip caches HTTP responses. I've just set a bunch of do-not-cache headers on all of the URLs so that going forward that should no longer happen, however the existing cache will still be a problem.

You can get pip to ignore its HTTP cache with the `--no-cache-dir` option, e.g.

    pip install --no-cache-dir https://get.octoprint.org/latest

I'll adjust the guides accordingly.

foosel, 2019-05-20:
Guides have been adjusted, workaround see above.

closed 2019-05-20 by foosel

foosel, 2019-05-20:
Additional note: Seems like `--no-cache-dir` really just ignores the cache but doesn't update it. And there doesn't seem to be any option to do the latter besides cleaning the cache completely. You can do that by deleting

  * `~/.cache/pip` on Linux & Unix
  * `~/Library/Caches/pip` on OSX
  * `%LocalAppData%\pip\Cache` on Windows