OctoPrint/OctoPrint #3166: Anonymous Usage Tracking request Config shown on every connection

djkirkendall, 2019-05-18:
#### What were you doing?

1. Go to 'Network'
2. Click on 'Any Prusa Octoprint Instance'
3. Scroll down to 'N/A'
4. Window appears asking to enable/disable AUT

#### What did you expect to happen? I expected the setting to stick and for the prompt to never appear again

#### What happened instead? The prompt is shown and I have to select & wait for the selection to apply before I can close the window and get to my printer control

#### Did the same happen when running OctoPrint in safe mode? Yes

#### Version of OctoPrint

#### Raspbian, version unknown. Came with Prusa MK3 image downloaded from prusa printers.org. Connecting via wifi using Windows 10 machine and Chrome. 

#### Seen on

#### Browser and version of browser, operating system running browser

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

djkirkendall, 2019-05-18:
[image]
[image]
[image]

djkirkendall, 2019-05-18:
[image]
[image]

No matter what I select, window appears on next connection in chrome from Network:

[image]

tedder, 2019-05-18:
We need your `octoprint.log`, and also need to know what version of Octoprint you're running. You said "Came with Prusa MK3 image downloaded from prusa printers.org"; if you don't know the version you may have to ask over there for help- we can help you with bugs in relatively current versions here, or [can help with tech support type issues in the octoprint forums](https://community.octoprint.org/).

Do you really have 15 Octoprint systems or is that part of the problem you're talking about?

djkirkendall, 2019-05-18:
Actually, I have 40 but the rest are powered down. I have a mix of Raspberry Pi Zero W (16 printers) and Raspberry Pi 3B (22 printers). It appears on both boards when I updated to 1.3.11.

I'm currently regressing 1.3.9 as I type this. I'll pull the logs from a machine as soon as it finishes. I'm running PrintHistory and DisplayZ plugins, but that's it.

djkirkendall, 2019-05-18:
As you can imagine, I close/reopen sessions from Windows Network as I queue up jobs, so this bug is particularly annoying as it adds 30 seconds to every transaction.

djkirkendall, 2019-05-18:
Here are the logs from #5, still running 1.3.11 and Octopi 0.15.0PE.

[octoprint (1).log](https://github.com/foosel/OctoPrint/files/3193633/octoprint.1.log)

djkirkendall, 2019-05-18:
Here are the logs from a RPi 3B running Octopi 0.14.0 and Octoprint 1.3.11. 
[octoprint (2).log](https://github.com/foosel/OctoPrint/files/3193634/octoprint.2.log)

tedder, 2019-05-18:
Hmm. Neither of those show errors. I'm curious what they have in .oprint/config.yaml in the plugins->tracking section. Is that section there? What are the values? You can blank out part of the unique id if they have one, but make sure to tell me if the value was empty or not.

djkirkendall, 2019-05-18:
The RPi 3B was imaged from the 2017-06-21-octopi-jessie-lite-0.14.0 with no changes made with the exception of adding "FilamentSensorReloaded", "PrintHistory", "DisplayZ", and "Marlin EEPROM Editor" plugins and upgrading from 1.3.9 to 1.3.11. 

[image]

djkirkendall, 2019-05-18:
I logged into CR10 #5 after looking at this, and it popped up the AUT window. The sequencing is strange in that the octoprint icon in chrome starts out green and shows the usual gui without any incoming data, then the icon turns brown and the AUT window pops up.

tedder, 2019-05-18:
The screenshot shows an empty uuid. There was a bug and it should have
resolved by itself, but go set a uuid in there, restart, try again.

After that we may need to be in a screenshare (or better, get me ssh in).

djkirkendall, 2019-05-18:
Yeah, I blanked it out.

djkirkendall, 2019-05-18:
I don't suppose this is as simple as testing at startup for "True" vs. "true"? I haven't dug into the code to check, but it would be easy to miss.

djkirkendall, 2019-05-18:
I've regressed my farm to 1.3.9 and am no longer in a position to collect data for you, sorry. I really like what was done with the Pi issue reporting, but this bug was driving me nuts and overshadowed all of the new goodness. I guess I'll wait for 1.3.12 and be more cautious about testing before rollout. All in all, this cost me about 12 hours of labor to roll out and then regress.

foosel, 2019-05-20:
OctoPrint will only show this wizard if `plugins.tracking.enabled` is neither `true` nor `false` (or rather, if it's `None`, which is the default if it's not explicitly set). From the looks of your `config.yaml` file it seems to be set just fine however. Which makes me wonder if there might be a caching issue at play here that has the wizard code populate even though it's not needed.

When you select one of the offered options, what happens next? It should disable the buttons and show you your decision in a line below. Does that happen?

Also, can you please share your browser's JS error console and also its Network tab from a run through the wizard?

I cannot reproduce this at all and your report is also the first and so far only instance of a not going away wizard, so if this is to be fixed I'll need your help in figuring out what's going on in the first place I fear.

Side note: you can also disable the tracking plugin by adding it to the `plugins._disabled` like this:

``` yaml
plugins:
  _disabled:
    - tracking
```

(just create the `_disabled` key if it doesn't yet exist)

That should make versions with it on board work for you.

djkirkendall, 2019-05-21:
I can do that, I have 1 machine I missed in the regression to 1.3.9. 

When the wizard appears on connection, I can select either option then finish to close it. It just reappers on the next browser instance into that same machine. 

How do I share the JS error console? The network tab?

foosel, 2019-05-21:
> It just reappers on the next browser instance into that same machine.

What happens if you go through that in incognito mode, if you clear the cache or if you switch to a completely different browser? What browser are you even using, with what extensions and on what OS (Windows, ok, but which one)? Your ticket lacks that information and it could be crucial here.

> How do I share the JS error console? The network tab?

Let's try screenshots first. Word of warning, I might have to ask you for some more deep diving.

djkirkendall, 2019-05-22:
Different Browser window - appears
Incognito - does not appear

foosel, 2019-05-22:
> Different Browser window - appears
> Incognito - does not appear

The plot thickens. That really sounds like a caching issue then. The question is why. Could you try holding the shift-key down and clicking the reload button in an affected browser window? Does the issue persist or does it go away?

djkirkendall, 2019-05-22:
Held shift and reloaded, did not appear. Opened new window after this, and it didn't appear.

foosel, 2019-05-22:
Then it's your browser for some reason deciding to ignore all cache headers that OctoPrint puts in (and there's quite some redundancy...) and caching to show the wizard. Which is weird and something I've so far not seen but since shift-reload solves it (which is a hard reload clearing the cache) it indeed appears to be the case.

I'll poke around a bit with Chrome, maybe I can figure something out, but I can't promise anything here.

djkirkendall, 2019-05-23:
Installed fresh instance of 1.3.11 on a regressed machine, settings were retained from previous install. 

At this point, with a valid (and apparently one-time) workaround, I'll roll this back out on my farm. 

Thank you so much for the help!!

yschroeder, 2019-06-07:
Just chiming in, as I had the same problem:

The config wizard also appeared on my machine every time I opened the website in the browser.

`Ctrl` + `F5` solved the problem.

foosel, 2019-06-07:
@yschroeder what browser on what OS?

msloth, 2019-06-13:
Had the same problem, using Chrome on w10. `Shift+F5` to hard reload solved it for me too. Thanks!

yschroeder, 2019-06-13:
Sorry for the late answer: Firefox on Arch Linux

stuartpb, 2019-06-16:
To be clear, this issue occurs in Firefox, too.

foosel, 2019-07-02:
I think #3211 is a duplicate of this issue, and while looking into that I was finally able to reproduce the behaviour (and have no idea why I wasn't able to earlier).

I've pushed 72167d14d2796e8527f09d183107fad99c0ab2fa which should fix this. I also pushed 59ae7e39f229310dab11ebf15c8b2959decd3aab and 530fca95e49000bead67bf677844f5f80a037b1f for some additional safeguarding against issues like this.

Pushed to `maintenance`, soon `devel`, to be released with 1.3.12.

Until then, a forced refresh (holding shift & clicking on the reload button) should solve this.

RealHabix, 2020-10-07:
Related. Keep getting the same message under different wizard..

[image]

There is no option to enable or disable it.. as you see the wizard behind the message is showing totally different content.

foosel, 2020-10-07:
Please open a new ticket.