OctoPrint/OctoPrint #3169: pyserial, error: (22, 'Invalid argument') with Klipper

RedSerenityDev, 2019-05-18:
#### What were you doing?

I usually install Octopi + Klipper on a RaspberryPi (and have never had any issues) but this time I wanted a more powerful system to run multiple instances of OctoPrint + Klipper.

I just did a fresh install of Octoprint on a CentOS 7 Dell laptop using the instructions from https://community.octoprint.org/t/setting-up-octoprint-on-a-raspberry-pi-running-raspbian/2337 (substituting yum for apt).

#### What did you expect to happen?

Octoprint to connect and be able to communicate with my printer.

#### What happened instead?

After installation, everytime I tried to connect to the printer I would get this error:

```
2019-05-18 11:43:27,449 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Error: Connection error, see Terminal tab"
2019-05-18 11:43:27,457 - octoprint.util.comm - ERROR - Unexpected error while connecting to serial port: /tmp/printer1 error: '(22, 'Invalid argument')' @ comm.py:_openSerial:2611 (hook default)
Traceback (most recent call last):
  File "/home/printer/OctoPrint1/venv/lib/python2.7/site-packages/octoprint/util/comm.py", line 2611, in _openSerial
    serial_obj = factory(self, self._port, self._baudrate, settings().getFloat(["serial", "timeout", "connection"]))
  File "/home/printer/OctoPrint1/venv/lib/python2.7/site-packages/octoprint/util/comm.py", line 2600, in default
    serial_obj = serial.Serial(**serial_port_args)
  File "/home/printer/OctoPrint1/venv/lib/python2.7/site-packages/serial/serialutil.py", line 240, in __init__
    self.open()
  File "/home/printer/OctoPrint1/venv/lib/python2.7/site-packages/serial/serialposix.py", line 272, in open
    self._reconfigure_port(force_update=True)
  File "/home/printer/OctoPrint1/venv/lib/python2.7/site-packages/serial/serialposix.py", line 438, in _reconfigure_port
    [iflag, oflag, cflag, lflag, ispeed, ospeed, cc])
error: (22, 'Invalid argument')
```

#### Did the same happen when running OctoPrint in safe mode?

Yes

#### Version of OctoPrint

1.3.11

#### Operating System running OctoPrint

CentOS 7 w/ latest updates

#### Printer model & used firmware incl. version

Ender 3, with Klipper 0.7.0 firmware

#### Browser and version of browser, operating system running browser

Chrome (Latest)

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem

#### Debugging

While trying to figure out the problem I came across https://github.com/foosel/OctoPrint/issues/3119 and after reading through the thread, I made the code changes that cevich did which immediately fixed my problem. But, as cevich said in that thread, this is not a good long-term solution.

Since the reason for using PARITY_ODD is to work around a Debian bug, I understand why you wouldn't want to change that part of the code. Would it be possible to instead have a configuration option in the serial settings to specify when your using a PTY and only use PARITY_NONE? I think this would provide a clean solution to work around both issues easily.

Thanks!

foosel, 2019-05-20:
Sounds reasonable. I've just pushed the above patch (and 1b82aa859fbe8bc9c8fe868c29dbd8e8d7ac4dd9 with a small fix of it) that does allow more granular control over this particular workaround.

[image]

RedSerenityDev, 2019-05-21:
Awesome! Thank you for the quick fix!

closed 2019-05-21 by RedSerenityDev

foosel, 2019-05-21:
I forgot to mention, this will be released as part of 1.3.12

ipatch, 2019-12-07:
> Sounds reasonable. I've just pushed the above patch (and [1b82aa8](https://github.com/foosel/OctoPrint/commit/1b82aa859fbe8bc9c8fe868c29dbd8e8d7ac4dd9) with a small fix of it) that does allow more granular control over this particular workaround.
> 
> [image]

wow I wish i would have found this a couple of days ago. this works great when using the physical GPIO pins on the raspberry pi to connect to serial port on a board, for my case the SKR v1.3 🙏

commit 12be637f (2019-05-20): ✨ Add setting for parity double open workaround

Can now be set to be applied if possibly needed (OS is a Debian Linux),
always or never.

Closes #3169, enhances #3167